google-deepmind/mujoco #901: Help needed viewer AttributeError__enter__

mertalagozlu, 2023-06-04:
**How to ask for help**

First, read our quick guide to
[asking good questions](https://github.com/deepmind/mujoco#asking-questions).
Below is a template for you to use:

Hi,

I'm a student and I'm trying to use MuJoCo for academic purposes.

I'm looking for some help with mujoco viewer. I am getting '    with mujoco.viewer.launch_passive(self.model, self.data) as viewer: AttributeError    with mujoco.viewer.launch_passive(self.model, self.data) as viewer:: 
AttributeError__enter__ : __enter__'. The code is basically exactly the same code in the documentation.

Here is a code, which explains my question:

 def simulate(self):
        with mujoco.viewer.launch_passive(self.model, self.data) as viewer:
            while viewer.is_running():
                step_start = time.time()
                for i in range(len(self.data.ctrl)):  # get number of controls
                    if self.predicted_values is not None:
                        self.data.ctrl[i] = self.predicted_values[i]

Thank you very much

kevinzakka, 2023-06-04:
What MuJoCo version are you using?

mertalagozlu, 2023-06-04:
version is 2.3.3

kevinzakka, 2023-06-04:
You need 2.3.5 (actually at least 2.3.4 but skip that one) for the context manager.

closed 2023-06-04 by kevinzakka