google-deepmind/mujoco #905: Use set_mjcb_sensor() to make a user sensor

MissMulledWine, 2023-06-05:
Hi,

I am stuck and lost :(

I'm trying to make a user sensor, that would work the same way as rangefinder works, but additionally would provide me with geom_id of the nearest geom. 

This is XML part of code, where I have defined the user sensor: 

```
<sensor>
    <user name="my_range" datatype="real" needstage="pos" dim="2"/>
</sensor>
```

I wrote a function that outputs distance and geom_id using mj_ray(), and used the callback function **set_mjcb_sensor()** inside of it: 

```
def my_sensor_call(model, data, site_name):
    site = data.site(site_name)
    pnt = site.xpos
    vec = site.xmat.reshape((3, 3))[:, 2]
    bodyexclude = model.site_bodyid[site.id]
    geomid_out = np.array([-1], np.int32)
    distance = mujoco.mj_ray(
        model, data, pnt, vec, geomgroup=None, flg_static=1,
        bodyexclude=bodyexclude, geomid=geomid_out)

    # callback function for user sensor

    mj.set_mjcb_sensor([distance, geomid_out[0]])
    return distance, geomid_out[0]
```

Then, If I call my func outside of mj.step() loop:

`my_sensor_call(model, data, 'site_sensor')`

I'm getting this error:

`ERROR: mj_ray: vector length is too small
`

And if I am trying to call the function inside of the loop:

```
while not glfw.window_should_close(window):
    simstart = data.time
    while (data.time - simstart < 1.0 / 60.0):
        # simulation step
        mj.mj_step(model, data)
        # Apply control
        data.ctrl[1] = 0.1
        data.ctrl[0] = -0.1
        mj.mj_fwdPosition(model, data)
        my_sensor_call(model, data, site_name='site_sensor')

    if (data.time >= simend):
        break
```

I'm getting this error: 

```
Traceback (most recent call last):
  File .../biped_main.py", line 276, in <module>
    my_sensor_call(model, data, site_name='site_sensor')
  File .../biped_main.py", line 261, in my_sensor_call
    mj.set_mjcb_sensor([distance, geomid_out[0]])
TypeError: callback is not an Optional[Callable]
```

I am really confused and would appreciate any help.
Thanks a lot!

closed 2023-06-09 by yuvaltassa

MissMulledWine, 2023-06-09:
Thank you very much!

I have read the test for Python callbacks and to be honest, as a beginner
it's quite challenging to understand a thing...
I understand that there is no need to call the callback function, however I
am still struggling to understand where and how I should provide my
function, so that later it can be called by the MuJoCo engine.

I am ashamed to ask, but if you could provide me with a code snippet, I
would really appreciate it!

saran-t, 2023-06-09:
The (legacy) `mjcb_sensor` API is unfortunately not modular, in the sense that MuJoCo only calls the callback once per computation stage, rather than per sensor.

The practical upshot of this is that your callback function needs to manually write into _all_ `<user>` sensordata slots in the `mjData`. There's also no way to specify in XML which site the ray should be cast from.

If all you need is a single `<user>` sensor in your entire model that just shoots a single ray from a single, fixed site, then this is sufficient. Otherwise, you'll likely find that the API is lacking (it's still doable, but it'll be rather nasty).

Our [sensor plugin](https://mujoco.readthedocs.io/en/stable/programming/extension.html) mechanism provides more flexibility to support exactly this sort of use cases, but at the moment you can only implement plugins in C or C++. See @yuvaltassa's [touch sensor](https://github.com/deepmind/mujoco/tree/main/plugin/sensor) for example.

MissMulledWine, 2023-06-09:
Thank you for your support!
Since I wanted to implement multiple user sensors then, as you said, it's
gonna be really challenging (especially for a beginner like me :) )

So it does make more sense just to use the mj_ray() function for all the
computations. However, in the simulation, calling the function doesn't give
any visual "ray" - representation. Is this also "doable, but nasty"? Or is
there an option/flag that could easily display those rays?

And again, I appreciate your support and guidance.

saran-t, 2023-06-09:
If you don't need your results written into the `sensordata` of `mjData`, your easiest option is just to call `mj_ray` yourself after each `mj_step` call, and store your results in your own NumPy array (for example).

That's pretty straightforward.

MissMulledWine, 2023-06-09:
Yeap, that's how I've already implemented it before, but visually those
rays can't be seen, whereas "range-finder" yellow rays from sensors can be
seen, if we simply set mjVIS_RANGEFINDER flag to be equal to 1. So, I was
curious if there is such an option for the mj_ray() function.

yuvaltassa, 2023-06-09:
You can draw 3D things directly into the mjvScene. There are a bunch of issues discussing this and a cell in the tutorial colab showing how to do it. Search in engine_vis_visualize.c for where the rangefinder rays are drawn. You just need to reproduce those couple of lines in Python. Would send you line numbers but am AFK (TGIF 🙂)

MissMulledWine, 2023-06-09:
I mean I am already very grateful for your help! Just wanted to make sure
that it's "doable". If so , then I can spend a couple of hours trying to
implement those things. Otherwise it would have been such a waste of time,
trying to implement something that is impossible to do in MuJoCo!

Thank you guys for your time!

yuvaltassa, 2023-06-12:
Definitely doable :slightly_smiling_face: 
Let us know if you get stuck.