google-deepmind/mujoco #908: Unknown actuator lag

le-horizon, 2023-06-07:
Hi,

I'm an engineer and I'm trying to use MuJoCo for sim to real deployment.

I'm looking for some help with actuator torque lag in simulation.

Here is the relevant part of the model xml:

```XML
        <default class="actuator">
          <general dyntype="none"
                   dynprm="0.005"
                   biastype="affine"
                   gainprm="50"
                   biasprm="0 -50 -1.0"
                   ctrlrange="-2.69653 -0.916298" forcerange="-35.55 35.55" />
        </default>
```

The response of the actuator lags behind the physical actuator.  See torque in orange, it is flat and near zero from t = 0 to 8.
Any idea what might be causing this lag?

Here is a screenshot, illustrating my question.

In the plot, I plotted both the command (target position) and the joint position reading, as well as the torque from joints.qfrc_actuator.

[image]

I'm using mujoco 2.3.2 on ubuntu 22.04.

Thanks,
Le

kevinzakka, 2023-06-07:
Why is dynprm != 1?

le-horizon, 2023-06-07:
> Why is dynprm != 1?

Will it matter if dyntype="none"?  I tweaked it a bit previously to see its effect, when I set dyntype="filter".

kevinzakka, 2023-06-07:
Could you post the full xml? I can investigate.

le-horizon, 2023-06-07:
We've tried using position control in the actuator, with similar lagged response.

Here it is the xml, thanks!

`<mujoco model="go1_description">
    <compiler angle="radian" meshdir="./meshes/" autolimits="true" />
    <option cone="elliptic" impratio="100"/>
    <size njmax="500" nconmax="100" />
    <option gravity='0 0 -9.806' iterations='50' solver='Newton' timestep='0.002'/>

    <default>
        <geom contype="1" conaffinity="1" friction="0.6 0.3 0.3" rgba="0.5 0.6 0.7 1" margin="0.001" group="0"/>

        <motor ctrlrange="-33.5 33.5" ctrllimited="true"/>
        <camera fovy="60"/>
        <joint damping="0.01" armature="0.01" frictionloss="0.2" />

        
        <default class="actuator_on_hip">
          <general dyntype="none"
                   dynprm="0.005"
                   biastype="affine"
                   gainprm="50"
                   biasprm="0 -50 -1.0"
                   ctrlrange="-0.802851 0.802851"
                   forcerange="-23.7 23.7" />
        </default>
        <default class="actuator_on_thigh">
          <general dyntype="none"
                   dynprm="0.005"
                   biastype="affine"
                   gainprm="50"
                   biasprm="0 -50 -1.0"
                   ctrlrange="-1.0472 4.18879" forcerange="-23.7 23.7" />
        </default>
        <default class="actuator_on_calf">
          <general dyntype="none"
                   dynprm="0.005"
                   biastype="affine"
                   gainprm="50"
                   biasprm="0 -50 -1.0"
                   ctrlrange="-2.69653 -0.916298" forcerange="-35.55 35.55" />
        </default>
        <default class="foot">
          
          <geom type="sphere" size="0.02" pos="0 0 -0.213" priority="1" solimp="0.015 1 0.01"
              condim="6" friction="0.8 0.02 0.01" rgba="0 0 0 1"/>
        </default>
    </default>
    <asset>
        <mesh name="trunk" file="trunk.stl" />
        <mesh name="hip" file="hip.stl" />
        <mesh name="thigh_mirror" file="thigh_mirror.stl" />
        <mesh name="calf" file="calf.stl" />
        <mesh name="thigh" file="thigh.stl" />
    </asset>

    <asset>
        <texture type="skybox" builtin="gradient" rgb1="1.0 1.0 1.0" rgb2="1.0 1.0 1.0" width="512" height="512"/>
        <texture name="plane" type="2d" builtin="flat" rgb1="1 1 1" rgb2="1 1 1" width="512" height="512" mark="cross" markrgb="0 0 0"/>
        <material name="plane" reflectance="0.0" texture="plane" texrepeat="3 3" texuniform="true"/>
    </asset>

    <worldbody>
        
        <body name="trunk" pos="0 0 0.325">
            <inertial pos="0.0116053 0.00442221 0.000106692" quat="0.0111438 0.707126 -0.00935374 0.706938" mass="1.801" diaginertia="0.0447997 0.0366257 0.0162187" />
            <freejoint/>
            <geom type="mesh" contype="0" conaffinity="0" group="1" rgba="0.913725 0.913725 0.847059 1" mesh="trunk" />
            <geom size="0.13 0.04675 0.057" type="box" rgba="0.913725 0.913725 0.847059 1" />
[... 132 lines cut ...]

        <jointvel name="FR_hip_vel"     joint="FR_hip_joint"/>
        <jointvel name="FR_thigh_vel"   joint="FR_thigh_joint"/>
        <jointvel name="FR_calf_vel"    joint="FR_calf_joint"/>
        <jointvel name="FL_hip_vel"     joint="FL_hip_joint"/>
        <jointvel name="FL_thigh_vel"   joint="FL_thigh_joint"/>
        <jointvel name="FL_calf_vel"    joint="FL_calf_joint"/>
        <jointvel name="RR_hip_vel"     joint="RR_hip_joint"/>
        <jointvel name="RR_thigh_vel"   joint="RR_thigh_joint"/>
        <jointvel name="RR_calf_vel"    joint="RR_calf_joint" />
        <jointvel name="RL_hip_vel"     joint="RL_hip_joint"/>
        <jointvel name="RL_thigh_vel"   joint="RL_thigh_joint"/>
        <jointvel name="RL_calf_vel"    joint="RL_calf_joint"/>

        <accelerometer name="Body_Acc" site="imu"/>

        <gyro name="Body_Gyro" site="imu"/>

        <framepos name="Body_Pos" objtype="site" objname="imu"/>

        <framequat name="Body_Quat" objtype="site" objname="imu"/>

    </sensor>
</mujoco>
`

yuvaltassa, 2023-06-07:
Do your analysis with step response, not these smooth curves.

From what I can tell this is just the delay coming from the 2nd orderness of the physics, but very hard to judge

robopianist, 2023-06-07:
FWIW, I've been working with the Go1 model for a few months now and have not experienced any control delays so unclear what is going on here.

yuvaltassa, 2023-06-07:
In that case, could it not be that the delay you're seeing is just gravity opposing your commands until they are actually strong enough to make the robot start moving?

Can you retry with [gravity disabled](https://mujoco.readthedocs.io/en/latest/XMLreference.html#option-gravity)?

breakds, 2023-06-07:
> In that case, could it not be that the delay you're seeing is just gravity opposing your commands until they are actually strong enough to make the robot start moving?
> 
> Can you retry with [gravity disabled](https://mujoco.readthedocs.io/en/latest/XMLreference.html#option-gravity)?

Thanks for the prompt response, @yuvaltassa. I might have understood this wrong - I thought `qfrc_acutator` is the torque generated by the actuator (e.g. a motor) instead of the net torque, is this right? Or is it the net torque (motor generated torque + torque generated by the gravity and others)?

le-horizon, 2023-06-07:
Even if I set gainprm and bias to 200 (from 50), the initial lag is still there, it's just less smooth.  So it's hard to imagine it's due to gravity:
[image]

breakds, 2023-06-07:
Thank you for taking the time to explain `qfrc_actuator` and the factors affecting the torque output. I appreciate your patience and your insightful advice. We will try to do more experiments.

Apologize if our questions seemed to over-complicate the issue or if it seemed like we were bypassing efforts to understand the model in a more basic setting. Our intention was merely to seek some expert guidance as we are not as knowledgeable in control systems.

le-horizon, 2023-06-08:
We discovered the reason for the mismatch.  We have domain randomization in simulation, we turned off some, but did not turn off center of mass randomization, which causes the responses to be different from real.

Clean step response matches real quite accurately.

The qfrc_actuator being zero initially is still counter intuitive, but does not seem to affect our energy computation in any significant way, given that the speed is zero when torque is zero.  There might be small differences in energy calculation, but the majority of the times, it seems to be correct.

Thanks for taking a look and the advices.

closed 2023-06-08 by le-horizon