google-deepmind/mujoco #909: Incorrect Return Code

nicholask42, 2023-06-08:
In `basic.cc`, the main function returns `1` at the end. 

https://github.com/deepmind/mujoco/blob/833d30858db7c4588a105199d6993a65842ad563/sample/basic.cc#LL189C10-L189C10

It seems like this return would not be an error. So, shouldn't it be returning `0`? Or is it an error if the window is closed and the end of `main()` is reached?

Also, I would recommend using `EXIT_SUCCESS` and `EXIT_FAILURE` instead of `0` or `1`.

saran-t, 2023-06-08:
Sure, send us a PR.

nicholask42, 2023-06-08:
Thanks. Just wanted to make sure I was understanding the use of the Mujoco library in the samples. I have created a PR for the changes: https://github.com/deepmind/mujoco/pull/910.

commit 5187a712 (2025-07-14): Standardize exit codes for samples.

Fixes #909 #910

PiperOrigin-RevId: 782951003
Change-Id: I093c0032f665bf34fecf71a75012b2b0c4c22aba
closed 2025-07-14 by copybara-service[bot]