google-deepmind/mujoco #912: Option to re-scale the size of markers for contact, force, perturbations, etc...

PeterMitrano, 2023-06-08:
Hi,

I'm a phd student using the Panda robot model from the Menagerie. I've noticed that for this and most other robot models I've tried, the scale of the visualization is really large, and obstructs the view of what's in contact, for example. The screenshot below shows that I mean (the two robot are holding tools, but you can hardly see anything with the giant markers).

Is there any way to adjust this without compiling from source? Could we expose a parameter for this in the viewer somewhere, like with font scale?

[image]

v-r-a, 2023-06-08:
Hi, I'm a MuJoCo user.

See: 
https://mujoco.readthedocs.io/en/latest/XMLreference.html#visual-scale

You can set this by adding the following to the XML (panda.xml):

```
<visual>
       <scale forcewidth = "0.1"  contactwidth = "0.1" contactheight = "0.1" />
 </visual>
```

closed 2023-06-08 by PeterMitrano

yuvaltassa, 2023-06-09:
btw, the latest version of the viewer lets you adjust these values in the GUI. Until the next release you'll need to build it from source.

yuvaltassa, 2023-06-09:
Ah there's even a screencast 🙂 
https://youtu.be/mXVPbppGk5I

happygaoxiao, 2023-07-06:
Hello, I watched your video. But I cannot have the option of `Visualization` below `Render`, with MuJoCo 2.3.5.
[image]

By the way, there is a function to add markers in [mujoco-py](https://github.com/openai/mujoco-py/blob/a13903b82f1ab316815e63b3575526005b3b2ae1/mujoco_py/mjrendercontext.pyx#L249) . We can set several types of markers. Is there any similar functions to do this? I cannot find any options.

Here is an example where I did with `mujoco-py`. The goal is to visualize all contact points during the hand exporation.
[image]

yuvaltassa, 2023-07-07:
This was released in MuJoCo 2.3.6