google-deepmind/mujoco #913: Reverse mouse scroll

Melanol, 2023-06-09:
My brain fails when I try to zoom in and out with my mouse scroll wheel, and the controls are reversed compared to everything else humanity has created.

Please add a way to reverse.

saran-t, 2023-06-09:
> the controls are reversed compared to everything else humanity has created.

This isn't necessarily true, the scroll directions are fairly standard before Apple started pushing their "natural" scrolling directions on macOS... FWIW I find the Apple directions counterintuitive outside of touchscreens, so this is a personal preference thing.

Having said that, we can consider adding an option.

yuvaltassa, 2023-06-09:
We could do something like 
```xml
<visual>
  <global naturalscroll="false/true">
</visual>
```
but that's weird, since this is not a model setting, but a user preference.

@Melanol do you build from source?

Is would be extremely easy to add this as a constant that you can flip. In the meantime can you replace `-=` with `+=` on [this line](https://github.com/deepmind/mujoco/blob/466c9aca82bc63a2a07f76cb1eb6d179a3bfa8b2/src/engine/engine_vis_interact.c#LL346C2-L346C2) and report if this is what you want?

saran-t, 2023-06-09:
Ideally we'd just look at the system setting to check if the user has natural scrolling enabled and respect that. No need for MuJoCo-specific knob that way.

Someone needs to research the right API to call though (should hopefully be straightforward on Windows and macOS, as always there's fragmentation on Linux so we can leave that aside for now).

Melanol, 2023-06-09:
> Ideally we'd...

Can't I just have a user preference?

yuvaltassa, 2023-06-09:
I agree that looking at system settings is tricky. E.g. on my machine I have different conventions for mouse and trackpad, and this is going to be super architecture specific and tricky to sort out.

@Melanol,

1. Can you please try my suggestion above and let us know if this is functionally what you want?
2. Where would you expect to pass in this user setting?

Melanol, 2023-06-09:
> 1. Can you please try...
I have 0 experience building from source. I just wanted to propose a feature.

Melanol, 2023-06-10:
> @Melanol do you build from source?
>
> Is would be extremely easy to add this as a constant that you can flip. In the meantime can you replace `-=` with `+=` on [this [...]

I have built it with `+=` and it is exactly what I wanted.

closed 2024-01-08 by yuvaltassa

GF-Huang, 2026-10-17:
Issue still 2026 on Windows 11.

```
> conda list | findstr mujoco
mujoco                       3.11.0           pypi_0                         pypi
mujoco-menagerie             2026.10.3        pypi_0                         pypi
mujoco-mjx                   3.8.1            pypi_0                         pypi
mujoco-warp                  3.11.0           pypi_0                         pypi
```