google-deepmind/mujoco #918: How to disable warm-start when doing mj_step()

Chuanfang-Neptune, 2023-06-13:
Hello,

I'm a student trying to use MuJoCo for muscle simulations and compare results with that from OpenSim Forward Dynamics. I encountered problems in disabling warm-start of the solver.

I applied actuation to a muscle actuator, used `mj_step()` to go one step and used `mj.MjData.joint()` to inspect joint states. I observed that I resulted in a very different `qacc_smooth` than `qacc_warmstart`. It seemed from [source code](https://github.com/deepmind/mujoco/blob/643400174763da17e0f226d3615b33d594ddb27b/src/engine/engine_forward.c#L447) that `qacc` was determined between the better of `qacc_smooth` and `qacc_warmstart`. I supposed `qacc_warmstart` was leading my result away from what I got in OpenSim and wanted to disable it.
```
  name: 'joint_LFTibia_pitch'
  qLDiagInv: array([95.32631846])
  qLDiagSqrtInv: array([9.76351978])
  qacc: array([728.55982479])
  qacc_smooth: array([4.06800485])
  qacc_warmstart: array([728.55982479])
  qfrc_actuator: array([0.01815944])
  qfrc_applied: array([0.])
  qfrc_bias: array([-0.01939574])
  qfrc_constraint: array([0.])
  qfrc_inverse: array([0.])
  qfrc_passive: array([-0.00010684])
  qfrc_smooth: array([0.03744834])
  qpos: array([1.06862884])
  qvel: array([0.36364967])
```
  I followed the [XML Reference](https://mujoco.readthedocs.io/en/latest/XMLreference.html#option-flag) to add a "disable" flag in option in my model for warm-start as below:
```
<option timestep="0.0005" gravity = "0 0 -9.8" >
      <flag warmstart = "disable"/>
</option>
```
I also checked with `bool(m.opt.disableflags & (1<<8))` which returned `true` that warmstart was indeed disabled. However, when I did `mj_step()` again and inspected the joint state with `mj.MjData.joint()`, I found that `qacc` was still taken to be the same as `qacc_warmstart`. The warmstart disable seemed not to work.
```
  name: 'joint_LFTibia_pitch'
  qLDiagInv: array([95.32631846])
  qLDiagSqrtInv: array([9.76351978])
  qacc: array([1068.76783786])
  qacc_smooth: array([2.36937381])
  qacc_warmstart: array([1068.76783786])
  qfrc_actuator: array([0.01790641])
  qfrc_applied: array([0.])
  qfrc_bias: array([-0.01845275])
  qfrc_constraint: array([0.])
  qfrc_inverse: array([0.])
  qfrc_passive: array([-0.00374336])
  qfrc_smooth: array([0.03261581])
  qpos: array([1.06907697])
  qvel: array([0.89624287])
```
Do you know what is the correct way to disable warm-start?

yuvaltassa, 2023-06-14:
https://mujoco.readthedocs.io/en/latest/XMLreference.html#option-flag-warmstart

yuvaltassa, 2023-06-14:
Ah sorry I misread, I see now that you already tried that. 

To clarify, `qacc` is always copied into`qacc_warmstart` in `mj_fwdConstraint`, but if the flag is disabled then it will be ignored. So I think the problem you're experiencing must be different.

It sounds like, though this is just a guess, that there are active constraints that you are not expecting, as this is the difference between `qacc_smooth` (no constraints) and `qacc` (with constraints).

Try disabling either [contacts](https://mujoco.readthedocs.io/en/latest/XMLreference.html#option-flag-contact), [limits](https://mujoco.readthedocs.io/en/latest/XMLreference.html#option-flag-limit) or [equality constraints](https://mujoco.readthedocs.io/en/latest/XMLreference.html#option-flag-equality), or just the entire [constraint solver](https://mujoco.readthedocs.io/en/latest/XMLreference.html#option-flag-constraint). And this will give you a hint to what is happening. Probably something like joint limits in radians that are interpreted as [degrees](https://mujoco.readthedocs.io/en/latest/XMLreference.html#compiler-angle), or some such mismatch.

Chuanfang-Neptune, 2023-06-15:
Thank you for your reply.

I see that `qacc` is always copied into 'qacc_warmstart' at line [488](https://github.com/deepmind/mujoco/blob/643400174763da17e0f226d3615b33d594ddb27b/src/engine/engine_forward.c#L488) of source code already. But before that, line [466](https://github.com/deepmind/mujoco/blob/643400174763da17e0f226d3615b33d594ddb27b/src/engine/engine_forward.c#L466) called function `warmstart(m, d)` to compute the new `qacc`. And from line [379](https://github.com/deepmind/mujoco/blob/643400174763da17e0f226d3615b33d594ddb27b/src/engine/engine_forward.c#L379) if the warmstart flag is disabled. `qacc_warmstart` should not be copied into qacc. Iistead `qacc_smooth` should be copied into qacc as in line [439](https://github.com/deepmind/mujoco/blob/643400174763da17e0f226d3615b33d594ddb27b/src/engine/engine_forward.c#L439).

The problem now is that for some reasons (maybe some constraints are wrong as you mentioned) `qacc_warmstart `doesn't yield close enough result to `qacc_smooth`. And I verified that `qacc_smooth` yields good value according to my manual calculation. So I do not want to use qacc_warmstart and let the solver "cold-start" to use `qacc_smooth`. I set that disable flag, which I expected to do the work but indeed not.

Here is a simplified example to reproduce the problem.

simple_warm_start_xml_string
```
<mujoco model="simple_warmstart_fly">
    <compiler inertiafromgeom="false" autolimits="true"/>
    <option timestep="0.001" gravity="0 0 -9.81" integrator="RK4">
        <flag warmstart = "disable"/>
    </option>
    <worldbody>
    <body name="Thorax" pos="       0 2.44929e-16        4" quat="1.0 -1.8354755412297943e-15 0.0 0.0">
      <light mode="trackcom" directional="false" pos="0 0 4.0" dir="0 0 -1"></light>
      <inertial pos="-0.431185 -0.000353423 -0.0613552" mass="0.000307167823005631" fullinertia="2.86391e-05 3.33587e-05 2.61127e-05 6.70632e-10 2.45623e-06 -3.76016e-10"></inertial>
      <body name="LFCoxa" pos="-0.0627668     0.27 -0.130773" quat="1.0 0.0 0.0 0.0">
        <inertial pos="0.00298103 -0.0315211 -0.253124" mass="5.741707610995013e-06" fullinertia="5.98342e-08 5.79746e-08 1.42077e-08 1.53551e-09 9.56886e-10 2.87286e-09"></inertial>
        <joint name="joint_LFCoxa_roll" type="hinge" pos="0 0 0" axis="       0        0        1" limited="true" range="0.32318438 0.32318439" ref="0"></joint>
        <joint name="joint_LFCoxa_yaw" type="hinge" pos="0 0 0" axis="       1        0        0" limited="true" range="-0.55791212 -0.55791211" ref="0"></joint>
        <joint name="joint_LFCoxa_pitch" type="hinge" pos="0 0 0" axis="       0        1        0" limited="true" range="0.253825960 0.253825961 " ref="0"></joint>
        <body name="LFFemur" pos="0.022831 -0.020991 -0.404196" quat="0.9975649688716339 -0.06974333573995524 0.0 0.0">
          <inertial pos="-0.0198577 -0.00674739 -0.359444" mass="7.992114412322805e-06" fullinertia="2.72679e-07 2.733e-07 1.57358e-08 2.25294e-10 -8.56227e-09 -1.66725e-08"></inertial>
          <site name="LFTibia_flex_93434-P1" pos="-0.036 0.09 -0.233"></site>
          <site name="LFTibia_flex_93434-P2" pos="0.001 0.103 -0.58"></site>
          <site name="LFTibia_extensor_93932-P1" pos="0.035 0.084 -0.212"></site>
          <site name="LFTibia_extensor_93932-P2" pos="0.056 0.077 -0.726"></site>
          <joint name="joint_LFFemur_pitch" type="hinge" pos="0 0 0" axis="       0        1        0" limited="true" range="-2.070059594 -2.070059593" ref="0"></joint>
          <joint name="joint_LFFemur_roll" type="hinge" pos="0 0 0" axis="       0        0        1" limited="true" range="0.295773329 0.295773330 " ref="0"></joint>
          <body name="LFTibia" pos="0.017791 0.081932 -0.683955" quat="1.0 0.0 0.0 0.0">
            <inertial pos="0.0113971 -0.00358758 -0.293024" mass="2.6268343212263377e-06" fullinertia="5.5039e-08 5.51919e-08 2.18285e-09 1.52678e-10 1.88903e-09 2.13508e-09"></inertial>
            <site name="LFTibia_flex_93434-P3" pos="-0.018 0.002 -0.041"></site>
            <site name="LFTibia_extensor_93932-P3" pos="0.025 -0.002 -0.024"></site>
            <joint name="joint_LFTibia_pitch" type="hinge" pos="0 0 0" axis="       0        1        0" limited="true" range="0.350943  2.68825" ref="0"></joint>
            <body name="LFTarsus1" pos="       0        0 -0.518382" quat="1.0 0.0 0.0 0.0">
              <inertial pos="-0.00459028 -0.00942873 -0.123415" mass="5.902334137063546e-07" fullinertia="2.66192e-09 2.67112e-09 2.13711e-10 -6.6962e-12 -8.49982e-11 2.47205e-11"></inertial>
              <body name="LFTarsus2" pos="       0        0 -0.225517" quat="1.0 0.0 0.0 0.0">
                <inertial pos="-0.0012889 -0.00532267 -0.0891431" mass="2.658976343874435e-07" fullinertia="5.10333e-10 5.11017e-10 6.69257e-11 -4.65293e-12 2.23192e-11 1.1418e-12"></inertial>
                <body name="LFTarsus3" pos="       0        0 -0.153669" quat="1.0 0.0 0.0 0.0">
                  <inertial pos="-0.00108021 -0.00345421 -0.0593525" mass="1.5048445915428457e-07" fullinertia="1.29438e-10 1.29733e-10 3.23759e-11 -1.63625e-12 7.38402e-12 -3.53697e-12"></inertial>
                  <body name="LFTarsus4" pos="       0        0 -0.0984974" quat="1.0 0.0 0.0 0.0">
                    <inertial pos="0.000869731 -0.00147298 -0.0499729" mass="1.1769954786975934e-07" fullinertia="7.57484e-11 7.61929e-11 2.28487e-11 -1.1218e-12 4.14266e-12 -1.87825e-12"></inertial>
[... cut ...]
The simple example loaded a simplifed version of my model where one joint is controlled by two muscles and where warmstart flag was disabled. I applied an actuation of [0.5,0.0001] to the two muscles and do `mj_step()`. 
As outputs `qacc`, `qacc_smooth` and `qacc_warmstart` were listed and it can be seen that although warmstart flag was disabled. `qacc_warmstart` was still copied into 'qacc' and played a role in joint dynamics.

Result:
```
qacc: [16962.32619857], qacc_smooth: [29357.9814531], qacc_warmstart: [16962.32619857]
```

yuvaltassa, 2023-06-15:
Forget about `qacc_warmstart`. Just disable it with the flag and ignore it. If you wish I can explain more about the flow of warmstart/coldstart logic, but trust me that `qacc_warmstart` is irrelevant for the diagnosis.

What you want to understand is which constraints are applying unexpected forces, which lead to the differences between `qacc_smooth` and `qacc`. I recommend doing this interactively in the simulate viewer, where forces can be visualised and constraint enabled/disabled interactively.

Chuanfang-Neptune, 2023-06-15:
Thanks again for your reply!

Could you check if I understand correctly, `qacc_warmstart`([16962.32619857]) just coincided with `qacc`([16962.32619857]) here in my result  and `qacc` was indeed not copied from `qacc_warmstart` as shown in line [379](https://github.com/deepmind/mujoco/blob/643400174763da17e0f226d3615b33d594ddb27b/src/engine/engine_forward.c#L379) but was determined by some possibly applied unexpected force and constraints. And what I should look for are those constraints.

yuvaltassa, 2023-06-15:
How qacc is calculated is too long  a story, you can find a summary [here](https://mujoco.readthedocs.io/en/latest/computation.html#constraint-solver).

The thing to understand is that two things are being calculated together: `qacc` and `qfrc_constraint` (which is a function of which depends on whether you use a primal or dual solver). At the end of the day the important lines are [here](https://github.com/deepmind/mujoco/blob/643400174763da17e0f226d3615b33d594ddb27b/src/engine/engine_forward.c#L582). The smooth forces and constraint forces are added together and `qacc` is overwritten with $\dot v = M^{-1} (f_s + f_c)$ (where `qacc_smooth` is $\dot v_s = M^{-1} f_s$). The reason for overwriting it like this is because the solver might not have converged, so we can't 100% trust that the solution the solver came up with is consistent in the Newton-mechanical sense. So the difference between `qacc_smooth` and `qacc` is just the constraint forces.

PS when linking to code, please use GH's permalink option, so people reading this discussion in the future, after files have changed, won't be led to the wrong line numbers.

Chuanfang-Neptune, 2023-06-15:
Great thanks for your answer and also your PS for linking the code correct way!

I understand now that `qacc` is calculated under the dual effect of `qfrc_smooth` and `qfrc_constraint`. One strange thing still is that when I inspected the joint property with `mj.MjData.joint()`, `qfrc_constraint` was actually shown to be zero. Does that mean I indeed didn't apply any unexpected constraint that generated `qfrc_constraint` which would add the extra part to `qfrc` and affect `qacc`? In such situation, is it still normal that `qacc_smooth` is different than `qacc`?

```
  name: 'joint_LFTibia_pitch'
  qLDiagInv: array([95.32631846])
  qLDiagSqrtInv: array([9.76351978])
  qacc: array([728.55982479])
  qacc_smooth: array([4.06800485])
  qacc_warmstart: array([728.55982479])
  qfrc_actuator: array([0.01815944])
  qfrc_applied: array([0.])
  qfrc_bias: array([-0.01939574])
  qfrc_constraint: array([0.])
  qfrc_inverse: array([0.])
  qfrc_passive: array([-0.00010684])
  qfrc_smooth: array([0.03744834])
  qpos: array([1.06862884])
  qvel: array([0.36364967])
```

yuvaltassa, 2023-06-15:
Hmmm, could you print `data.qfrc_constraint` to rule out a bug in the bindings?

Chuanfang-Neptune, 2023-06-15:
I confirm data.qfrc_constraint to yield the same zero.
And indeed `qfrc_constraint` is working and applying force on other joints where I constrained them to be fixed.

yuvaltassa, 2023-06-15:
This is disturbing. I need to take a closer look. Would you mind posting the full Python script wherein you see constraints doing things but `qfrc_constraint` is 0?

I'll likely only get to this in a few days.

yuvaltassa, 2023-07-02:
You're only looking at the last element of `qfrc_constraint`, the forces are in the other elements.

Chuanfang-Neptune, 2023-07-03:
Thanks for your response!
I checked and found that there is `qfrc_constraint` in other joints. I think I am having them because I locked all other joints than the joint I am actuating (by setting their upper and lower limits to a same value. ) Under such situation, can I just see all the other elements as a whole rigid body? What should be the correct way to lock the other joints I'm not actuating in MuJoCo to avoid having unexpected `qfrc_constraint`?

yuvaltassa, 2023-07-03:
The correct way is to remove the joints.

Even if you have some reason not to do that, locking a joint with tight joint limits is a bad idea. This is why we have [bilateral equality constraints](https://mujoco.readthedocs.io/en/latest/XMLreference.html#equality-joint). 

Yet another option is to give a joint very high armature, like 1e6. This will lock it in a very stable way.

closed 2023-07-03 by yuvaltassa