google-deepmind/mujoco #920: Exception:GLFW window does not exist but you tried to render

liuchen199501115, 2023-06-14:
**How to ask for help**

First, read our quick guide to
[asking good questions](https://github.com/deepmind/mujoco#asking-questions).
Below is a template for you to use:

Hi,

I'm a student and I'm trying to use MuJoCo for learning

I'm looking for some help with code

Here is a model which explains my question:
Traceback (most recent call last):
     File "scripts/debug_stepper.py",line 137,in <module>
       main()
     File "scipts/debug_stepper.py",line 133 in main
       run(env,policy)
    File "scripts/debug_stepper.py",line 93,in run
      env.render()
    File "home/sky/anaconda3/envs/mujoco2.3.5/envs/RobBIEcommon/mujoco_env.py",line 78,in render
      self.viewer.render()
    File "home/sky/anaconda3/envs/mujoco2.3.5/lib/python3.8/site-packages/mujoco_viewer.py",line 309,in render
     raise Exception(
    Exception:GLFW window does not exist but you tried to render.

kevinzakka, 2023-07-19:
This is hard to debug without more information regarding the code you ran and the OS / python version you are using.

saran-t, 2023-07-19:
Neither `mujoco_env.py` nor `mujoco_viewer.py` is code that comes from this repo (or `dm_control`)...

closed 2023-07-19 by saran-t