google-deepmind/mujoco #922: API provided to record data

lyuyiqi, 2023-06-14:
Hi, thank you for developing this tool. I am trying to run my robot and record some data from it. However, it was a pain to store data to a vector by using for loop and indexing the mjtNum data. Also, the function mj_printData stores all the data instead of what I need. Could you please give me some hint about this, like a built-in API for storing specific part of the data?

saran-t, 2023-06-14:
Can you be more specific about exactly what it is that you're trying to save?

lyuyiqi, 2023-06-14:
Thank you for your reply. I am trying to continuously save time, qpos, qvel, ctrl, sensordata, xpos and squat in mjData at each timestep. I failed to convert mjtNum to vectors and store them.

closed 2023-06-14 by yuvaltassa