google-deepmind/mujoco #923: Customized button in the Agent section

lyuyiqi, 2023-06-14:
Hi, I am a student working to collect some data of my robot on mujoco. Now I wish to add a button to start and stop my data collection in the Agent section of the UI. I tried to add some status flags in the header, and then in the source file, I defined the default values of them and add two lines in the UI defining function. However, when I tried to run the simulator it just got frozen and could even load the model. Could you give me some instruction to fix this?

Here are what I added:
-----------------------------------------------------------------------------------
agent.h:
  // status flags, logically should be bool, but mjUI needs int pointers
  int plan_enabled;
  int action_enabled;
  int visualize_enabled;
  int allocate_enabled;
  int plot_enabled;
  int gui_task_id = 0;
  // int record_enabled;
  // int save_enabled;

-----------------------------------------------------------------------------------
agent.cc:

  // status
  plan_enabled = false;
  action_enabled = true;
  visualize_enabled = false;
  allocate_enabled = true;
  plot_enabled = true;
  // record_enabled = false;
  // save_enabled = false;

  // ----- agent ----- //
  mjuiDef defAgent[] = {
      {mjITEM_SECTION, "Agent", 1, nullptr, "AP"},
      {mjITEM_BUTTON, "Reset", 2, nullptr, " #459"},
      {mjITEM_SELECT, "Planner", 2, &planner_, ""},
      {mjITEM_CHECKINT, "Plan", 2, &plan_enabled, ""},
      {mjITEM_CHECKINT, "Action", 2, &action_enabled, ""},
      {mjITEM_CHECKINT, "Plots", 2, &plot_enabled, ""},
      {mjITEM_CHECKINT, "Traces", 2, &visualize_enabled, ""},
      // {mjITEM_CHECKINT, "Record", 2, &record_enabled, ""},
      // {mjITEM_CHECKINT, "Save", 2, &save_enabled, ""},
      {mjITEM_SEPARATOR, "Agent Settings", 1},
      {mjITEM_SLIDERNUM, "Horizon", 2, &horizon_, "0 1"},
      {mjITEM_SLIDERNUM, "Timestep", 2, &timestep_, "0 1"},
      {mjITEM_SELECT, "Integrator", 2, &integrator_, "Euler\nRK4\nImplicit\nFastImplicit"},
      {mjITEM_SEPARATOR, "Planner Settings", 1},
      {mjITEM_END}};
-----------------------------------------------------------------------------------

yuvaltassa, 2023-06-14:
Could you please post this question (with better formatting please) in the correct GitHub repository? (`mujoco_mpc`)

closed 2023-06-14 by yuvaltassa

lyuyiqi, 2023-06-14:
Sorry for my mistake.