google-deepmind/mujoco #924: Deprecated Makefile for MuJoCo-2.3.5 simulate

gywhitel, 2023-06-15:
Version: MuJoCo-2.3.5 release for linux on x86_64 platforms

The current makefile for MuJoCo-2.3.5 release  (mujoco/simulate/Makefile) contains deprecated `uitools.c`, which I guess is changed to `platform_ui_adapter.cc`. Besides, a few source code files is not compiled in the makefile. Users cannot build simulate using current makefile.

The old makefile 
```make
...
all:
	$(CXX) $(COMMON) -std=c++17 -c simulate.cc
	$(CC)  $(COMMON) -std=c11   -c uitools.c
	$(CXX) $(COMMON) -std=c++17 main.cc simulate.o uitools.o -lmujoco -lglfw -o ../bin/simulate
	rm uitools.o simulate.o
```
should change to the makefile below
```make
...
all:
	$(CXX) $(COMMON) -std=c++17 -c simulate.cc
	$(CC)  $(COMMON) -std=c++17 -c platform_ui_adapter.cc
	$(CC)  $(COMMON) -std=c++17 -c glfw_adapter.cc
	$(CC)  $(COMMON) -std=c++17 -c glfw_dispatch.cc
	$(CXX) $(COMMON) -std=c++17 main.cc simulate.o platform_ui_adapter.o glfw_adapter.o glfw_dispatch.o -lmujoco -lglfw -o ../bin/simulate2
	rm platform_ui_adapter.o simulate.o glfw_adapter.o glfw_dispatch.o
```

yuvaltassa, 2023-06-15:
As you know, we only officially support cmake. The make files are indeed legacy and left purely for user convenience, but are not supported by us.

Could you please send your changes as a PR?

Thanks

gywhitel, 2023-06-15:
Sure, I submitted PR #925, which fixed the problem.

yuvaltassa, 2023-06-15:
Thanks, this requires the CLA, see PR.

gywhitel, 2023-06-16:
Sorry, I used an unauthorized email in last PR. Now I use a correct email in PR #929.

gywhitel, 2023-06-16:
Sorry, I used an unanthorized email in last PR. Now I use the correct one in PR #929.

closed 2023-06-16 by gywhitel