google-deepmind/mujoco #926: add -keep_inertials args to the mujoco.mj_saveLastXML?

HuaweiWang, 2023-06-15:
Hi MuJoCo Team, 

We are creating musculoskeletal (MSK) models in MuJoCo and using `mujoco-bindings`, one thing we noticed is that the `mujoco.mj_saveLastXML` function does not save inertia properties when there is no geometry (but with joints) attached to the body. 

In MSK modelling, we needs to create quite some virtual bodies (as conditional sites). The issue we encountered can be reproduced in this simple example, in which load/save/reload does not work due to the inertia properties are not saved. 

```xml
<mujoco model="Virtual">

  <compiler inertiafromgeom="auto" angle="radian" balanceinertia="true" boundmass="0.001" boundinertia="0.001" inertiagrouprange="0 1">
    <lengthrange inttotal="50" useexisting="true"/>
  </compiler>

  <size njmax="1000" nconmax="400" nuser_jnt="1" nkey="1"/>

  <worldbody>
  
        <body name="virtual" pos="0 0 -0.25" euler="0 0 0">
          <!--<geom name="virtual_geom" type="box" size = "0.03 0.03 0.03" rgba="1 0 0 1"/>-->
          <joint name="virtual_x" pos="0 0 0" axis="0 0 1" type="slide" limited = "true" range="0 0.05"/>
        </body>
        
  </worldbody>
  
</mujoco>
```

A solution could be to add the -keep_inertials args, or automatically save the inertia properties when no geometry is included (in this case, the inertia properties need to be saved to be reload correctly)

An alternative is to have the common bounds setup saved in the headline `boundmass="0.001" boundinertia="0.001" inertiagrouprange="0 1"`. With this information, the model can be loaded even when there is no geometry and inertia property attached to the virtual body.

@aikkala

yuvaltassa, 2023-06-16:
I'm confused, the model you posted doesn't have an explicit <inertial> tag, so the inertia is entirely inferred from boundmass and boundinertia, right?

Are you saying that these attributes are missing from <compiler> in the saved model?

Or are you saying that you want explicit <inertial> tags to be saved even though none were present in the original model?

yuvaltassa, 2023-08-29:
Completed by https://github.com/deepmind/mujoco/commit/55173e58f9ff6fcb01ee336d7b25c1be0df9dc52

closed 2023-08-29 by yuvaltassa