google-deepmind/mujoco #928: Coupling Motors via Fixed Tendons - Search for Best Practices

psclklnk, 2023-06-15:
### Discussed in https://github.com/deepmind/mujoco/discussions/927

<div type='discussions-op-text'>

<sup>Originally posted by **psclklnk** June 15, 2023</sup>
Hi everyone,

I am currently aiming to create a fairly accurate model of the Barrett WAM 4DoF Robotic Arm. Doing some first Sys-Id experiments, I realized that the coupling of joints 2 and 3 (e.g. described on Page 68 of the [WAM Documentation](https://web.barrett.com/support/WAM_Documentation/WAM_UserManual_AH-00.pdf)) makes it hard to accurately capture their dynamics without explicitly modeling this motor coupling.

My idea for coupling the motors was to add two additional joints (representing the motors) that are coupled with the actual joints via a fixed tendon. In XML, this corresponds to 

```xml
<body name="wam/links/base" pos="0 0 1.2" euler="0 0 0">
            ...

            <body name="wam/rotor1" pos="0.22 0 0.173" euler="0 0 0">
                <joint name="wam/joints/motor_pitch" pos="0 0 0" axis="0 0 1" armature="0.053249" damping="0.1"
                       frictionloss="0.1" limited="false"/>
                <inertial pos="0 0 0" mass="1e-3" diaginertia="1e-7 1e-7 1e-7" quat="1 0 0 0"/>
                <geom class="viz" type="cylinder" size="0.03 0.01" rgba="1 0 0 1"/>
            </body>

            <body name="wam/rotor2" pos="0.22 0.28 0.173" euler="0 0 0">
                <joint name="wam/joints/motor_roll" pos="0 0 0" axis="0 0 1" armature="0.0564972" damping="0.1"
                       frictionloss="0.1" limited="false"/>
                <inertial pos="0 0 0" mass="1e-3" diaginertia="1e-7 1e-7 1e-7" quat="1 0 0 0"/>
                <geom class="viz" type="cylinder" size="0.03 0.01" rgba="1 0 0 1"/>
            </body>

            <body name="wam/links/shoulder_yaw" pos="0.22 0.14 0.346" euler="0 0 0">
                ....
               <body name="wam/links/shoulder_pitch" pos="0. 0. 0." euler="-1.570796 0 0">
                    <joint name="wam/joints/shoulder_pitch" pos="0 0 0" axis="0 0 1" range="-2 2" damping="0.1"
                           frictionloss="0" armature="0"/>
                    ...

    <tendon>
        <fixed name="pitch_coupling" limited="true" range="0. 0.001">
            <joint joint="wam/joints/motor_pitch" coef="0.01768812"/>
            <joint joint="wam/joints/motor_roll" coef="-0.01768812"/>
            <joint joint="wam/joints/shoulder_pitch" coef="1"/>
        </fixed>
        ...
```

While this works nicely in simulation (Sys-Id is to be done with this model), I was wondering whether there is a more elegant solution that does not require to specify additional rotating bodies. What I don't like about this solution is the need to specify additional inertias and masses. Can this be avoided by representing the actuated joint differently?</div>

yuvaltassa, 2023-06-21:
I don't understand why you need the extra joint. WAM has very little backlash.
Why aren't you just [coupling the joints directly](https://mujoco.readthedocs.io/en/latest/XMLreference.html#equality-joint)?

psclklnk, 2023-06-21:
Good point with the equality constraints. I remember looking at it but then concluding that I cannot easily (or at all) model the desired effect using the equality constraints. 

If I understood the documentation correctly, the equality element only allows to express relations between two joints (albeit non-linear ones). However,  the coupling of the BarrettWAM motors is expressed by two linear equations, each equation featuring three variables (the two motor positions and one of the two joint positions).

This is why I rather went for modeling the linear equations directly via the fixed tendons instead of trying to find a formulation via the equality element (if there even exists an equivalent formulation).

Does this reasoning make sense?

yuvaltassa, 2023-06-21:
Yes that makes sense. Perhaps some variables can be eliminated? Depends on the system. Can you make a minimal (planar) model with the two (I guess four now?) relevant joints and we can think about it together?

Anyway even in your formulation, unless you want backlash, I suspect you probably want a bilateral equality constraint on your tendon, rather than two unilateral inequality constraints that you'd get with the length range you have now.

PS remember with the WAM you'll want to use precisely computed joint armature to get the correct dynamics. The motor rotation is highly backdrivable and reflected inertia plays an important role.

psclklnk, 2023-06-23:
> PS remember with the WAM you'll want to use precisely computed joint armature to get the correct dynamics. The motor rotation is highly backdrivable and reflected inertia plays an important role.

You're right! On a second thought, modeling a rotating body that is attached to the motor may actually turn out to be important. I now included the rotor inertias at the bottom of the [WAM Inertia Document](https://web.barrett.com/support/WAM_Documentation/WAM_InertialSpecifications_AC-02.pdf) by setting the diagonal elements of the rotating bodies to the specified values, resulting in [this model](https://github.com/deepmind/mujoco/files/11846570/wam4_with_tendons.zip) (the model comes with meshes and everything so I hope you load it without problems).

I also checked whether modeling coupled motors can improve the realism of the simulation. For that, I tuned the parameters of my friction model (basically a Stribeck model) on trajectories that have been generated using a PD+Gravity Compensation controller. I then compared the model with motor coupling and without motor coupling on an unseen trajectory. The error in the position is negligible. However, when looking at the velocities, we see a benefit of the motor coupling:

[image]
[image]

Particularly in q_1 between second 0 and 3, the coupled model captures the behavior much better. Looking at torques, I would also say that the coupled model does a better job during movement.

[image]
[image]

However, it cannot compensate for some "steady-state errors" at the end of the trajectory and even increases this error in q_1. Maybe I can get rid of these errors by allowing to slightly deviate from the official inertia specifications.

yuvaltassa, 2023-08-29:
Sorry, misclosed this.

yuvaltassa, 2024-01-08:
Is there anything still pending here?

I'm closing for now but feel free to keep discussing.

commit 55173e58 (2023-08-29): Save `<compiler {boundmass/boundinertia}>` if nonzero.

Fixes #928

PiperOrigin-RevId: 560967976
Change-Id: Idc8c043278362d8925ce41061c65f3e678ad50ce
closed 2024-01-08 by yuvaltassa