google-deepmind/mujoco #940: How to increase stiffness in slide joints

jwcho0609, 2023-06-20:
Hello, I am using MuJoCo's python binding to create a robot that can pick up and place items from different heights. The robot consists of an arm that is attached to a vertical sliding joint, with all joints controlled through a custom implemented PID controller. 

Currently, I am able to grasp the object, but when I twist the wrist joint to lift the object, it instead seems to apply a stronger torque on the vertical slide joint, which causes the end effector to stay in the same position and makes the robot translate about the sliding joint instead. 

My question is, how can I stiffen sliding joints to prevent it from being affected by external forces? I have tried increasing armature, frictionloss, and damping for the slide joint in the xml file, but have had no success. Thank you. 

This is not for a personal project, so I am unable to share detailed visualization or xml files. I apologize in advance.

yuvaltassa, 2023-06-21:
Perhaps you can't share the actual model, but can you make a MWE? 

It is very difficult to diagnose your problem as is.

Balint-H, 2023-06-23:
Hello. just to check, have you tried increasing the P and D gains of your controller for the slide joint?

jwcho0609, 2023-06-28:
I apologize for the late response. I have tried diagnosing the issue and have found that the inertial components of the robot joints were actually incorrectly defined. I will try to fix this first and see if that improves the robot behavior. 

I'll close the issue for the time being, but if I am unable to resolve the problem, I will reopen the issue.

closed 2023-06-28 by jwcho0609