google-deepmind/mujoco #941: Unstable / Oscillating Contact Forces

llach, 2023-06-20:
Hi,
I am currently developing a simple MuJoCo object grasping scenario: a 2-DoF parallel-jaw gripper with individually controlled fingers is grasping a primitive object. I'm interested in the contact forces measured in the fingers during the grasp. For that, I've introduced bounding-box geoms around the fingers to calculate the contact forces as such:

```python
def total_contact_force(model, data, g1, g2):
    ft = np.zeros((6,))
    ncontacts = 0

    for i, c in enumerate(data.contact):
        name1 = data.geom(c.geom1).name
        name2 = data.geom(c.geom2).name

        if (g1==name1 and g2==name2) or (g1==name2 and g2==name1):
            c_ft = np.zeros((6,))
            mujoco.mj_contactForce(model, data, i, c_ft)
            ft += c_ft
            ncontacts += 1
    return ft[:3], ft[3:], ncontacts

left_force = np.sum(np.abs(total_contact_force(self.model, self.data, "object", "left_finger_bb")[0])
```
For the part of the model defining `"left_finger_bb"`, see below.

**What's the problem?**
The force measurements seem quite noisy / oscillate a lot. I've done some grasps where the fingers are simply commanded to reach a fixed goal position, penetrating the object, and plotted the contact forces. I've repeated the grasp with different MuJoCo settings that were named in affecting the stability of forces in some issues here.

[image]

Although enabling `multiccd` and using the `RK4` solver drastically decreased the measurement noise, the contact force measured in one finger is still quite noisy. 
I've tried setting `noslip_iterations` to various values > 0, but the results were never consistently better, sometimes they were even worse.

**Model Info**
These are the MuJoCo options / settings of my MJCF + the actuator parameters (taken from the Franka Panda gripper from the Menagerie):
```xml
<compiler angle="radian" autolimits="true" />
<size njmax="500" nconmax="100"/>

<option integrator="RK4" noslip_iterations="0" >
	<flag multiccd="enable"/>
</option>

<default>
	<general dyntype="none" biastype="affine" forcerange="-100 100" ctrlrange="0 0.045" gainprm="100 0 0" biasprm="0 -100 -10" />
</default>
```

Here's the left finger model:
```xml
<body name="gripper_left_finger_link" pos="0 0 0.076575" quat="0 -0.707107 0.707107 0">

  <inertial pos="-0.00447 0.010766 -0.121013" quat="0.944415 0.0576483 0.0941322 -0.309671" mass="0.10998" diaginertia="0.000161667 0.000161667 0.000161667"/>
  <joint name="gripper_left_finger_joint" pos="0 0 -0.076575" axis="-1 0 0" type="slide" range="0 0.045" />
  
  <geom quat="0 0 0 1" contype="0" conaffinity="0" type="mesh" rgba="0 0 0 1" mesh="gripper_finger_link_left"/>
  <geom name="left_finger_bb" pos="-0.004 0 -0.159" type="box" rgba="1 0 0 0" size="0.004 0.025 0.06" mass="0.01" solimp="0 0.95 0.01 0.2 2" />
</body>
```

... the right finger model ...
```xml
<body name="gripper_right_finger_link" pos="0 0 0.076575" quat="0 -0.707107 0.707107 0">

  <inertial pos="-0.00447 0.010766 -0.121013" quat="0.944415 0.0576483 0.0941322 -0.309671" mass="0.10998" diaginertia="0.000161667 0.000161667 0.000161667"/>
  <joint name="gripper_right_finger_joint" pos="0 0 -0.076575" axis="1 0 0" type="slide" range="0 0.045" />
  
  <geom quat="0 0 0 1" contype="0" conaffinity="0" type="mesh" rgba="0 0 0 1" mesh="gripper_finger_link_right"/>
  <geom name="right_finger_bb" pos="0.004 0 -0.159" type="box" rgba="1 0 0 0" size="0.004 0.025 0.06" mass="0.01" solimp="0 0.95 0.01 0.2 2"/>
</body>
```

... and the object:
```xml
<body name="object" pos="0 0 0.65" >
	<freejoint name="object_joint" />
	<geom name="object" type="cylinder" size="0.02 0.05" mass="0.5" priority="1" solimp="0 0.95 0.01 0.2 2" />  
</body> 
```

Note, that the object's `solimp` parameters are modified to simulate a softer object (as opposed to the very "stiff" default values). The contact force behavior is practically the same when using the default values however.

My open questions are:
* How can I reduce the remaining noise in the force measurements? Is it even possible to do so?
* Why is it that one finger's measurements always have more noise than the other's?
* Any other suggestions for improvements for the model and/or force calculation?

All help is welcome! Thanks!

yuvaltassa, 2023-06-21:
Not sure RK4 is what you want here. Try implicitfast.

`biasprm[2]` might want to be larger (in absolute terms).

Also, depending on what kind of motor you are modeling, your finger joints might want a dash of armature inertia.

llach, 2023-06-24:
Thanks for the quick response!
I tried your suggestions, but was unsuccessful in finding a configuration that produced consistently acceptable force measurements. I also tried varying other joint parameters besides `armature` (`damping` and `stiffness`), but had no luck either.

I have figured it out though: I started out with a known-good gripper (Franka Emika's Panda hand from the Menagerie), added force readouts and compared them to what my model produced. The ones from the Panda hand were super stable and practically noise-free!

The problem with my model (I think) was that the collision geoms used for the contact were quite long and the object was located at their tips, leading to oscillations in the fingers. I don't know why that happens though, see below for a pointer to a code example of that.

In any case, the forces look like that now:

[image]

I uploaded the [working example here.](https://github.com/llach/franka_force/tree/master) On [another branch in that repo is the broken version](https://github.com/llach/franka_force/tree/oscillations) which is similar to what I was using before opening up this issue. Hope it helps somebody.
I'll continue with using the Panda model with slight modifications to fit my real-life setup.

closed 2023-06-24 by llach