google-deepmind/mujoco #945: UI. A way to zero an actuator in simulate

Melanol, 2023-06-22:
Not very important. Also accidentally wrote this FR for dm_control. Reposting here, so as it is not forgotten.

There is a "Clear all" button in simulate to zero all actuators, but I sometimes find it frustrating that there is no way to zero a particular actuator, and I have to try to click as close to 0 as possible, and it is not helpful, as even the smallest value results in movement. This would be useful for manual tests.

[image]

yuvaltassa, 2024-01-08:
I tried a while back to make double-clicking do this, but it was too hard, sorry.
Maybe you can figure it out? 🙂

closed 2024-01-08 by yuvaltassa

Melanol, 2024-01-08:
How about right click? My C/C++ skills aren't great.

yuvaltassa, 2024-01-08:
Hmmm okay, I'll look at right click.