google-deepmind/mujoco #951: unknown default class Element 'general'

lyuyiqi, 2023-06-26:
Hi,

I am a student trying to train a RL agent using mujoco python package on Unitree GO1 quadruped robot downloaded in Menagerie. I changed the original position type actuator to general type, which worked well when I put it in the mujoco_mpc platform. However, with python environment I got an error saying unknown default class Element 'general', which I couldn't sovle for a long time. Could you please give me some guildance?

---
The actuators defined in the original XML file:
```XML  
 <actuator>
    <position class="abduction" name="FR_hip" joint="FR_hip_joint"/>
    <position class="hip" name="FR_thigh" joint="FR_thigh_joint"/>
    <position class="knee" name="FR_calf" joint="FR_calf_joint"/>
    <position class="abduction" name="FL_hip" joint="FL_hip_joint"/>
    <position class="hip" name="FL_thigh" joint="FL_thigh_joint"/>
    <position class="knee" name="FL_calf" joint="FL_calf_joint"/>
    <position class="abduction" name="RR_hip" joint="RR_hip_joint"/>
    <position class="hip" name="RR_thigh" joint="RR_thigh_joint"/>
    <position class="knee" name="RR_calf" joint="RR_calf_joint"/>
    <position class="abduction" name="RL_hip" joint="RL_hip_joint"/>
    <position class="hip" name="RL_thigh" joint="RL_thigh_joint"/>
    <position class="knee" name="RL_calf" joint="RL_calf_joint"/>
  </actuator>
```
The actuators I use:
```XML
  <actuator>
    <general class="torque" name="FR_hip" joint="FR_hip_joint"/>
    <general class="torque" name="FR_thigh" joint="FR_thigh_joint"/>
    <general class="torque" name="FR_calf" joint="FR_calf_joint"/>
    <general class="torque" name="FL_hip" joint="FL_hip_joint"/>
    <general class="torque" name="FL_thigh" joint="FL_thigh_joint"/>
    <general class="torque" name="FL_calf" joint="FL_calf_joint"/>
    <general class="torque" name="RR_hip" joint="RR_hip_joint"/>
    <general class="torque" name="RR_thigh" joint="RR_thigh_joint"/>
    <general class="torque" name="RR_calf" joint="RR_calf_joint"/>
    <general class="torque" name="RL_hip" joint="RL_hip_joint"/>
    <general class="torque" name="RL_thigh" joint="RL_thigh_joint"/>
    <general class="torque" name="RL_calf" joint="RL_calf_joint"/>
  </actuator>
```
The error I got:
```
Traceback (most recent call last):
  File "train.py", line 155, in <module>
    trainer = EncoderTrainer(args)
  File "train.py", line 74, in __init__
    self.env = QuadrupedEnv(args.max_step)
  File "/home/yiqi/Workspace/Python/quadruped-offline-main/mujoco_legged_gym/env.py", line 13, in __init__
    self.model = mujoco.MjModel.from_xml_path(asset_path)
ValueError: XML Error: unknown default class
Element 'general', line 0
```

kevinzakka, 2023-06-27:
Make sure the `torque` class is defined in the same XML (I think it's in the task XML IIRC).

closed 2023-06-27 by kevinzakka

yuvaltassa, 2023-06-27:
Read [this](https://mujoco.readthedocs.io/en/latest/modeling.html#actuator-shortcuts).
All actuator types are just macros for "general"