google-deepmind/mujoco #952: Derivative of

biegunk, 2023-06-27:
**How to ask for help**

First, read our quick guide to
[asking good questions](https://github.com/deepmind/mujoco#asking-questions).
Below is a template for you to use:

Hi,

I'm a (student / professor / engineer) and I'm trying to use MuJoCo for _____.

I'm looking for some help with ____.

Here is a model which explains my question:

<details>
  <summary>minimal XML</summary>

```XML
<mujoco>
  <worldbody>
    <light pos=".4 -.4 .3" dir="-2 2 -1.5" diffuse=".6 .6 .6"/>
    <light pos="-.2 -.4 .3" dir="1 2 -1.5" diffuse=".6 .6 .6"/>
    <geom type="plane" size=".5 .5 .01"/>
    <body name="ball" pos="0 0 0.05">
      <freejoint/>
      <geom size=".03"/>
    </body>
    <body name="hammer" pos="-.05 0 0.18">
      <joint axis="0 1 0"/>
      <geom type="capsule" size=".01" fromto="0 0 0 -.15 0 0"/>
      <geom type="box" size=".02 .02 .03" pos="-.15 0 0"/>
    </body>
  </worldbody>
</mujoco>
```

</details>

Here is a screenshot / video, illustrating my question:

yuvaltassa, 2023-06-27:
I the observation code is in Python you'll have to do the finite-differencing in Python.

closed 2023-06-27 by yuvaltassa

biegunk, 2023-06-27:
Hi, the observation is in python but is a function of elements of the MJData struct, in particular `geom_xpos`. Given the jacobian d_geom_xpos/da it would be trivial to get the jacobian of the observation. Is there a simple way of computing this Jacobian without manual finite difference implementation?

yuvaltassa, 2023-06-29:
https://mujoco.readthedocs.io/en/latest/APIreference/APIfunctions.html#mj-jacgeom