google-deepmind/mujoco #954: Different time step size makes my simulation dramatically different and even blow up

Kobaya627, 2023-06-28:
I'm a student and I'm trying to use MuJoCo doing a torque-based [impedance control](https://ieeexplore.ieee.org/document/8328305) simulation.

After finish programming my cartesian impedance controller with C++ , I find my simulation goes weird with different **time step size** which is modified in MJCF as below.

`  <option integrator="RK4" timestep="0.0005"/>`

Below i show the simulation result setting `timestep` equal to `0.0004 `, `0.0005 `, `0.005 ` respectively .

**timestep = "0.0004"**

[image]

**timestep = "0.0005"**

[image]

**timestep = "0.005"**

[image]

Some warning as below also occur in the case that `timestep = "0.005"`

```
WARNING: Nan, Inf or huge value in CTRL at ACTUATOR 0. The simulation is unstable. Time = 2.9200.

WARNING: Nan, Inf or huge value in QACC at DOF 0. The simulation is unstable. Time = 2.9200.

WARNING: Nan, Inf or huge value in CTRL at ACTUATOR 0. The simulation is unstable. Time = 0.0175.

WARNING: Nan, Inf or huge value in QPOS at DOF 2. The simulation is unstable. Time = 0.0200.

WARNING: Nan, Inf or huge value in CTRL at ACTUATOR 0. The simulation is unstable. Time = 0.0175.

WARNING: Nan, Inf or huge value in CTRL at ACTUATOR 0. The simulation is unstable. Time = 0.0175.
```

> Tips : The simulation is unstable only when i drag the manipulator , it is stable if i do not perturb it .

The model i am using is kuka iiwa14 provided by [mujoco_menagerie](https://github.com/deepmind/mujoco_menagerie) and i made some changes to it.

<details>
  <summary>minimal XML</summary>

```XML
<mujoco model="iiwa14">
  <compiler angle="radian" meshdir="assets" autolimits="true"/>

  <option integrator="RK4" timestep="0.0003"/>

  <default>
    <default class="iiwa">
      <material specular="0.5" shininess="0.25"/>
      <joint axis="0 0 1" damping="0.2"/>
      <general gaintype="fixed" biastype="affine" gainprm="2000" biasprm="0 -2000 -200"/>
      <default class="joint1">
        <joint range="-2.96706 2.96706" damping="0.2"/>
        <general ctrlrange="-2.96706 2.96706"/>
        <default class="joint2">
          <joint range="-2.0944 2.0944" damping="0.2"/>
          <general ctrlrange="-2.0944 2.0944"/>
        </default>
      </default>
      <default class="joint3">
        <joint range="-3.05433 3.05433" damping="0.2"/>
        <general ctrlrange="-3.05433 3.05433"/>
      </default>
      <default class="visual">
        <geom type="mesh" contype="0" conaffinity="0" group="2" material="gray"/>
      </default>
      <default class="collision">
[... 160 lines cut ...]
	<position name="position_servo_7" joint="joint7" kp="100"/>
	<velocity name="velocity_servo_7" joint="joint7" kv="2"/>
  </actuator>
  
  <sensor>
	<force name="Force" site="attachment_site" />
	<torque name="Torque" site="attachment_site"/>
	<framepos name="ee_pos_sensor" objtype="site" objname="attachment_site" />
        <framequat name="ee_quat_sensor" objtype="site" objname="attachment_site" />
	<framelinvel name="ee_linvel_sensor" objtype="site" objname="attachment_site"/>
	<frameangvel name="ee_angvel_sensor" objtype="site" objname="attachment_site"/>
  </sensor>
    
  <!--keyframe>
    <key name="home" qpos="0 0.785398 0 -1.5708 0 0 0" ctrl="0 0.785398 0 -1.5708 0 0 0"/>
  </keyframe-->
</mujoco>
```

</details>

**Expected solution**

I am not sure if this situation is normal of if there is something wrong with my controller , i do know that large time step size can cause divergence but the result is still surprising to me .
I hope you can give me some insight into what is happening , thanks very much.

yuvaltassa, 2023-06-28:
This generally expected, but at larger timesteps. Can you try the implicit or implicitfast integrators?

Kobaya627, 2023-06-28:
> This generally expected, but at larger timesteps. Can you try the implicit or implicitfast integrators?

Thanks for your reply , i have tried implicit and implicitfast integrators before and i confirmed simulation results just now . 
They both go unstable when setting `timestep = "0.0004"` , the results are similar to the RK4 integrator setting `timestep = "0.005" which is showed above.

saran-t, 2023-06-28:
Can you share your controller code too?

Kobaya627, 2023-06-28:
> Can you share your controller code too?

Sure , my mjcb_control is showed below .

```
void mycontroller(const mjModel* m, mjData* d){
    Eigen::MatrixXd J(6,7),J_T(7,6),Lambda(6,6),J_dot(6,7),I6(6,6),I7(7,7),M(7,7),J_T_MP(6,7),JMinvJT(6,6),prjMat(7,7),J_MP(7,6);
    Eigen::Matrix3d rotMatd,deltaRot,rotMat;
    Eigen::VectorXd transVecd(3);
    Eigen::VectorXd dX(6),X(6),tau_c(7),tau_e(7),tau_n(7),tau_0(7),q_dot(7),q(7),X_dot(6),dq(7),C(7),qd_dot(7),P(7),F(7);
    Eigen::VectorXd k1(7),k2(7),k3(7),k4(7);
    I6 = Eigen::MatrixXd::Identity(6,6);
    I7 = Eigen::MatrixXd::Identity(7,7);
    Eigen::Quaterniond quat;
    Eigen::MatrixXd Kd(6,6),Bd(6,6),Md(6,6);
    int body = 8;
    
    // Avoidance of Inertia Shaping
    bool AOIS = true;

    Kd = Eigen::MatrixXd::Zero(6,6);               
    Md = Eigen::MatrixXd::Zero(6,6);               
    Bd = Eigen::MatrixXd::Zero(6,6);               
    Kd(0,0) = Kd(1,1) = Kd(2,2) = 100;
    Kd(3,3) = Kd(4,4) = Kd(5,5) = 10;
    Md(0,0) = Md(1,1) = Md(2,2) = 10; 
    Md(3,3) = Md(4,4) = Md(5,5) = 2;  

    if (AOIS){
        Bd(0,0) = Bd(1,1) = Bd(2,2) = 2 * sqrt(Kd(0,0));
        Bd(3,3) = Bd(4,4) = Bd(5,5) = 2 * sqrt(Kd(3,3));
    }
    else{
      
        Bd(0,0) = Bd(1,1) = Bd(2,2) = 2 * sqrt(Md(0,0)*Kd(0,0));
        Bd(3,3) = Bd(4,4) = Bd(5,5) = 2 * sqrt(Md(3,3)*Kd(3,3));
    }
    
    for (int i = 0;i<7;i++){
      q(i) = d->qpos[i];
      q_dot(i) = d->qvel[i];
    }

    // calculate current pose
    X(0) = d->sensordata[6];
    X(1) = d->sensordata[7];
    X(2) = d->sensordata[8];
    quat = Eigen::Quaterniond(d->sensordata[9],d->sensordata[10],d->sensordata[11],d->sensordata[12]);
    dX.head(3) = X.head(3)-Xd.head(3);
    dX.tail(3) = calculateOrientationError(quatd,quat);
    // std::cout << "dX = \t"<<dX.transpose()<<std::endl;

    // calculate jacobian
    double jacp[21]={0},jacr[21]={0},point[3]={X(0),X(1),X(2)};
    mj_jac(m,d,jacp,jacr,point,body);
    for (int i = 0;i<3;i++)
      for (int j = 0;j<7;j++){
        J(i,j) = jacp[7*i+j];
        J(i+3,j) = jacr[7*i+j];
      }
    J_T = J.transpose();
    J_T_MP = pseudoInverse(J_T);
    J_MP = pseudoInverse(J);

    // calculate(q)
    double fullM[49]={0};
    mj_fullM(m,fullM,d->qM);
    for (int i=0;i<7;i++)
      for (int j = 0;j<7;j++)
        M(i,j) = fullM[7*i+j];
    
    // calculateLambda(6,6)
      JMinvJT = J*M.inverse()*J_T;
      Lambda = JMinvJT.inverse();

    // 计算nle : F = C*q_dot + g
    for (int i=0;i<7;i++)
      F(i) = d->qfrc_bias[i];

    // calculate tau_e
      for (int i = 0; i < 7; i++)
        tau_e(i) = 0;

    // cartesian impedance controller
    if (AOIS)
      // tau_c = J_T*(-Kd*dX -Bd * J *q_dot )+F; 
      tau_c = J_T*(-Kd*dX -Bd * J *q_dot )+F; 
    else 
      tau_c = J_T*(Lambda*Md.inverse()-I6)*J_T_MP*tau_e - J_T*Lambda*Md.inverse()*Bd*J*q_dot - J_T*Lambda*Md.inverse()*Kd*dX+F;
      
    // std::cout <<"Bd = \n"<<Bd<<std::endl;
    // std::cout <<"J = \n"<<J<<std::endl;
    // std::cout <<"q_dot = \n"<<q_dot.transpose()<<std::endl;
    
    // ctrl
    for (int i = 0;i<7;i++)
      d->ctrl[3*i] = tau_c(i);
}
```

yuvaltassa, 2023-07-02:
This is difficult to diagnose. Can you add [actuator force sensors](https://mujoco.readthedocs.io/en/latest/XMLreference.html#sensor-actuatorfrc) and re-render your videos with those sensors visualised (F4 in simulate)? I expect that there is some bug in your actuation code that is causing huge values. Do you have control or force limits set for these actuators?

Kobaya627, 2023-07-02:
> This is difficult to diagnose. Can you add [actuator force sensors](https://mujoco.readthedocs.io/en/latest/XMLreference.html#sensor-actuatorfrc) and re-render your videos with those sensors [...]

Thanks for your reply ， i do not have control and force limits for actuators . I'll add torque sensors and check the torque curve .

yuvaltassa, 2024-01-08:
Closing due to lack of activity.

closed 2024-01-08 by yuvaltassa