google-deepmind/mujoco #955: Converting the freebody into a static body and vice versa

jwcho0609, 2023-06-28:
Hello, I am using MuJoCo's python binding to create a robot that can pick up and place items from different heights. 

I have found that if I spawn too many items (~1000 different objects) into the simulation environment, the simulation crashes with a stack overflow error. I am assuming the potential cause of this is due to too many contacts being detected and calculated. 

As a work around, I was thinking of initializing these objects as static, but when the object is picked up (or maybe if the robot end effector is near the object) the object becomes a free body so that it can be picked up. 

My question is, is there a way to convert a body into a freebody from a static body and vice versa on the fly? These attributes are predefined in the xml files, so I was wondering if there is a way to maybe "toggle" it during simulation in code.

Thank you.

saran-t, 2023-06-28:
Are all of your boxes overlapping at t=0? Usually an easy fix would be to just set box qpos to different places.

jwcho0609, 2023-06-29:
No, they are all generated at different positions with no overlap on each other.

yuvaltassa, 2023-06-29:
Try using the Conjugate Gradient solver.

kevinzakka, 2023-12-03:
@jwcho0609 Were you able to solve your problem?

jwcho0609, 2023-12-06:
No, not necessarily; we just ended up having to decrease the number of bodies being used in simulation. I am still wondering if there are ways of disabling and enabling the freejoint attribute of a geom.