google-deepmind/mujoco #956: Inconsistent Behavior of Weld Constraints when Updating Body's Position and Orientation, aswell as the Welds Relative Position and Orientation at Runtime

acodeodyssey, 2023-06-28:
Hi,

I'm a research assistant at Uni Bielefeld,

and we want to fixate a body in the world with a softness, and tried to do so by using a weld constraint between the body and the worldbody.
Our use case requires us to update the desired position and orientation of the body at runtime, i.e. the position and orientation of the weld constraint. A changed position works, a changed orientation, however, doesn't match the expected orientation of the body. In other words, when we try to modify the relative orientation using eq_data, it doesn't accurately reflect the orientation of the body. Is this behavior wanted and how could I set the desired orientation?

<details>
<summary>Minimal XML </summary>

```XML
<mujoco model="empty">
    <option collision="all" timestep="0.001" gravity="0 0 -9.81" cone="elliptic" />
    <compiler angle="radian" />

    <visual>
        <headlight ambient="0.4 0.4 0.4" diffuse="0.4 0.4 0.4" specular="0.0 0.0 0.0" active="1" />
    </visual>

    <asset>
        <texture builtin="checker" height="512" name="texplane" rgb1=".2 .3 .4" rgb2=".1 .15 .2" type="2d" width="512" />
        <material name="MatPlane" reflectance="0.5" shininess="0.01" specular="0.1" texrepeat="1 1" texture="texplane" texuniform="true" />
    </asset>

g    <worldbody>
        <light pos="0 0 1000" castshadow="false" />
        <body name="hogBox">
            <joint type="free"/>
            <geom type="box" size="0.25 0.25 0.25"/>
        </body>
        <geom name="ground_plane" type="plane" size="5 5 10" material="MatPlane" rgba="1 1 1 1"/>
    </worldbody>
    <equality>
        <weld body1="hogBox" body2="Box" active="false"/>
    </equality>
</mujoco>
```
</details>

yuvaltassa, 2023-06-29:
@quagla could you reply?

quagla, 2023-07-04:
Hi @acodeodyssey, I assume you have already seen this code:

https://github.com/deepmind/mujoco/blob/main/src/engine/engine_setconst.c#L205

which is where we update the `eq_data` specified in the XML and obtain the one that is actually used in the simulation?

acodeodyssey, 2023-07-04:
Hi @quagla ,

yes, i have seen it.

quagla, 2023-07-04:
Could you please share how are you updating `eq_data` during the simulation?

acodeodyssey, 2023-07-05:
@quagla I'm updating it here, https://github.com/acodeodyssey/mujoco_hog_plugin/blob/noetic-devel/src/mujoco_ros_hog_plugin.cpp#L304

quagla, 2023-07-06:
Why is it `i * mjNEQDATA + j + 3` instead of `i * mjNEQDATA + j + 6`?

acodeodyssey, 2023-07-07:
@quagla It's j (>=3)+3, so it adds up to 6 in total. I used i as an index, in i * mjNEQDATA + j + 3 .

quagla, 2023-07-12:
Have you checked that the weld constraint is actually satisfied after the update? Maybe the solver is struggling with keeping up with the change in orientation. You can enable constraint rendering and check that the rendered spheres (corresponding to the pose of body1 and body2 as defined in `eq_data`) overlap.

acodeodyssey, 2023-07-14:
@quagla I checked the spheres, and they overlap after some time. I think you are right, about the solver not keeping up.

closed 2023-07-28 by quagla

DavidPL1, 2023-09-12:
@STAY-Melody, we also had difficulties finding out how `eq_data` is organized. The content of the constraint data -- which should correspond to the content of eq_data -- is actually described [here](https://mujoco.readthedocs.io/en/stable/computation.html?highlight=constraints#equality) in the documentation text for each constraint type, but it's not as clear that this is in fact the order in the data array. Maybe this should be mentioned explicitly.

As a more practical example with code, you can have a look at the xml reader code to find out which data is written in which order for each constraint. Here is the section relevant for weld constraints:

https://github.com/google-deepmind/mujoco/blob/2b192cebb6694f9af4b62f8c19a75f2d7daab402/src/xml/xml_native_reader.cc#L1570-L1577

In this case `pequality->data` would correspond to `eq_data[eq_id*mjNEQDATA]`
So retrieving the relpose of a weld constraint with id `eq_id` should be possible with the following code:

```c++
mjtNum eq_relpose[3];
mju_copy3(m->eq_data[eq_id*mjNEQDATA + 3])
``` 

(correct me if I got that wrong, @yuvaltassa & @quagla)

DavidPL1, 2023-09-12:
The function name and description indicate that mocap body weld constraints should be reset but this function just "resets" all weld constraints. Generally this does not seem like a sane thing to do.

- The code you referenced was last changed in a commit from Sep 14, 2022 which also updated the pyproject.toml requirement to mujoco 2.2.2. This might explain outdated code **(I just checked and it seems that mjNEQDATA did not change since 2.2.2, so no outdated API here)**.

It seems that this is a special case constructed by Gymnasium for an actuation model. This code sets the anchor of each weld constraint to (0, 0, 0), the xyz component of the relpose to (0, 0, 0) and the first orientation component of the relpose to 1 rad while the rest of the orientation and the torquescale remain untouched.
This suggests that they change the anchor and part of the relpose to actuate bodies welded to mocap bodies. But I'm not too familiar with Gymnasium mujoco environments. You should look into the Gymnasium documentation if this is explained anywhere or open an issue on their repo and ask for an explanaition.