google-deepmind/mujoco #959: Github actions to test rendering on multiOS

Vittorio-Caggiano, 2023-06-28:
Hi team,

I am trying to set up a github action for testing rendering capabilities (via `mujoco.render_test`) on different operating systems. Here the minimal example:

```
jobs:
  build:
    runs-on: ${{ matrix.os }}
    strategy:
      matrix:
        os:
          - ubuntu-latest
          - macos-latest
          - windows-latest
    steps:
      - name: Set up Python 3.9
        uses: actions/setup-python@v3
        with:
          python-version: "3.9"

      - name: Prepare Linux
        if: ${{ runner.os == 'Linux' }}
        run: |
          sudo apt-get install -y \
              libgl1-mesa-dev \
              libgl1-mesa-glx \
              libglew-dev \
              libosmesa6-dev \
              software-properties-common
          sudo apt-get install -y patchelf
          export MUJOCO_GL=EGL

      - shell: bash
        run: |
          pip install mujoco
          python -m mujoco.render_test
```

However, it seems there are issues to instantiate GLFW on all the OSs
Linux:
``` 
/opt/hostedtoolcache/Python/3.9.17/x64/lib/python3.9/site-packages/glfw/__init__.py:916: GLFWError: (65544) b'X11: The DISPLAY environment variable is missing'
```
Windows:
```
 C:\hostedtoolcache\windows\Python\3.9.13\x64\lib\site-packages\glfw\__init__.py:916: GLFWError: (65542) b'WGL: The driver does not appear to support OpenGL'
```
MacOS:
```    
self.free()
  File "/Users/runner/hostedtoolcache/Python/3.9.17/x64/lib/python3.9/site-packages/mujoco/cgl/__init__.py", line 60, in free
    if self._context:
AttributeError: 'GLContext' object has no attribute '_context'
```

saran-t, 2023-06-28:
It's basically impossible to test GPU-based rendering on these VMs unfortunately. There's a reason we disabled rendering tests.

saran-t, 2023-06-28:
Maybe I misunderstood: are you wanting to test CPU-based rendering? I think that's only easy to do on Linux (I've never tried to get Mesa working on Windows or macOS). In that case you should set `MUJOCO_GL=osmesa`.

Vittorio-Caggiano, 2023-06-29:
Hi @saran-t yes, sorry for being unclear. I am trying to check CPU-based rendering.
Unfortunately when I set `MUJOCO_GL=osmesa` on linux I still get the same error
```
/opt/hostedtoolcache/Python/3.9.17/x64/lib/python3.9/site-packages/glfw/__init__.py:916: GLFWError: (65544) b'X11: The DISPLAY environment variable is missing'
```

I think this is something to do specifically with gh-actions as if I run the same in a colab, it works.

DavidPL1, 2023-06-29:
The VMs probably don't seem to be running an X server. I'm not too familiar with GH Actions VMs, but in case you can configure Xvbf to be installed, you can create and use a simulated X like in the below script:

```bash
#!/bin/bash

set -e

# Use CREATE_DISPLAY env variable or default to 4 
CREATE_ID=${CREATE_DISPLAY:-4}

echo "CREATE_DISPLAY: ${CREATE_DISPLAY} | CREATE_ID: ${CREATE_ID}"

# Let Xvfb create the display
Xvfb :${CREATE_ID} &

echo "trying to create virtual display named ${CREATE_ID}"

display_ready=0
tries=0

# Wait for the display to be ready
while [ $display_ready = 1 ] || [ $tries -lt 10 ]
do
    if xdpyinfo -display :${CREATE_ID} >& /dev/null ; then break
    else echo "Virutal display not ready yet (try $tries)" && sleep 1 ; fi
    tries=`expr $tries + 1`
done

if xdpyinfo -display :${CREATE_ID} >& /dev/null ; then echo "Display exists"
else echo "Display invalid, too many tries" && exit -1 ; fi

# Export DISPLAY env variable to use it in following commands
export DISPLAY=:${CREATE_ID}
```

After the export you are good to go and glfw should not complain anymore.

Vittorio-Caggiano, 2023-06-30:
Thanks for the suggestion @DavidPL1 ... the simplest solution that worked for Linux is 

```
jobs:
  build:
    env:
      DISPLAY: :0

    defaults:    
      run:
        shell: bash -l {0}
  
    runs-on: ${{ matrix.os }}
    strategy:
      matrix:
        os: 
          - ubuntu-latest
#           - macos-latest
#           - windows-latest
    steps:
      - name: Checkout repository and submodules
        uses: actions/checkout@v3
    
      - name: Set up Python 3.9
        uses: actions/setup-python@v3
        with:
          python-version: "3.9"
         
      - name: Prepare Linux
        if: ${{ runner.os == 'Linux' }}
        run: |
          sudo apt-get install -y xvfb x11-xserver-utils
          sudo /usr/bin/Xvfb $DISPLAY -screen 0 1280x1024x24  & 

      - name: Install mujoco and test rendering
        run: |
          pip3 install mujoco
          python3 -m mujoco.render_test
```

can you perhaps point me in the right direction for macOS and windows too? I tried `xquartz` on MacOS and it didn't work

cc: @saran-t

saran-t, 2023-07-01:
Ideally you wouldn't try to depend on X11 at all. OSMesa on Linux will allow you to headlessly test software rendering. I don't believe that there's a standard/easy solution on macOS or Windows.

saran-t, 2023-07-01:
(Also, we _do_ run rendering tests internally, so it's not like the renderer is completely untested :) )

closed 2024-08-30 by kevinzakka