google-deepmind/mujoco #961: XML error regarding instances when importing XML flies created from loaded XML models containing plugin and composites. Is this a bug or just not implemented yet?

ce-gh, 2023-06-29:
Hi, 
First of all, thank you for the great work and opening up this promising piece of tech.
I'm a researcher and I am currently evaluating MuJoCo (ver. 2.3.6 on Win and Linux), when I came across the elasticity plugin.
As more context is likely irrelevant to my issue, as it can be replicated using the example models, I'll cut to the chase:

**Replicate**
Using Simulator, open one of the examples from the "model\plugin" folder and save the XML from the GUI.
Try to load it back into the Simulator results in:
```
Element 'plugin', line 32
XML Error: Schema violation: unrecognized attribute: 'instance'
```
The actual line number depends on the first occurrence of "instance" in the object (s) section of the file.

I managed to dig deeper into the problem and found that restructuring the attribute 'instance' into it's own element removes the XML error:
<details>

**Before** 

```XML
<mujoco model="Floppy">
  <compiler angle="radian" meshdir="asset/" texturedir="asset/" autolimits="true"/>
  <option timestep="2e-06" tolerance="1e-06"/>
  <size memory="500M"/>
  <visual>
    <map fogstart="1" fogend="15" zfar="40" haze="1" shadowscale="0.5"/>
    <rgba haze="0.15 0.25 0.35 1"/>
  </visual>
  <statistic meansize="0.05" extent="2" center="0 0 0"/>
  <extension>
    <plugin plugin="mujoco.elasticity.solid">
      <instance name="composite">
        <config key="nx" value="22"/>
        <config key="ny" value="6"/>
        <config key="nz" value="2"/>
        <config key="young" value="5e5"/>
        <config key="poisson" value="0.2"/>
        <config key="damping" value="0.008"/>
      </instance>
    </plugin>
  </extension>
  <asset>
    <texture type="skybox" builtin="gradient" rgb1="0.3 0.5 0.7" rgb2="0 0 0" width="512" height="3072"/>
    <texture type="2d" name="texplane" builtin="checker" mark="cross" rgb1="0.2 0.3 0.4" rgb2="0.1 0.15 0.2" markrgb="0.8 0.8 0.8" width="512" height="512"/>
    <material name="matplane" texture="texplane" texuniform="true" texrepeat="10 10" reflectance="0.3"/>
  </asset>
  <worldbody>
    <geom name="ground" size="0 0 1" type="plane" condim="1" material="matplane"/>
    <light pos="0 0 2" dir="0 0 -1" castshadow="false" diffuse="0.4 0.4 0.4" specular="0.1 0.1 0.1"/>
    <light pos="0 0 4" dir="0 0 -1" directional="true" diffuse="0.8 0.8 0.8" specular="0.2 0.2 0.2"/>
    <body name="B0_0_0" pos="-0.12 -0.06 0.03" instance="composite"> <!-- *** this is the offending line *** -->
      <joint pos="0 0 0" axis="1 0 0" type="slide" group="3"/>
      <joint pos="0 0 0" axis="0 1 0" type="slide" group="3"/>
      <joint pos="0 0 0" axis="0 0 1" type="slide" group="3"/>
      <geom size="0.005" group="4" priority="1" solref="0.01 1" solimp="0.95 0.99 0.0001 0.5 2" rgba="0.8 0.2 0.1 1"/>
    </body>
...
```

**After**
```XML
...
    <body name="B0_0_0" pos="-0.12 -0.06 0.03" > <!-- *** removed attribute instance here *** -->
      <plugin instance="composite"/>  <!-- *** added it here as an element *** -->
      <joint pos="0 0 0" axis="1 0 0" type="slide" group="3"/>
      <joint pos="0 0 0" axis="0 1 0" type="slide" group="3"/>
      <joint pos="0 0 0" axis="0 0 1" type="slide" group="3"/>
      <geom size="0.005" group="4" priority="1" solref="0.01 1" solimp="0.95 0.99 0.0001 0.5 2" rgba="0.8 0.2 0.1 1"/>
    </body>
...
```

</details>

However, it generates a new error:
```
Element 'body', line 98
Error: engine error: This body does not have the requested plugin instance
```

Before I waste my time on something that might be on the road map already, I stopped my efforts.
I would be grateful to know if this will be fixed (loading of generated XML from models with composites and plugins).

Greetings,
Chris

yuvaltassa, 2023-06-30:
Thanks @ce-gh, I'm AFK but this looks like a bug. The surprising thing is that we have a test for exactly this that runs on all the models in `/model/` (look at the bottom of `xml_native_writer_test.cc`).

Perhaps @quagla can diagnose?

ce-gh, 2023-06-30:
Thanks for the quick reply.

The "plugin" dir is excluded from testing in `xml_native_writer_test.cc`. This is why I assumed it to be WIP.

> TEST_F(XMLWriterTest, WriteReadCompare) {
  FullFloatPrecision increase_precision;
  // Loop over all xml files in data
  std::vector<std::string> paths = {GetTestDataFilePath("."),
                                    GetModelPath(".")};
  std::string ext(".xml");
  for (auto const& path : paths) {
    for (auto &p : std::filesystem::recursive_directory_iterator(path)) {
      if (p.path().extension() == ext) {
        std::string xml = p.path().string();

        // if file is meant to fail, skip it
        if (absl::StrContains(p.path().string(), "malformed_") ||
            absl::StrContains(p.path().string(), "plugin")) {
          continue;
        }

closed 2023-07-04 by copybara-service[bot]