google-deepmind/mujoco #962: Custom decorative objects

zxwang96, 2023-07-01:
Hi,

I'm using mujoco for a project that moves a cuboid to the target area. I use a thin `cylinder` object as the receptacle/target area indicator. But I found no matter how thin the cylinder is, the cuboid will be stuck near the edge of the cylinder. My question is: is there any way to add custom decorative objects, which are not considered by the physics engine of mujoco, to my MJCF? If not, could you suggest possible solutions to my problem?

yuvaltassa, 2023-07-01:
Yes, of course. Run the tutorial colab, this is explained in the last cell.

closed 2023-07-01 by yuvaltassa

zxwang96, 2023-07-01:
Thanks. And there is a more direct solution: set `contype="2" conaffinity="2"` in MJCF to avoid collision

saran-t, 2023-07-01:
> Thanks. I found there is a more direct solution: set `contype="2" conaffinity="2"` in MJCF to avoid collision

This isn't correct. You should set both to zero.

yuvaltassa, 2023-07-01:
Or just use sites instead of geoms. Sites are designed for this.

https://mujoco.readthedocs.io/en/latest/overview.html#kinematic-tree

zxwang96, 2023-07-02:
Thanks, both using site and setting cons to zero work.