google-deepmind/mujoco #964: Official Julia bindings for MuJoCo?

nic-barbara, 2023-07-02:
Hi MuJoCo team!

I was wondering if there are any plans to develop official [Julia](https://julialang.org/) bindings for MuJoCo in the near future? 

Julia is a great language in which to quickly write fast code to simulate control or reinforcement learning experiments, but at the moment there are no officially-supported bindings for MuJoCo. The only options I can see are:

- Adapting the old [Lyceum/MuJoCo.jl](https://github.com/Lyceum/MuJoCo.jl/tree/master) package for newer versions of MuJoCo
- Calling the Python bindings for MuJoCo from Julia with packages like [PythonCall.jl](https://github.com/cjdoris/PythonCall.jl) (eg: [this implementation](https://github.com/adubredu/MuJoCo.jl))

So far I've been tinkering with the second option and calling Python (which calls C...) from Julia. The first option is more efficient in the long run, but the second was easier to get started with. This is fine for very simple cases, but becomes unnecessarily computationally expensive when many calls to Python are required in more complicated experiments/algorithms. 

An example is simulating feedback control in batches (eg: for reinforcement learning). Methods like `rollout()` are great and help to reduce the number of cross-language function calls required for simulating large batches simultaneously, but there are plenty of other methods for which this is not the case (see my question in https://github.com/deepmind/mujoco/issues/203#issuecomment-1616421921 regarding a batch-mode `mj_transitionFD`). 

It would be amazing to have officially-supported Julia bindings some time in the near future. In the mean time, if you have any suggestions for those of us working with MuJoCo from Julia, any help would be greatly appreciated.

Thank you for all the fantastic work you've done to make MuJoCo open-source and accessible to all :)

yuvaltassa, 2023-07-02:
IIRC Lyceum was quite good, we would certainly support its revival. Perhaps @klowrey (original author of the Julia bindings) would be willing to assist?

Regarding batch processing in Python, @PeterMitrano has recently shown in #897 that Python threading is quite good (only insignificantly slower than C++, at least for not-very-small models), so this this is what we'd recommend for `mjd_transitionFD` as well. If you'd like to add a cell to the [tutorial colab](https://github.com/deepmind/mujoco/blob/main/python/tutorial.ipynb) showing people how to do this (as it's currently hidden in the rollout test), that would be welcome 🙂

closed 2023-07-02 by yuvaltassa

nic-barbara, 2023-07-02:
Thanks for the suggestions @yuvaltassa. 

Regarding `mjd_transitionFD`, wouldn't we have to unlock the GIL first to be able to multi-thread it? And it would still be useful to have a batch-mode method independently of multi-threading (i.e: write the for loop in C/C++) for extra speed on single-thread computations.

Happy to help out with adding a note on threading in the tutorial when I get a chance :)

yuvaltassa, 2023-07-02:
Hmmm I don't think so.

@saran-t is it not the case that all single calls to MuJoCo functions are GIL-free and can be multithreaded using Python threads?

klowrey, 2023-07-05:
Updating the Mujoco.jl wrapper shouldn’t be hard, but I haven’t kept up to date with new Mujoco developments. The only thing in the past that didn’t transfer well was some UI features using struct unions; all the physics stuff was great.

The was no particular reason mujoco.jl was released under the Lyceum project as it is standalone. Feel free to reach out to me to coordinate updates.

saran-t, 2023-07-05:
> @saran-t is it not the case that all single calls to MuJoCo functions are GIL-free and can be multithreaded using Python threads?

The vast majority of functions are called with GIL unlocked. You can find the bindings for `mjd_transitionFD` [here](https://github.com/deepmind/mujoco/blob/main/python/mujoco/functions.cc#L1242-L1272), and the GIL is released [here](https://github.com/deepmind/mujoco/blob/main/python/mujoco/functions.h#L182).