google-deepmind/mujoco #965: Cant use mjr_uploadHField when using passive viewer because of missing context

ripkea, 2023-07-04:
Hi,

I'm currently trying to run a simulation of a dynamic heightfield using the mujoco python bindings. I would like to use the passive viewer (launch_passive) rendering, since it comes with a lot of neat capabilities, but there is no way for me to execute mjr_uploadHField, which is needed to reflect changes in the heightfield, since it requires an mjrContext.  

For the heightfield to chance dynamically,  I need to use _mujoco.mjr_uploadHField_ which requires a context, that the viewer does not expose. I have tried using _context = mujoco.MjrContext(model, mujoco.mjtFontScale.mjFONTSCALE_150.value)_  , however this will crash the application with exit code -1073741819. I've had no luck accesing the context any other way.

Is this intended? If yes, how else should I access the context to execute uploadHField or uploadMesh etc.? Please change the label accordingly, if this is not a bug.

Thanks !

This is a small example of how I tried to run my code: 

`import mujoco
import mujoco.viewer as v
import time
import numpy as np
import random

field_size = 32
base_height = 30
height_perturbation_range = 5
heightfield = np.array([0] * field_size * field_size)

def change_heightfield(heightfield):
    for x in range(int(field_size / 2)):
        for y in range(int(field_size / 2)):
            height = random.uniform(base_height, height_perturbation_range + base_height)
            heightfield[2 * y + 2 * x * field_size] = height
            heightfield[2 * y + 1 + 2 * x * field_size] = height
            heightfield[2 * y + (2 * x + 1) * field_size] = height
            heightfield[2 * y + 1 + (2 * x + 1) * field_size] = height
    return heightfield

xml ="""
<mujoco>
    <asset>
        <hfield name="terrain" nrow="{field_size}" ncol="{field_size}" size="{world_field_size} {world_field_size} 1 1"/>
    </asset>
  <worldbody>
    <light name="top" pos="0 0 1"/>
      <geom name="red_sphere" type="box" size=".1 .5 .5" rgba="1 0 0 1" pos="0 0 4"/>
        <geom name="heightfield" type="hfield" hfield="terrain" 
     />
  </worldbody>
</mujoco>
""".format(field_size=field_size, world_field_size=field_size*0.1)

model = mujoco.MjModel.from_xml_string(xml)
data = mujoco.MjData(model)
mujoco.mj_step(model, data)

with v.launch_passive(model, data) as v:

  start = time.time()
  context = mujoco.MjrContext(model, mujoco.mjtFontScale.mjFONTSCALE_150.value)

  while v.is_running() and time.time() - start < 60:

    model.hfield_data = heightfield.astype(float) * 0.1
    renderer = mujoco.Renderer(model)
    step_start = time.time()

    heightfield = change_heightfield(heightfield)
    model.hfield_data = heightfield.astype(float) * 0.1
    mujoco.mjr_uploadHField(model, context, 0)
    mujoco.mj_step(model, data)
    v.sync()
    time_until_next_step = model.opt.timestep - (time.time() - step_start)
    if time_until_next_step > 0:
      time.sleep(time_until_next_step)`

gaiyi7788, 2023-07-14:
I meet same problem. Have you finish this problem yet?

closed 2023-07-14 by copybara-service[bot]

saran-t, 2023-07-20:
This feature is now available in version 2.3.7 which was just released. Please consult [our documentation](https://mujoco.readthedocs.io/en/stable/python.html#passive-viewer) for the new viewer API for this.

commit 0ccfef73 (2023-07-14): Add `update_{hfield,mesh,texture}` methods to the passive viewer.

Fixes #812.
Fixes #958.
Fixes #965.

PiperOrigin-RevId: 548181282
Change-Id: I6eb0cd2ce52c66083e83dd08d69e0cfb4178dbba