google-deepmind/mujoco #966: collision and contact error

jheyeo222, 2023-07-05:
**How to ask for help**

Hi,

I'm a student  and I'm trying to use MuJoCo for fun.

I'm looking for some help with collision and contact.

Here is a model which explains my question:

<details>
  <summary>minimal XML</summary>

 <option timestep="0.001" integrator="RK4" gravity="0 0 -9.8">
		<flag sensornoise="disable" contact="enable" energy="enable"  multiccd="enable"/>
</option>

  <worldbody>
    <light name="spotlight" mode="targetbodycom" target="wrist_2_link" pos="0 -1 2"/>
   
    
		<geom pos = "0.7 0 0.5" type="box" size="0.5 0.7 0.9" rgba="0 0.9 0 1" mass="5"/>

    <body name="base" >
      <inertial mass="4.0" pos="0 0 0" diaginertia="0.00443333156 0.00443333156 0.0072"/>
      <geom mesh="base_0" material="black" class="visual"/>
      <geom mesh="base_1" material="jointgray" class="visual"/>

      <body name="shoulder_link" pos="0 0 0.163">
        <inertial mass="3.7" pos="0 0 0" diaginertia="0.0102675 0.0102675 0.00666"/>
        <joint name="shoulder_pan_joint" class="size3" axis="0 0 1"/>
        <geom mesh="shoulder_0" material="urblue" class="visual"/>
        <geom mesh="shoulder_1" material="black" class="visual"/>
        <geom mesh="shoulder_2" material="jointgray" class="visual"/>
        <geom class="collision" size="0.06 0.06" pos="0 0 -0.04"/>

        <body name="upper_arm_link" pos="0 0.138 0" >
          <inertial mass="8.393" pos="0 0 0.2125" diaginertia="0.133886 0.133886 0.0151074"/>
          <joint name="shoulder_lift_joint" class="size3"/>
          <geom mesh="upperarm_0" material="linkgray" class="visual"/>
          <geom mesh="upperarm_1" material="black" class="visual"/>
          <geom mesh="upperarm_2" material="jointgray" class="visual"/>
          <geom mesh="upperarm_3" material="urblue" class="visual"/>
          <geom class="collision" pos="0 -0.04 0" quat="1 1 0 0" size="0.06 0.06"/>
          <geom class="collision" size="0.05 0.2" pos="0 0 0.2"/>
.
.
.
.

</details>

Here is a screenshot / video, illustrating my question:
[image]

I'm trying to implement precise force control by attaching a force sensor to the end of the manipulator. For this to work, the UR5 needs to collide with the green wall. However, instead of colliding, the UR5 is penetrating through the wall. How can I resolve this issue?

For your information, I'm using Python and running the simulation with mj.step.

[mujoco_issu.zip](https://github.com/deepmind/mujoco/files/11954661/mujoco_issu.zip)

saran-t, 2023-07-05:
I loaded your `ur5e.xml` in `simulate` and everything seems to be working fine. Are you sure your Python code isn't disabling contacts somehow?

[image]

jheyeo222, 2023-07-05:
> I loaded your `ur5e.xml` in `simulate` and everything seems to be working fine. Are you sure your Python code isn't disabling contacts somehow?
> 
> [image]

I apologize for troubling you with such a simple matter.

I have attached a Python file I wrote to Alzip. Essentially, it's a code to solve IK. I've included the Python file "manipulator_position4" in the mujoco_issue folder.

However, collisions are continuously ignored in the Python code. Could you possibly identify the cause?

Also, it seems like there are not enough Mujoco Python examples available.

Could you recommend a good site with Mujoco Python examples for a Windows OS, Python user to study?

I'm particularly interested in various topics such as IK, force control, RL, etc.
[mujoco_issue.zip](https://github.com/deepmind/mujoco/files/11959204/mujoco_issue.zip)

jheyeo222, 2023-07-05:
The file is too large to attach via email. I've added a new response and
attached my Python file to this issue:
https://github.com/deepmind/mujoco/issues/966. Could you possibly provide a
response there?

> Message ID: ***@***.***>
>

jheyeo222, 2023-07-05:
> I loaded your `ur5e.xml` in `simulate` and everything seems to be working
fine. Are you sure your Python code isn't disabling contacts somehow?
>
> ![Screenshot 2023-07-05 at 12 40 43 PM](
https://user-images.githubusercontent.com/2200217/251138554-30cc86f0-5010-40a4-9458-e05569e4ac53.png
)

I apologize for troubling you with such a simple matter.

I have attached a Python file I wrote to Alzip. Essentially, it's a code to
solve IK. I've included the Python file "manipulator_position4" in the
mujoco_issue folder.

The following code is the loop that runs the simulation in
manipulator_position4.py. However, there's nothing particularly unusual
about it.
```
while not glfw.window_should_close(window):
    time_prev = time

    while (time - time_prev < 1.0/100.0):

        position_Q = data.site_xpos[0]
        rotation_R = data.xmat[7]
        rotation_R = rotation_R.reshape(3,3)

        roll = np.arctan2(rotation_R[2, 1], rotation_R[2, 2])
        pich = np.arctan2(-rotation_R[2,0], np.sqrt((rotation_R[2,1]**2) +
(rotation_R[2, 2]**2)) )
        yaw = np.arctan2(rotation_R[1,0], rotation_R[0,0])

        jacp = np.zeros((3,6)) #3 is for x,y,z and 2 is for theta1 and
theta2
        jacr = np.zeros((3,6)) #3 is for x,y,z and 2 is for theta1 and
theta2
        mj.mj_jac(model,data,jacp,jacr,position_Q,7)

        J = np.vstack((jacp, jacr))

        J_transpose = J.T

        Jsudoinv = np.dot(J_transpose,np.linalg.inv(np.dot(J,J_transpose) +
0.01* np.eye(6)))

        Rotation_mat_ref = euler_to_rotation_matrix(roll_ref[i],
pich_ref[i], yaw_ref[i])
        delta = np.dot(Rotation_mat_ref, rotation_R.T)

        delta_roll = np.arctan2(delta[2, 1], delta[2, 2])
        delta_pich  = np.arctan2(-delta[2,0], np.sqrt((delta[2,1]**2) +
(delta[2, 2]**2)) )
        delta_yaw = np.arctan2(delta[1,0], delta[0,0])

        dX = np.array([x_ref[i]-position_Q[0],   y_ref[i]- position_Q[1],
  z_ref[i]-position_Q[2],
                       delta_roll,         delta_pich,            delta_yaw
    ])

        dq = 0.01*Jsudoinv.dot(dX)

        theta1  +=  dq[0]
        theta2  +=  dq[1]
        theta3  +=  dq[2]
        theta4  +=  dq[3]
        theta5  +=  dq[4]
        theta6  +=  dq[5]

        data.qpos[0] = theta1;
        data.qpos[1] = theta2;
        data.qpos[2] = theta3;
        data.qpos[3] = theta4;
        data.qpos[4] = theta5;
        data.qpos[5] = theta6;

        mj.mj_step(model, data)

        time +=dt
```

However, collisions are continuously ignored in the Python code. Could you
possibly identify the cause?

Also, it seems like there are not enough Mujoco Python examples available.

Could you recommend a good site with Mujoco Python examples for a Windows
OS, Python user to study?

I'm particularly interested in various topics such as IK, force control,
RL, etc.
[mujoco_issue.zip](
https://github.com/deepmind/mujoco/files/11959204/mujoco_issue.zip)

saran-t, 2023-07-05:
You're manually setting `qpos`, presumably to a self-intersecting configuration. MuJoCo can't magically change those values for you. The single call to `mj_step` is going to _try_ and apply large forces to resolve the penetration, but you're only taking a single step before overriding the `qpos` again.

What exactly is it that you're trying to do? Are you try to place the end effector at the closest possible point to some target without any insection?

jheyeo222, 2023-07-05:
I'm aiming to apply model-based force control to a robot. To achieve this,
I plan to install force sensors on each joint and at the end effector of
the robot. The idea is to maintain a constant force against a wall while
moving the robot's end effector along a predetermined path. You can
visualize this process as a welding robot moving along a defined path on a
wall, maintaining a steady force as it welds.

My second objective is to implement this process through reinforcement
learning and then compare it with the model-based force control approach.
This comparison is my ultimate goal.

To achieve this, I've solved inverse kinematics (IK) as a basic step using
the above code. The IK is working well, and I can send the robot to the
desired location. However, there's an issue where the robot is passing
through the wall.

Moreover, there is a scarcity of examples designed for Windows Python,
which is making the process challenging.

jheyeo222, 2023-07-05:
> You're manually setting `qpos`, presumably to a self-intersecting configuration. MuJoCo can't magically change those values for you. The single call to `mj_step` is going to _try_ and apply large [...]

I'm aiming to apply model-based force control to a robot. To achieve this, I plan to install force sensors on each joint and at the end effector of the robot. The idea is to maintain a constant force against a wall while moving the robot's end effector along a predetermined path. You can visualize this process as a welding robot moving along a defined path on a wall, maintaining a steady force as it welds.

My second objective is to implement this process through reinforcement learning and then compare it with the model-based force control approach. This comparison is my ultimate goal.

To achieve this, I've solved inverse kinematics (IK) as a basic step using the above code. The IK is working well, and I can send the robot to the desired location. However, there's an issue where the robot is passing through the wall.

Moreover,it's challenging because there aren't any examples available that are tailored for Windows Python. This makes the process difficult.

yuvaltassa, 2023-07-05:
You're supposed to use actuators to apply forces. You can't just set positions or velocities. The world is second order. `f = M * a` means you can only control second derivatives. You set the forces via `ctrl`, MuJoCo does the rest.

closed 2023-07-19 by kevinzakka