google-deepmind/mujoco #971: How to get Force Distribution over multiple tendons

AmirSamanMirjalili, 2023-07-07:
Hi,

I'm a student and I'm trying to use MuJoCo for my project.

I'm looking for some help with sensing tension force inside tendons, in the following model i have used four tendons which suppur a box with 5kg mass  i attach four sensors to the site points on the box and I'm expected to read at least 50/4=12.5N for each sensor when the box is in a symmetrical stationary pose but im reading 49N for each sensor which is not correct , what am i doing wrong here .

Here is a XML model which explains my question:

<details>
  <summary>minimal XML</summary>

```XML
<mujoco>
  <option />
  <worldbody>
    <geom type="plane" size="10 10 .1"/>
    <light pos="0 0 10"/>
    <site name="anchor" pos="0 0 5" size="0.03" rgba="0 0 1 1"/>
    <body name="box" pos="0 0 2">
      <freejoint/>
      <geom type="box" mass="5" size=".5 .5 .5"/>
      <site name="a1" pos=".5 .5 .5" rgba="0 0 1 1" size="0.03"/>
      <site name="a2" pos=".5 -.5 .5" rgba="0 0 1 1" size="0.03"/>
      <site name="a3" pos="-.5 .5 .5" rgba="0 0 1 1" size="0.03"/>
      <site name="a4" pos="-.5 -.5 .5" rgba="0 0 1 1" size="0.03"/>
    </body>
  </worldbody>
  <tendon>
    <spatial name="tendon1" limited="true" range="0 1" damping="1">
      <site site="anchor"/>
      <site site="a1"/>
    </spatial>
    <spatial name="tendon2" limited="true" range="0 1" damping="1">
      <site site="anchor"/>
      <site site="a2"/>
    </spatial>
    <spatial name="tendon3" limited="true" range="0 1" damping="1">
      <site site="anchor"/>
      <site site="a3"/>
    </spatial>
    <spatial name="tendon4" limited="true" range="0 1" damping="1">
      <site site="anchor"/>
      <site site="a4"/>
    </spatial>
  </tendon>
  <actuator>
    <position
      name="tendon1" tendon="tendon1"
      ctrllimited="true" ctrlrange="1 2"
      kp="100000"/>
    <position
      name="tendon2" tendon="tendon2"
      ctrllimited="true" ctrlrange="1 2"
      kp="100000"/>
    <position
      name="tendon3" tendon="tendon3"
      ctrllimited="true" ctrlrange="1 2"
      kp="100000"/>
    <position
      name="tendon4" tendon="tendon4"
      ctrllimited="true" ctrlrange="1 2"
      kp="100000"/>
  </actuator>

  <sensor>
	<force site="a1"/>
	<force site="a2"/>
	<force site="a3"/>
	<force site="a4"/>
	</sensor>
</mujoco>
```

</details>

Here is a screenshot , illustrating my question:

[image]

yuvaltassa, 2023-07-07:
Read the [force sensor documentation](https://mujoco.readthedocs.io/en/latest/XMLreference.html#sensor-force) carefully, it doesn't do what you think it does. I believe you are looking for the [tendon limit force sensor](https://mujoco.readthedocs.io/en/latest/XMLreference.html#sensor-tendonlimitfrc)

yuvaltassa, 2023-07-07:
Relatedly, there is a known open bug (#833) regarding force measurement in models that include tendons (however in a different setting than yours). Please let us know if you find that this bug is blocking for you, that could help us prioritise.

AmirSamanMirjalili, 2023-07-08:
thanks, @yuvaltassa for your kind guidance, right now in the following scenario i have tried out using the `tendonlimitfrc` for the tendons force sensing on the site point of my box , from the symmetrical and stationary pose of the box (figure1) one can infer that each tendon should bear 5(box mass in kg) * 9.8 / 4 (four tendon supporting the mass)=12.25(N) but as illustrated from the following screenshot each tendon shows a tension around 2.5(N) which is completely false . I hope I have not made a mistake here, in that case i appreciate if you correct me , 

```xml
<mujoco>
  <option gravity="0 0 -9.81" />
  <worldbody>
    <geom type="plane" size="10 10 0.1" />
    <light pos="0 0 10" />
    <site name="anchor1" pos="0.5 0.5 5" size="0.03" rgba="0 0 1 1" />
    <site name="anchor2" pos="0.5 -0.5 5" size="0.03" rgba="0 0 1 1" />
    <site name="anchor3" pos="-0.5 0.5 5" size="0.03" rgba="0 0 1 1" />
    <site name="anchor4" pos="-0.5 -0.5 5" size="0.03" rgba="0 0 1 1" />
    <body name="box" pos="0 0 2">
      <freejoint />
      <geom type="box" mass="1" size=".5 .5 .5" rgba="0.8 0.8 0.8 0.3"/>
      <site name="a1" pos=".5 .5 .5" rgba="0 0 1 1" size="0.03" />
      <site name="a2" pos=".5 -.5 .5" rgba="0 0 1 1" size="0.03" />
      <site name="a3" pos="-.5 .5 .5" rgba="0 0 1 1" size="0.03" />
      <site name="a4" pos="-.5 -.5 .5" rgba="0 0 1 1" size="0.03" />
    </body>
  </worldbody>
  <tendon>
    <spatial name="tendon1" limited="true" range="0 1" damping="100">
      <site site="anchor1" />
      <site site="a1" />
    </spatial>
    <spatial name="tendon2" limited="true" range="0 1" damping="100" >
      <site site="anchor2" />
      <site site="a2" />
    </spatial>
    <spatial name="tendon3" limited="true" range="0 1" damping="100">
      <site site="anchor3" />
      <site site="a3" />
    </spatial>
    <spatial name="tendon4" limited="true" range="0 1" damping="100">
      <site site="anchor4" />
      <site site="a4" />
    </spatial>
  </tendon>
  <actuator>
    <motor name="tendon11" tendon="tendon1" ctrllimited="true" ctrlrange="-6 2" gear="4" />
    <motor name="tendon22" tendon="tendon2" ctrllimited="true" ctrlrange="-6 2" gear="4" />
    <motor name="tendon33" tendon="tendon3" ctrllimited="true" ctrlrange="-6 2" gear="4" />
    <motor name="tendon44" tendon="tendon4" ctrllimited="true" ctrlrange="-6 2" gear="4" />
    <!-- <position name="tendon1" tendon="tendon1" ctrllimited="true" ctrlrange="1 2" kp="100000"/>
    <position name="tendon2" tendon="tendon2" ctrllimited="true" ctrlrange="1 2" kp="100000"/>
    <position name="tendon3" tendon="tendon3" ctrllimited="true" ctrlrange="1 2" kp="100000"/>
    <position name="tendon4" tendon="tendon4" ctrllimited="true" ctrlrange="1 2" kp="100000"/> -->
  </actuator>

  <sensor>
    <tendonlimitfrc tendon="tendon1" />
    <tendonlimitfrc tendon="tendon2" />
    <tendonlimitfrc tendon="tendon3" />
    <tendonlimitfrc tendon="tendon4" />
  </sensor>
</mujoco>
```

figure 1 :
[image]

yuvaltassa, 2023-07-08:
You're setting `mass = "1"`?

AmirSamanMirjalili, 2023-07-09:
@yuvaltassa  so sorry for this  awkward mistake , i just have one last question . in the following video ( XML model also attached) when i change the tendon length range (from (0,1) to (0, 5) i get zero in force sensor `tendonlimitfrc ` i suppose the reason is that the length of the tendons is in under tension range so that the sensor reads zero and so there is no force in tendon,

how can i avoid this behavior so that in any length of tendon it bears the tension force. i appreciate your time and kindness

```xml
<mujoco>
  <option gravity="0 0 -9.81" />
  <worldbody>
    <geom type="plane" size="10 10 0.1" />
    <light pos="0 0 10" />
    <site name="anchor1" pos="0.5 0.5 5" size="0.03" rgba="0 0 1 1" />
    <site name="anchor2" pos="0.5 -0.5 5" size="0.03" rgba="0 0 1 1" />
    <site name="anchor3" pos="-0.5 0.5 5" size="0.03" rgba="0 0 1 1" />
    <site name="anchor4" pos="-0.5 -0.5 5" size="0.03" rgba="0 0 1 1" />
    <body name="box" pos="0 0 2">
      <freejoint />
      <geom type="box" mass="1" size=".5 .5 .5" rgba="0.8 0.8 0.8 0.3"/>
      <site name="a1" pos=".5 .5 .5" rgba="0 0 1 1" size="0.03" />
      <site name="a2" pos=".5 -.5 .5" rgba="0 0 1 1" size="0.03" />
      <site name="a3" pos="-.5 .5 .5" rgba="0 0 1 1" size="0.03" />
      <site name="a4" pos="-.5 -.5 .5" rgba="0 0 1 1" size="0.03" />
    </body>
  </worldbody>
  <tendon>
    <spatial name="tendon1" limited="true" range="0 1" damping="100">
      <site site="anchor1" />
      <site site="a1" />
    </spatial>
    <spatial name="tendon2" limited="true" range="0 1" damping="100" >
      <site site="anchor2" />
      <site site="a2" />
    </spatial>
    <spatial name="tendon3" limited="true" range="0 1" damping="100">
      <site site="anchor3" />
      <site site="a3" />
    </spatial>
    <spatial name="tendon4" limited="true" range="0 1" damping="100">
      <site site="anchor4" />
      <site site="a4" />
    </spatial>
  </tendon>
  <actuator>
    <motor name="tendon11" tendon="tendon1" ctrllimited="true" ctrlrange="-6 2" gear="4" />
    <motor name="tendon22" tendon="tendon2" ctrllimited="true" ctrlrange="-6 2" gear="4" />
    <motor name="tendon33" tendon="tendon3" ctrllimited="true" ctrlrange="-6 2" gear="4" />
    <motor name="tendon44" tendon="tendon4" ctrllimited="true" ctrlrange="-6 2" gear="4" />
    <!-- <position name="tendon1" tendon="tendon1" ctrllimited="true" ctrlrange="1 2" kp="100000"/>
    <position name="tendon2" tendon="tendon2" ctrllimited="true" ctrlrange="1 2" kp="100000"/>
    <position name="tendon3" tendon="tendon3" ctrllimited="true" ctrlrange="1 2" kp="100000"/>
    <position name="tendon4" tendon="tendon4" ctrllimited="true" ctrlrange="1 2" kp="100000"/> -->
  </actuator>

  <sensor>
    <tendonlimitfrc tendon="tendon1" />
    <tendonlimitfrc tendon="tendon2" />
    <tendonlimitfrc tendon="tendon3" />
    <tendonlimitfrc tendon="tendon4" />
  </sensor>
</mujoco>
```

tendons zero sensor read out in under tension zone

[image]

yuvaltassa, 2023-07-09:
That can't be implemented with a length range. Sounds like maybe you want to attach a position actuator to your tendon, and control that? Look at #803 and the model and video therein, and compare to the video and model below.

https://github.com/deepmind/mujoco/assets/6387526/9377a454-ce30-4068-b4cc-3e37c052a275

<details>
  <summary>tendon actuator model</summary>
  
```XML
<mujoco>
  <compiler autolimits="true"/>

  <default>
    <site type="sphere" rgba="1 0 0 1" size="0.005"/>
    <tendon rgba="0 1 0 1"/>
  </default>
  
  <visual>
    <headlight diffuse=".7 .7 .7"/>
  </visual>

  <worldbody>
    <body name="arm">
      <joint name="arm" axis="0 1 0" damping=".04"/>
      <geom name="shoulder" type="cylinder" size=".05 .01" zaxis="0 1 0"/>
      <geom name="arm" type="box" size=".01" fromto="0 0 0 -.3 0 0"/>
      <geom name="crossbar" type="box" size=".01" fromto="-.1 0 -.05 -.1 0 .05"/>
      <site name="arm" pos="-.1 0 .05"/>
    </body>
    <site name="right" pos=".1 0 0"/>
    <body name="slider" pos=".05 0 -.2">
      <joint name="slider" type="slide" damping="1"/>
      <geom name="slider" type="box" size=".01 .01 .01"/>
      <site name="slider" pos="0 0 .01"/>
    </body>
  </worldbody>

  <tendon>
    <spatial name="rope" range="0 .35">
      <site site="slider"/>
      <geom geom="shoulder" sidesite="right"/>
      <site site="arm"/>
    </spatial>
  </tendon>

  <actuator>
    <position name="slider" joint="slider" ctrlrange="-.05 .05" kp="100"/>
  </actuator>
</mujoco>
```

</details>

closed 2023-07-09 by yuvaltassa