google-deepmind/mujoco #973: needing guidence for cable driven robot simulation

MohammadrezaDindarloo, 2023-07-08:
hi to everyone

I have some questions in simulating a cabel robot in mujoco. actually my problem is to modeling the cables. at first I used tendon to modeling my cables because of some good features but I ended up to using cable for this means because the cable have some features like bending and twisting that are suitable. my simulation in cable comming up with this: 
[image]

this was good but thre was a problem that when I increase the count of composite type of cable the simulation time was very slow and it couldn't be handeled to working with this(my cable length are almost 4*10 meters) and when I decrease the count of cable composite the simulation couldn't handle the simulating and it get unstable and reset simulation. I tried a lot to find the solution but I couldn't. if anyone have any seggestion please help in this matter.
In second scenario I tried to simulate the robots cables with tendon. in this matter I did some simulation like bellow:
[image]

this simulation will be good fore me if I could measure the force in tendons and also impelement a scenario that an actuator handle my cable length (in reality we see that if we want to drive a box with two cable when we pull one of the cables the other cable get sag with out supporting any force) I tried to simulate this reality but I couldn't. I don't want the tendon haveing any push when I get the motor free but is do like the below :
[image]
now I don't know if there is any solution for me? can I handle this with tendon module? or any idea to handle my cable model? (I expect to have 4 motos in the top an endeffector that can drive the end effector with these motors through cables) thank you anyone for your helping.

quagla, 2023-07-11:
Have you tried to decrease the cable stiffness to make the simulation stable? I would recommend using tendons if your cable is in a tensile configuration and the cable composite if it is in a bent/twisted configuration. What do you mean by the tendon not having any push?

quagla, 2023-08-04:
I will close it as there is no update I assume you managed to solve the issue.

closed 2023-08-04 by quagla