google-deepmind/mujoco #982: what does `data.cinert` consist of

Kallinteris-Andreas, 2023-07-13:
Hi,

I'm a maintainer of Gymnasium-Robotics & Gymnasium, and I'm trying to use MuJoCo to document The Gymnasium/MuJoCo environments.

What are the 10 elements of `data.cinert` per body part.

All I could find in the documentation is this:  https://mujoco.readthedocs.io/en/latest/APIreference/APItypes.html?highlight=cinert#mjdatam, which simple states that there are 10 elements (but not what those elements represent).

Thank you!

closed 2023-07-14 by copybara-service[bot]

yuvaltassa, 2023-07-14:
I added [a note](https://mujoco.readthedocs.io/en/latest/APIreference/APItypes.html#c-frame-variables)

dt1729, 2023-12-09:
Quick question @yuvaltassa, the com frame of reference Moment of inertia is 6 dim? If so, my analytical controller requires 3 dim in com frame of reference, how do I convert? Also it's mentioned upper triangular 6 variables are given what about the rest? I believe I'm missing something major here, any guidance will be of great help, thanks!

yuvaltassa, 2023-12-09:
The answers to your specific questions are:

1. You discard the first 3. The order is {rotation, translation} (aka Featherstone notation).
2. The matrix is symmetric.

That said, I don't think those are the answers you want, I get the sense that you haven't quite formulated the question. Without really knowing what you are doing, it is very rare that someone needs direct access to the c-frame variables. It is almost certain these are not the variables you are looking for.

dt1729, 2023-12-10:
Thanks for such a prompt reply: I'm trying to develop an MPC controller and for that I have found simplified model of the quadruped, the model requires the Moment of Inertia of the torso in the CoM frame. The moment of inertia is 3x3 in the analytical model, now, coming to mujoco, i found that cinert would give me the moment of inertia of every body in it's CoM frame, however it turns out that cinert is in Fetherstone's notation. In that case would this be an appropriate equation to follow? If not how do I convert the 6 dim cinert into 3 dim (cartesian) moment of inertia? 

I am thinking this is the analytical way to write it: 
[image]

yuvaltassa, 2023-12-10:
Hold on, taking a step back, why would you want to use some simplified model when the full model is fast enough?

Here is a screen capture of real-time MPC quadruped control, running on a laptop: https://youtu.be/esLuwaWz4oE

This is from our open-source MJPC tool at https://github.com/google-deepmind/mujoco_mpc

dt1729, 2023-12-10:
I have seen MJPC I wanted to recreate results of some paper.

yuvaltassa, 2023-12-10:
You don't want `cinert`, you just want the body inertia.

In Python: `model.body('torso').inertia`
In C: `mjModel.body_inertia + 3*torso_id`

Note these are 3 numbers since they are already in the (diagonalized) inertial frame.

dt1729, 2023-12-10:
Thanks, I revisited the equations and saw that it's just a global frame. Thanks for all the help @yuvaltassa !

Aryan-Satpathy, 2024-09-15:
Hi,
I am trying to retrieve the moment of inertia tensor for entities programmatically. However, for a body containing a sphere geom, I get 0 in all values of `data.cinert` except for the last entry (mass), and I get 0 in all values in `model.body('m_0').inertia`. Interestingly, I get non-zero values for other entities in the scene. Would you have any idea about what's wrong @yuvaltassa ?

<details>
  <summary>Here is the XML and some code to view the values (and their corresponding outputs):</summary>
 
 variable values
  ```python
  mass= 5
  g = 2
  ```
 xml is python string formatted by the above variables
 ```xml
 <mujoco model="sphere_on_incline">
    <compiler angle="radian" autolimits="true"/>
    <compiler meshdir="asset" texturedir="asset" assetdir="asset"/>
    <!-- <include file="scene.xml"/> -->
    <option gravity="0 0 {g}"/>
    <default class="main">
        <geom rgba="0.9 0.7 0.1 1"/>
    </default>
    <visual>
        <headlight diffuse="0.7 0.7 0.7"/>
        <global offwidth="1000" offheight="1000"/>
        <quality offsamples="32"/>
    </visual>
    <asset>
        <texture type="2d" name="checks1" builtin="checker" rgb1="1 1 1" rgb2="0 0 0" width="256" height="256"/>
        <material name="test_color" texture="checks1" texuniform="true" texrepeat="2 2"/>
    </asset> 
<worldbody>
<body name="incline_0" pos="0.0 0.0 0.0">
<geom type="box" size="0.7 0.015 0.015" pos="0 0 0" quat="0.8660254037844387 0.0 0.49999999999999994 0.0" rgba="0.5 0.5 0.5 1" mass="1" condim="1" />
<site name="incline_0" pos="0.0075 0.0 0.01299038105676658"/>  
<!-- <joint type="slide" axis="1 0 0"/>  -->
</body>
<body name="m_0" pos="-0.10498254121265795 0.0 0.23183509528801702">
<geom type="sphere" pos="0 0 0" size="0.01" rgba="0.5 0.5 0.5 1" mass="{mass}" quat="0.8660254037844387 0.0 0.49999999999999994 0.0" condim="1"/>
<site name="m_0_center" pos="0.0 0.0 0.00"/>
<site name="m_0_top" pos="0.0 0.0 0.01"/>
<joint type="free" stiffness="0" damping="0" frictionloss="0" armature="0"/>
</body>
</worldbody>
<equality>
</equality>
<sensor>
<accelerometer name="m_0_acc" site="m_0_top"/>
</sensor> 
</mujoco>
 ```
 
 To view the values:
 ```python
 mujoco.mj_step(model, data)
 mujoco.mj_comPos(model, data)
 mujoco.mj_rnePostConstraint(model, data)
 
 print(data.cinert)
 # output
 # [[0.    0.    0.    0.    0.    0.    0.    0.    0.    0.   ]
 # [0.123 0.163 0.041 0.    0.071 0.    0.    0.    0.    1.   ]
 # [0.    0.    0.    0.    0.    0.    0.    0.    0.    5.   ]]
 
 print(model.body('m_0').inertia)
 # output
 # [0. 0. 0.]
 
 print(model.body_inertia)
 # output
 # [[0.    0.    0.   ]
 # [0.    0.163 0.163]
 # [0.    0.    0.   ]]
 ```

</details>

yuvaltassa, 2024-09-15:
I think this is a print precision thing. Either increase print precision or make your geom bigger, and you will see the inertia. Remember that inertia scales like $r^2$, so `<geom ... size="0.01" ...` adds a coefficient of $10^{-4}$.

Aryan-Satpathy, 2024-09-15:
Yes, that was the case, it seems, I changed the radius to 0.1, and I can see the values that I was expecting. Thanks for the help.

commit b5bef1b6 (2023-07-14): Describe c-frame variables.

Fixes #982.

PiperOrigin-RevId: 548060614
Change-Id: I2164afc07c02e657b1b924a5afc59209356b57eb
commit 1885d518 (2023-12-10): Use the string `(rot:lin)` in mjdata.h to clarify where 6D motion vectors use the Featherstone convention. Also add clarifying text to the appropriate note in the documentation.

Fixes #982.

PiperOrigin-RevId: 589556339
Change-Id: I3e7ff185a089715a386694321d4b89d09d481cf9