google-deepmind/mujoco #984: [Question] Ground-truth Dynamics Model

KehanLong, 2023-07-14:
Hi, could anyone tell me if we could access the ground-truth dynamics model of the mujoco models? For example, is there a place we could find the equations (dynamics) for the models (e.g. cart-pole, half-cheetah, and so on). I understand there are some parameters in the xml file, but I could not find any equations that express the motions of the model.

yuvaltassa, 2023-07-14:
Your question is unclear, the equations _are_ the simulator.
The code that implements the dynamics is [here](https://github.com/deepmind/mujoco/tree/main/src/engine).

closed 2023-07-14 by yuvaltassa

IntrovertInDisguise, 2025-06-11:
More towards analytical models that may be used as a reference

yuvaltassa, 2025-06-12:
MuJoCo is a C implementation of the analytical model. 
If you find C hard to read then I recommend a pen and paper, or perhaps something like Mathematica.