google-deepmind/mujoco #986: Asking for advice on improving the simulation performance

breakds, 2023-07-14:
Hi,

Greetings! We have recently been utilizing MuJoCo for our project involving the simulation of a Unitree Go1 quadrupedal robot navigating across varied terrains. However, we're facing some performance bottlenecks during our training cycles. Consequently, we're seeking your insights and advice on potential ways to enhance the simulation speed.

Below, I've provided some context and specifics from our experiment conducted on an AMD Ryzen 9 5950x CPU, which has 16 cores and 32 threads.

**Question 1**: Discrepancies in Simulation Speed Factors

The [MuJoCo MPC paper](https://arxiv.org/abs/2212.00541) indicates that a single CPU thread can simulate a 27-DoF humanoid system 4000 times faster than real-time. 

> for example, the humanoid (a 27-DoF system) can be simulated 4000 times faster than realtime on a single CPU thread.

To replicate this performance, we ran `testspeed` on the humanoid model and achieved a real:sim factor of roughly 150 times for a single thread. Although this surpasses the speed reported in post #897, it falls far short of the 4000x speed cited in the paper.

```
Running 10000 steps per thread at dt = 0.005 ...

Summary for all 16 threads

 Total simulation time  : 0.44 s
 Total steps per second : 366686
 Total realtime factor  : 1833.43 x
 Total time per step    : 2.7 µs

Details for thread 0

 Simulation time      : 0.34 s
 Steps per second     : 29722
 Realtime factor      : 148.61 x
 Time per step        : 33.6 µs

 Broadphase accuracy  : 44.85%
 Midphase accuracy    : 40.36%
 Contacts per step    : 12.56
 Constraints per step : 43.56
 Degrees of freedom   : 27

 Internal profiler for thread 0 (µs per step)
             step :   33.5  (100.00 %)
          forward :   31.8  ( 95.04 %)
         position :   17.1  ( 51.03 %)
         velocity :    1.7  (  5.08 %)
        actuation :    0.3  (  0.92 %)
     acceleration :    0.6  (  1.72 %)
       constraint :   12.0  ( 35.76 %)
   pos_kinematics :    2.4  (  7.07 %)
      pos_inertia :    2.0  (  5.84 %)
    pos_collision :    8.3  ( 24.73 %)
         pos_make :    4.3  ( 12.80 %)
      pos_project :    0.0  (  0.08 %)
            other :    1.8  (  5.49 %)
```

Can anyone guide us on how we can approach this 4000x magnitude? Are there any specific configurations, compilation flags, or model modifications that we need to consider? We are building MuJoCo from the source and any advice regarding this matter would be highly appreciated!

**Question 2**:

For our specific Go1 robot and terrain map setting ([see the model here](https://files.breakds.org/wl/?id=YzB5ipmAAZstXnBTUE8bxo0cz3e6T3RX)), the simulation speed dramatically decreases compared to the humanoid model.

```
Running 10000 steps per thread at dt = 0.005 ...

Summary for all 16 threads

 Total simulation time  : 10.41 s
 Total steps per second : 15377
 Total realtime factor  : 76.88 x
 Total time per step    : 65.0 µs

Details for thread 0

 Simulation time      : 8.27 s
 Steps per second     : 1209
 Realtime factor      : 6.05 x
 Time per step        : 826.9 µs

 Broadphase accuracy  : 24.62%
 Midphase accuracy    : 12.47%
 Contacts per step    : 40.70
 Constraints per step : 254.77
 Degrees of freedom   : 18

 Internal profiler for thread 0 (µs per step)
             step :  826.7  (100.00 %)
          forward :  825.7  ( 99.88 %)
         position :  178.0  ( 21.53 %)
         velocity :    2.2  (  0.26 %)
        actuation :    0.2  (  0.03 %)
     acceleration :    0.3  (  0.04 %)
       constraint :  642.7  ( 77.74 %)
   pos_kinematics :    2.2  (  0.26 %)
      pos_inertia :    1.0  (  0.12 %)
    pos_collision :  160.1  ( 19.37 %)
         pos_make :   14.5  (  1.75 %)
      pos_project :    0.0  (  0.00 %)
            other :    3.3  (  0.40 %)
```

This MJCF model works adequately for policy training, but the simulation speed is considerably slower, which hinders our process. The terrain map consists of 1600 x 1600 points. Are there any suggestions on improving the performance for this specific scenario?

Thank you for your time and consideration. Your insights and assistance are greatly valued!

breakds, 2023-07-14:
Below is the build log when I build MuJoCo from source. AVX should be enabled by default.

<details>
  <summary>Build Log for MuJoCo</summary>
  
  ```
mujoco> unpacking sources
mujoco> unpacking source archive /nix/store/wxivcwsqz6zylr31hnfx1cckbwl8wr3i-source
mujoco> source root is source
mujoco> patching sources
mujoco> applying patch /nix/store/fpka17kjcczv7zc1cmq4j7p7dpf49pcp-dependencies.patch
mujoco> patching file cmake/MujocoDependencies.cmake
mujoco> patching file simulate/cmake/SimulateDependencies.cmake
mujoco> configuring
mujoco> fixing cmake files...
mujoco> cmake flags: -DCMAKE_FIND_USE_SYSTEM_PACKAGE_REGISTRY=OFF -DCMAKE_FIND_USE_PACKAGE_REGISTRY=OFF -DCMAKE_EXPORT_NO_PACKAGE_REGISTRY=ON -DCMAKE_BUILD_TYPE=Release -DBUILD_TESTING=OFF -DCMAKE_INSTALL_LOCALEDIR=/nix/store/hj2d8zyd3yjb5n6s5m1m1k22i5rbniwh-mujoco-2.3.6/share/locale -DCMAKE_INSTALL_LIBEXECDIR=/nix/store/hj2d8zyd3yjb5n6s5m1m1k22i5rbniwh-mujoco-2.3.6/libexec -DCMAKE_INSTALL_LIBDIR=/nix/store/hj2d8zyd3yjb5n6s5m1m1k22i5rbniwh-mujoco-2.3.6/lib -DCMAKE_INSTALL_DOCDIR=/nix/store/hj2d8zyd3yjb5n6s5m1m1k22i5rbniwh-mujoco-2.3.6/share/doc/mujoco -DCMAKE_INSTALL_INFODIR=/nix/store/hj2d8zyd3yjb5n6s5m1m1k22i5rbniwh-mujoco-2.3.6/share/info -DCMAKE_INSTALL_MANDIR=/nix/store/hj2d8zyd3yjb5n6s5m1m1k22i5rbniwh-mujoco-2.3.6/share/man -DCMAKE_INSTALL_OLDINCLUDEDIR=/nix/store/hj2d8zyd3yjb5n6s5m1m1k22i5rbniwh-mujoco-2.3.6/include -DCMAKE_INSTALL_INCLUDEDIR=/nix/store/hj2d8zyd3yjb5n6s5m1m1k22i5rbniwh-mujoco-2.3.6/include -DCMAKE_INSTALL_SBINDIR=/nix/store/hj2d8zyd3yjb5n6s5m1m1k22i5rbniwh-mujoco-2.3.6/sbin -DCMAKE_INSTALL_BINDIR=/nix/store/hj2d8zyd3yjb5n6s5m1m1k22i5rbniwh-mujoco-2.3.6/bin -DCMAKE_INSTALL_NAME_DIR=/nix/store/hj2d8zyd3yjb5n6s5m1m1k22i5rbniwh-mujoco-2.3.6/lib -DCMAKE_POLICY_DEFAULT_CMP0025=NEW -DCMAKE_OSX_SYSROOT= -DCMAKE_FIND_FRAMEWORK=LAST -DCMAKE_STRIP=/nix/store/d9fndiing52fkalp5knfalrvlb3isi6w-gcc-wrapper-12.2.0/bin/strip -DCMAKE_RANLIB=/nix/store/d9fndiing52fkalp5knfalrvlb3isi6w-gcc-wrapper-12.2.0/bin/ranlib -DCMAKE_AR=/nix/store/d9fndiing52fkalp5knfalrvlb3isi6w-gcc-wrapper-12.2.0/bin/ar -DCMAKE_C_COMPILER=gcc -DCMAKE_CXX_COMPILER=g++ -DCMAKE_INSTALL_PREFIX=/nix/store/hj2d8zyd3yjb5n6s5m1m1k22i5rbniwh-mujoco-2.3.6
mujoco> -- The C compiler identification is GNU 12.2.0
mujoco> -- The CXX compiler identification is GNU 12.2.0
mujoco> -- Detecting C compiler ABI info
mujoco> -- Detecting C compiler ABI info - done
mujoco> -- Check for working C compiler: /nix/store/d9fndiing52fkalp5knfalrvlb3isi6w-gcc-wrapper-12.2.0/bin/gcc - skipped
mujoco> -- Detecting C compile features
mujoco> -- Detecting C compile features - done
mujoco> -- Detecting CXX compiler ABI info
mujoco> -- Detecting CXX compiler ABI info - done
mujoco> -- Check for working CXX compiler: /nix/store/d9fndiing52fkalp5knfalrvlb3isi6w-gcc-wrapper-12.2.0/bin/g++ - skipped
mujoco> -- Detecting CXX compile features
mujoco> -- Detecting CXX compile features - done
mujoco> -- Performing Test CAN_BUILD_AVX
mujoco> -- Performing Test CAN_BUILD_AVX - Success
mujoco> -- Performing Test SUPPORTS_LLD
mujoco> -- Performing Test SUPPORTS_LLD - Failed
mujoco> -- Performing Test SUPPORTS_GOLD
mujoco> -- Performing Test SUPPORTS_GOLD - Success
mujoco> -- Performing Test SUPPORTS_GC_SECTIONS
mujoco> -- Performing Test SUPPORTS_GC_SECTIONS - Success
mujoco> -- mujoco::FindOrFetch: checking for targets in package `qhull`
mujoco> -- mujoco::FindOrFetch: checking for targets in package `qhull` - target `qhull` not defined.
mujoco> -- mujoco::FindOrFetch: Using FetchContent to retrieve `qhull`
mujoco> -- Looking for sys/types.h
mujoco> -- Looking for sys/types.h - found
mujoco> -- Looking for inttypes.h
mujoco> -- Looking for inttypes.h - found
mujoco> -- Looking for stddef.h
mujoco> -- Looking for stddef.h - found
mujoco> -- Looking for stdint.h
mujoco> -- Looking for stdint.h - found
mujoco> -- Check size of off_t
mujoco> -- Check size of off_t - done
mujoco> -- Looking for fseeko
mujoco> -- Looking for fseeko - found
mujoco> -- Looking for ftello
mujoco> -- Looking for ftello - found
mujoco> -- Looking for PRIdMAX
mujoco> -- Looking for PRIdMAX - found
mujoco> --
mujoco> -- ========== qhull Build Information ==========
mujoco> -- Build Version:                             8.1-alpha1
mujoco> -- Install Prefix (CMAKE_INSTALL_PREFIX):     /nix/store/hj2d8zyd3yjb5n6s5m1m1k22i5rbniwh-mujoco-2.3.6
mujoco> -- Binary Directory (BIN_INSTALL_DIR):        bin
mujoco> -- Library Directory (LIB_INSTALL_DIR):       lib
mujoco> -- Include Directory (INCLUDE_INSTALL_DIR):   include
mujoco> -- Documentation Directory (DOC_INSTALL_DIR): share/doc/qhull
mujoco> -- Man Pages Directory (MAN_INSTALL_DIR):     share/man/man1
mujoco> -- CMake Directory (CMAKE_INSTALL_DIR):       lib/cmake/QHull
mujoco> -- PkgConfig Directory (PKGCONFIG_INSTALL_DIR):lib/pkgconfig
mujoco> -- Build Type (CMAKE_BUILD_TYPE):             Release
mujoco> -- Build static libraries:                    ON
mujoco> -- Build shared library:                      OFF
mujoco> -- Use shared library for linking apps:       OFF
[... 398 lines cut ...]
mujoco> -- Installing: /nix/store/hj2d8zyd3yjb5n6s5m1m1k22i5rbniwh-mujoco-2.3.6/share/mujoco/model/cube/assets/blue_red_yellow.png
mujoco> -- Installing: /nix/store/hj2d8zyd3yjb5n6s5m1m1k22i5rbniwh-mujoco-2.3.6/share/mujoco/model/cube/assets/red.png
mujoco> -- Installing: /nix/store/hj2d8zyd3yjb5n6s5m1m1k22i5rbniwh-mujoco-2.3.6/share/mujoco/model/cube/assets/green_orange_white.png
[... cut ...]
mujoco> -- Installing: /nix/store/hj2d8zyd3yjb5n6s5m1m1k22i5rbniwh-mujoco-2.3.6/share/mujoco/model/cube/cube_3x3x3.xml
mujoco> -- Installing: /nix/store/hj2d8zyd3yjb5n6s5m1m1k22i5rbniwh-mujoco-2.3.6/share/mujoco/model/mug
mujoco> -- Installing: /nix/store/hj2d8zyd3yjb5n6s5m1m1k22i5rbniwh-mujoco-2.3.6/share/mujoco/model/mug/mug.obj
mujoco> -- Installing: /nix/store/hj2d8zyd3yjb5n6s5m1m1k22i5rbniwh-mujoco-2.3.6/share/mujoco/model/mug/mug.png
mujoco> -- Installing: /nix/store/hj2d8zyd3yjb5n6s5m1m1k22i5rbniwh-mujoco-2.3.6/share/mujoco/model/mug/mug.xml
  ```
</details>

saran-t, 2023-07-14:
You indicated that you built from source. As with all questions relating to performance, I'd strongly urge you to get baseline numbers from our official binary build first.

yuvaltassa, 2023-07-14:
Hi, thanks for your questions!

**Question 1:** I apologize, I believe the line in the paper is a (rather serious) mistake. "on a single CPU thread" should read "on a single CPU". Note however that real-time ratios scale with the timestep. The [humanoid model](https://github.com/deepmind/mujoco/blob/0ccfef731487af98e92367a9a96729034d6c9448/model/humanoid/humanoid.xml#L17) in the MuJoCo repository has a conservative **5ms** timestep, but the one used by the MJPC planner has a [timestep of **15ms**](https://github.com/deepmind/mujoco_mpc/blob/28bc73029c214dfae8ca3917f814adaf22b35cc4/mjpc/tasks/humanoid/stand/task.xml#L9) which leads to a total real-time ratio of **4000x** on an M1 Mac and **5400x** on your machine. Of course large timesteps are "dangerous" in the sense that the physics can sometimes diverge, but when used in the context of short planner trajectories (that handle divergence gracefully), this is not a problem. The humanoid *environment* controlled by the MJPC planner actually has the default timestep of **2ms**, for high stability and fidelity. Such are the benefits of asynchronous MPC, you can choose timesteps to suit your needs 🙂 

**Question 2:** I think I can see multiple issues in the timing breakdown. 
1. I find it curious that you have so many contacts. You say you have a quadruped on some terrain? How do you get to an average of 40 contacts / step? You have lots of geoms on the feet or something? Are you using a very dense height-field? (don't do that).
2. I find the constraint/contact ratio curious. I think you must be using 6D contacts? These are not free...
3. I find the 77% time spent in constraint resolution too high. Try switching solver and/or setting [jacobian = "sparse"](https://mujoco.readthedocs.io/en/latest/XMLreference.html#option-jacobian).
4. The 19% time spent in collision detection is also pretty high. This is something we can help with. In [version 2.3.3](https://mujoco.readthedocs.io/en/latest/changelog.html#version-2-3-3-march-20-2023) we introduced a new mid-phase pruning stage that significantly enhanced collision detection speed ([credit](https://github.com/deepmind/mujoco/commit/70959c1a2ccf4169b30e31eec596c183e167d8c3) to @quagla!). The speedup was made possible because a user sent us [models](https://www.youtube.com/watch?v=e0babIM8hBo) that we could benchmark against. The same thing that was done for individual geoms could and should be done with heightfields: use a [binary tree search](https://github.com/deepmind/mujoco/blob/0ccfef731487af98e92367a9a96729034d6c9448/src/engine/engine_collision_driver.c#L217) instead of [nested for loops](https://github.com/deepmind/mujoco/blob/0ccfef731487af98e92367a9a96729034d6c9448/src/engine/engine_collision_convex.c#L711). Unfortunately we ourselves do not use heightfields much and do not have example benchmark models. If you send us your model we can see if we can add tree-based pruning to heightfields. But I would really ask that you first fix whatever you did that made constraint resolution so slow. 

Very generally, MuJoCo has a built in profiler that is visible in the `simulate` app GUI (F3 key), that will show you how different choices affect compute speed (this is printed out in `testspeed`). Use it!

Allow me to apologize again for the typo in the paper, it is rather embarrassing 😬

breakds, 2023-07-15:
Really appreciate your fast response!

>  I suspect you're missing the INTERPROCEDURAL_OPTIMIZATION flag.)

Thanks! This did boost the realtime factor from 1833 to 2200. Is this close to what you have @saran-t ?

@yuvaltassa Thanks a lot for the clarification on the MuJoCo MPC paper. The paper is very nice and the various benefits of using MuJoCo for planner rollout looks like a good fit.

On Question 2:

> I find it curious that you have so many contacts. You say you have a quadruped on some terrain? How do you get to an average of 40 contacts / step? You have lots of geoms on the feet or something? Are you using a very dense height-field? (don't do that).

Unfortunately yes, we do have a very dense height field, with 1600 x 1600 vertex in 12m x 12m ground. The reason we have this is that since we are training a policy with model-free reinforcement learning, we need a relatively dense terrain to encourage the learned locomotion policy to lift feet hand enough. 

> I find the constraint/contact ratio curious. I think you must be using 6D contacts? These are not free...

Yes we set `condim="6"` on all feet based on code from [mujoco-menagerie](https://github.com/deepmind/mujoco_menagerie/blob/main/unitree_go1/go1.xml#L62). When lower this down to "3", the simulation is indeed 20% faster. I am not sure whether this will affect sim2real at this moment.

> I find the 77% time spent in constraint resolution too high. Try switching solver and/or setting [jacobian = "sparse"](https://mujoco.readthedocs.io/en/latest/XMLreference.html#option-jacobian).

Using "CG" solver did help double the simulation performance. Setting "jacobian" to "sparse" does not seem to further improve the performance.

> In [version 2.3.3](https://mujoco.readthedocs.io/en/latest/changelog.html#version-2-3-3-march-20-2023) we introduced a new mid-phase pruning stage that significantly enhanced collision detection [...]

I noticed this when 2.3.3 was released (really appreciate your effort on actively developing it and regularly making new releases!). The algorithm was clever. I understand that currently it does not seem to work on heightfields - but in the end we just want some terrain with irregular ups and downs. Do you think it would be better if instead we create the terrain with a bunch of cubes on the ground which cannot move?

 > Very generally, MuJoCo has a built in profiler that is visible in the simulate app GUI (F3 key)
Nice. This is very useful. Thank you!

Thanks again for your time and patiently answer the questions!

breakds, 2023-07-18:
So the only unresolved problem (and the biggest) is that we have a rather dense `hfield`. We need this to allow the training to get a policy where the quadruped can lift its legs higher.

In terms of performance, instead of using `hfield`, is it going to be faster if we replace the terrain with (a huge amount of) blocks, so that the binary tree search trick can kick in?

breakds, 2023-07-18:
Another trick I can think of is to place 20 robots on a single terrain (i.e. a model with 20 robots) for training. I can carefully assign the `contype` and `conaffinity` to make those robots "disjoint" for conflict detection purpose. This seems to increase the "per-robot" performance a lot, but I think this might not be recommended and can even have harmful side effects that I am not aware of.

yuvaltassa, 2023-07-18:
Replacing the hfield with primitives should help, though I suspect that you might be able to get away with a somewhat less dense hfield.

I can't think of side-effects of making a multi-robot scene, this should be fine.

Closing for now but will keep following this thread.

closed 2023-07-18 by yuvaltassa