google-deepmind/mujoco #989: Making model smaller cause WARNING: Nan, Inf or huge value in QACC at DOF <xx>. The simulation is unstable.

muchvo, 2023-07-17:
**How to ask for help**

First, read our quick guide to
[asking good questions](https://github.com/deepmind/mujoco#asking-questions).
Below is a template for you to use:

Hi,

I'm a (student) and I'm trying to use MuJoCo for RL.

I'm looking for some help with modeling.

Here is a model which explains my question:

<details>
  <summary>minimal XML</summary>

This is the origin model:

```XML
<!--
Copyright 2022-2023 OmniSafe Team. All Rights Reserved.

 Licensed under the Apache License, Version 2.0 (the "License");
 you may not use this file except in compliance with the License.
 You may obtain a copy of the License at

     http://www.apache.org/licenses/LICENSE-2.0

 Unless required by applicable law or agreed to in writing, software
 distributed under the License is distributed on an "AS IS" BASIS,
 WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
 See the License for the specific language governing permissions and
 limitations under the License.
-->

<mujoco model="doggo">
  <option timestep="0.01"/>
  <compiler angle="radian"/>
  <size njmax="3000" nconmax="1000"/>
  <default>
    <joint limited="true" stiffness="0.01"/>
    <geom condim="6" size="0.032" rgba="0.8 0.6 0.4 1" type="capsule" density="5"/>
    <motor ctrllimited="true" ctrlrange="-1.0 1.0" gear=".0125"/>
    <site size="0.036" type="sphere" rgba="1 0 0 0"/>
  </default>
  <worldbody>
    <geom name="floor" size="5 5 0.1" type="plane" condim="6"/>
    <body name="agent" pos="0 0 0.22">
      <camera name="vision" pos=".125 0 .2" xyaxes="0 -1 0 .4 0 1" fovy="100"/>
      <!-- <geom name="cam" pos=".15 0 .2" type="sphere" size="0.005"/> -->
      <site name="agent" pos="0 0 0"/>
      <geom name="agent" pos="0 0 0" size="0.075" fromto="0 0 0 .2 0 0" type="cylinder" density="0.5"/>
      <freejoint name="agent"/>

      <body name="leg_1" pos="0 0 0">
        <geom fromto=".1 0.0 0.0 0.2 0.1 0.0" name="aux_1" rgba="0.0039 0.1529 0.3961 1"/>
        <body name="aux_1" pos="0.2 0.1 0">
          <joint axis="0 0 1" name="hip_1_z" pos="0.0 0.0 0.0" range="-0.174533 0.523599" type="hinge" springref="0"/>
          <joint axis="0 1 0" name="hip_1_y" pos="0.0 0.0 0.0" range="-1.308998 0.2618" type="hinge" springref="-0.174533"/>
          <geom fromto="0.0 0.0 0.0 0.098 0.0566 -.05" name="hip_1" rgba="0.0039 0.1529 0.3961 1"/>
          <body pos="0.098 0.0566 -.05">
            <joint axis="-.5 .866 0" name="ankle_1" pos="0.0 0.0 0.0" range="-1.308998 0" type="hinge" springref="-0.349066"/>
            <geom fromto="0.0 0.0 0.0 -0.1176 -0.0679 -.1" name="ankle_1"/>
            <site name="ankle_1a" pos="0 0 0"/>
            <site name="ankle_1b" pos="-0.1176 -0.0679 -.1"/>
          </body>
        </body>
      </body>
      <body name="leg_4" pos="0 0 0">
[... 256 lines cut ...]
    <subtreeangmom body="agent" name="subtreeangmom"/>
  </sensor>
  <actuator>
    <motor name="hip_1_z" joint="hip_1_z"/>
    <motor name="hip_2_z" joint="hip_2_z"/>
    <motor name="hip_3_z" joint="hip_3_z"/>
    <motor name="hip_4_z" joint="hip_4_z"/>
    <motor name="hip_1_y" joint="hip_1_y"/>
    <motor name="hip_2_y" joint="hip_2_y"/>
    <motor name="hip_3_y" joint="hip_3_y"/>
    <motor name="hip_4_y" joint="hip_4_y"/>
    <motor name="ankle_1" joint="ankle_1"/>
    <motor name="ankle_2" joint="ankle_2"/>
    <motor name="ankle_3" joint="ankle_3"/>
    <motor name="ankle_4" joint="ankle_4"/>
  </actuator>
</mujoco>
```

</details>

Here is a screenshot / video, illustrating my question:
[image]
**To reproduce it**:
Drag this model into the simulator window and click `1.0` at `hip_1_y` then quickly switch it to `-1.0`.

yuvaltassa, 2023-07-17:
Add damping and/or armature to the joints.

closed 2023-07-17 by yuvaltassa

muchvo, 2023-07-17:
> Add damping and/or armature to the joints.

I have tried due to related issues, but it seems hard to recover it to act like before. Did it mean I should make more effort on tuning these two parameters and exclude other possible mistakes? And thanks a lot for your rapid reply.

yuvaltassa, 2023-07-17:
Wait, are you trying to have the two models behave identically?

If yes then you need to use dimensional analysis to scale all the relevant quantities appropriately.

yuvaltassa, 2023-07-17:
I see you scaled the gains, but you didn't scale e.g. the stiffness.

muchvo, 2023-07-17:
> Wait, are you trying to have the two models behave identically?
> 
> If yes then you need to use dimensional analysis to scale all the relevant quantities appropriately.

Yeah, ideally, I want to reach this, but at one level which cannot be distinguished by a human is also great.

muchvo, 2023-07-17:
> I see you scaled the gains, but you didn't scale e.g. the stiffness.

Well, due to the change in mass, so I try to change this value.

muchvo, 2023-07-17:
Okay, I will work on it. thanks a lot for your careful and on-time reply.