google-deepmind/mujoco #991: Caught egl initial wrong when use multiprocessing

TypeFloat, 2023-07-19:
I'm new in mujoco. And I'm using mujoco for reinforcement research. I have designed an environmnet using mujoco which needs render to get version data. I use ssh connect with my server whose operator system is ubuntu 20.04 and use egl as my render backend. At the beginning, it runs with any problem. To accelerate the data collection process, I try to use multi-processing. I use python(py3.8 and mujoco python bindings is 2.3 which is the latest version) and once I run multi-process, I caught 

AttributeError: 'GLContext' object has no attribute '_context'.

I'm sure there is no problem with my code because it works well in another computer whose os is windows.

saran-t, 2023-07-19:
Please post a full stack trace.

TypeFloat, 2023-07-20:
Exception ignored in: <function GLContext.__del__ at 0x7f465c83ae50>
Traceback (most recent call last):
  File "/home/liuzekun/anaconda3/envs/robot/lib/python3.8/site-packages/mujoco/egl/__init__.py", line 130, in __del__
    self.free()
  File "/home/liuzekun/anaconda3/envs/robot/lib/python3.8/site-packages/mujoco/egl/__init__.py", line 120, in free
    if self._context:
AttributeError: 'GLContext' object has no attribute '_context'

When I run 20 process paralley of mujoco to collect trainning data，every process caught this problem. But if I run process one by one, it runs collectedly.
Interesting, I ran a single process yesterday, and it works correctly till now. But when I try to reproduct the bug, the process also caught GLContext error.

HeegerGao, 2023-07-21:
Same problem! Is there any progress?

TypeFloat, 2023-09-15:
It's strange that I change the method to create mujoco process from multiprocessing.Process to multiprocessing.Pool, it runs successfully. Anyway, this issue can be closed.

TypeFloat, 2023-09-15:
I have tried for long time. When I run my program in m1 chip macOS, it runs successfully, but it occur "egl init wrong" on x86 chip ubuntu.

kevinzakka, 2023-12-15:
Hi @TypeFloat, can you check if the latest release (3.1.0) fixes your issue? We've added a `close` method that you should call to free up resources associated with the renderer object. You can use it as follows:

```python
# Option 1.
renderer = Renderer(...)
# Use renderer.
renderer.close()

# Option 2.
with Renderer(...) as renderer:
# Use renderer.
```

TypeFloat, 2023-12-15:
I appreciate your attention, but unfortunately, the close function doesn't work. My requirement was to enable multiple renderers at the same time(because I use multiprocessing), not turn them on and off in sequence, so freeing resources didn't work. Below is a simple example I implemented, reproducing my problem on my server (ubuntu20.04 OS, python=3.9, mujoco=3.1.0).

import mujoco

def para_func():
    xml_path = "assets/robot-0.xml"
    model = mujoco.MjModel.from_xml_path(xml_path)
    data = mujoco.MjData(model)
    with mujoco.Renderer(model, 960, 1280) as r:
        r.update_scene(data)
        r.render()

if __name__ == "__main__":
    import multiprocessing

    with multiprocessing.Pool(3) as pool:
        for i in range(3):
            pool.apply_async(para_func, error_callback=print)
        pool.close()
        pool.join()

Traceback is below:

Traceback (most recent call last):
  File "/home/*/miniconda3/envs/mujoco/lib/python3.9/site-packages/mujoco/egl/__init__.py", line 130, in __del__

    self.free()
  File "/home/*/miniconda3/envs/mujoco/lib/python3.9/site-packages/mujoco/egl/__init__.py", line 120, in free
    if self._context:
AttributeError: 'GLContext' object has no attribute '_context'
Exception ignored in: <function Renderer.__del__ at 0x7f6f90215790>
Traceback (most recent call last):
  File "/home/*/miniconda3/envs/mujoco/lib/python3.9/site-packages/mujoco/renderer.py", line 327, in __del__
    self.close()
  File "/home/*/miniconda3/envs/mujoco/lib/python3.9/site-packages/mujoco/renderer.py", line 315, in close
    if self._gl_context:
AttributeError: 'Renderer' object has no attribute '_gl_context'

kevinzakka, 2024-01-03:
Hi @TypeFloat, can you try adding `multiprocessing.set_start_method('spawn')` right after the import? It solves my multiprocessing issue.

TypeFloat, 2024-01-03:
@kevinzakka Amazing, thanks for you helping! The code works well on ubuntu.

closed 2024-01-03 by TypeFloat