google-deepmind/mujoco #996: Problems encountered when using robotiq 2f-85 for gripping simulation

wangrp1997, 2023-07-22:
https://github.com/deepmind/mujoco/assets/58319077/d1a77858-6bf8-490c-a49e-4d58b2381433

Hi,

I'm a student and I'm trying to use MuJoCo for academic research.

I'm looking for some help implementing a grasping simulation in MuJoCo.

When I use the robotiq_2f85 gripper in mujoco_menagerie for object grasping simulation, stable gripping and lifting cannot be achieved, there seems to be a collision between the gripper and the object that is not properly defined.

I just control the position of the jaws and apply an external force on the drive joint, code show as below:

def init_controller(model,data):
    ...
    data.ctrl[6] = 90
    data.qfrc_applied[7] = 1
    data.qfrc_applied[11] = 1
    ...

Then this is the xml code of the object：
```XML
<body name="grabbable_object" pos="-0.134 0.79213164 0.3">
    <geom type="box" size="0.03 0.03 0.1" rgba="1 0 0 1" friction="1 1 1"
        solimp="0.95 0.99 0.001" solref="0.004 1" condim="3" priority="0"/>
    <joint armature="0" axis="0 0 1" damping="0" limited="false" name="box_1_z" pos="0 0 0" stiffness="1" type="slide"/>
    <inertial pos="0 0 0" mass="0.01" diaginertia="1.98e-5 1.98e-5 5.45e-6"/>
</body>
```

Here is a screenshot / video, illustrating my question:

wangrp1997, 2023-07-22:
Thanks in advance.

yuvaltassa, 2023-12-10:
I believe this is now fixed at HEAD, see [this comment](https://github.com/google-deepmind/mujoco/issues/786#issuecomment-1849009946). 

Closing for now, let us know here if this did not solve your problem.

closed 2023-12-10 by yuvaltassa