google-deepmind/mujoco #997: force and torque sensor

jheyeo222, 2023-07-24:
Hi,

I'm a student and I'm trying to use MuJoCo for a project.

I received help previously on collision and contact error #966, and I'm extremely grateful for your assistance.

I'm currently considering how to obtain force and torque values. I've experimented with two methods: using `mj_contactForce` and utilizing force and torque sensors. For my model, I've adapted the Franka Emika Panda.XML from the MuJoCo Menagerie.

I have some questions regarding these methods.

1. `void mj_contactForce(const mjModel* m, const mjData* d, int id, mjtNum result[6]);`

Firstly, as I'm coding in Python, I've implemented mj_contactForce like so: `mj.mj_contactForce(model, data, 0, force_and_torque);`

Additionally, I've defined the robot's XML in the following :
```
<geom type="mesh" contype="0" conaffinity="0" group="2"/>
.
.
.
.

<geom name="wall" type="box" size="0.1 1.5 1.5" pos="0.7 0 0" rgba="0.5 0.5 0.5 1" contype="1" conaffinity="1"/>
```

[image]

`void mj_contactForce(const mjModel* m, const mjData* d, int id, mjtNum result[6]);`

1-1
Could you please clarify what exactly the 'id' in mj_contactForce refers to, and how it should be properly used? An example would be greatly appreciated.

1-2
As shown in the figure, why are the force values measured but not the torque values?

1-3
It appears that mj_contactForce measures the force and torque values based on the world frame. Could you please confirm if this is correct?

2. torque sensor and force sensor

The sensors have been coded as follows:

```
                    <geom mesh="link7_5" material="black" class="visual"/>
                    <geom mesh="link7_6" material="black" class="visual"/>
                    <geom mesh="link7_7" material="white" class="visual"/>
                    <geom mesh="link7_c" class="collision"/>
                    <site name="attachment_site" pos="0 0 0.107" />
                    <site name="tip" size="0.1" pos="0 0 0"/>

                  </body>
                </body>
              </body>
            </body>
          </body>
        </body>
      </body>
    </body>
  </worldbody>
  <sensor>
    <framepos objtype="site" objname="tip"/>
    <force site ='tip'/>
    <torque site='tip' />
  </sensor>
```

2-1 
The measurements seem to be working well, but it appears that the force and torque are measured based on the coordinate system of the immediately preceding link, rather than the world frame coordinates. Is this correct?

2-2 
I would like to obtain values based on the world coordinate system. Should I manually convert these using a homogeneous transform, or is there a specific function available for this purpose?

yuvaltassa, 2023-07-24:
> 1-1 Could you please clarify what exactly the 'id' in mj_contactForce refers to, and how it should be properly used? An example would be greatly appreciated.

this is the id of the contact. you want something like

```py
for id,c in enumerate(data.contact):
  mujoco.mj_contactForce(.., id, ..)
```
There is an example of this in the [tutorial](https://github.com/deepmind/mujoco#getting-started).

> 1-2 As shown in the figure, why are the force values measured but not the torque values?

These are the forces at the contact. By default contacts are 3D (only normal and and tangential friction). Torsional and rolling friction needs to be [enabled](https://mujoco.readthedocs.io/en/latest/XMLreference.html#body-geom-condim).

> 1-3 It appears that mj_contactForce measures the force and torque values based on the world frame. Could you please confirm if this is correct?

No, it is in the frame of the contact (normal, tangent, tangent, torsion, rolling, rolling). This is [documented](https://mujoco.readthedocs.io/en/latest/APIreference/APIfunctions.html#mj-contactforce).

> 2. torque sensor and force sensor
>
> The sensors have been coded as follows:
>
> ```
>                     <geom mesh="link7_5" material="black" class="visual"/>
>                     <geom mesh="link7_6" [...]

It is in the frame of the associated site, as [documented](https://mujoco.readthedocs.io/en/latest/XMLreference.html#sensor-force).

> 2-2 I would like to obtain values based on the world coordinate system. Should I manually convert these using a homogeneous transform, or is there a specific function available for this purpose?

You can use [`mj_local2global`](https://mujoco.readthedocs.io/en/latest/APIreference/APIfunctions.html#mj-local2global).

closed 2023-07-24 by yuvaltassa

jheyeo222, 2023-07-25:
> > 1-1 Could you please clarify what exactly the 'id' in mj_contactForce refers to, and how it should be properly used? An example would be greatly appreciated.
> 
> this is the id of the contact. you want something like
> 
> ```python
> for id,c in enumerate(data.contact):
>   mujoco.mj_contactForce(.., id, ..)
> ```
> 
> There is an example of this in the [tutorial](https://github.com/deepmind/mujoco#getting-started).
> 
> > 1-2 As shown in the figure, why are the force values measured but not the torque values?
> 
> These are the forces at the contact. By default contacts are 3D (only normal and and tangential friction). Torsional and rolling friction needs to be [enabled](https://mujoco.readthedocs.io/en/latest/XMLreference.html#body-geom-condim).
> 
> > 1-3 It appears that mj_contactForce measures the force and torque values based on the world frame. Could you please confirm if this is correct?
> 
> No, it is in the frame of the contact (normal, tangent, tangent, torsion, rolling, rolling). This is [documented](https://mujoco.readthedocs.io/en/latest/APIreference/APIfunctions.html#mj-contactforce).
> 
> > 2. torque sensor and force sensor
> > 
> > The sensors have been coded as follows:
> > ```
> >                     <geom mesh="link7_5" material="black" class="visual"/>
> >                     <geom mesh="link7_6" material="black" class="visual"/>
> >                     <geom mesh="link7_7" material="white" class="visual"/>
> >                     <geom mesh="link7_c" class="collision"/>
> >                     <site name="attachment_site" pos="0 0 0.107" />
> >                     <site name="tip" size="0.1" pos="0 0 0"/>
> > 
> > 
> >                   </body>
> >                 </body>
> >               </body>
> >             </body>
> >           </body>
> >         </body>
> >       </body>
> >     </body>
> >   </worldbody>
> >   <sensor>
> >     <framepos objtype="site" objname="tip"/>
> >     <force site ='tip'/>
> >     <torque site='tip' />
> >   </sensor>
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > 2-1 The measurements seem to be working well, but it appears that the force and torque are measured based on the coordinate system of the immediately preceding link, rather than the world frame coordinates. Is this correct?
> 
> It is in the frame of the associated site, as [documented](https://mujoco.readthedocs.io/en/latest/XMLreference.html#sensor-force).
> 
> > 2-2 I would like to obtain values based on the world coordinate system. Should I manually convert these using a homogeneous transform, or is there a specific function available for this purpose?
> 
> You can use [`mj_local2global`](https://mujoco.readthedocs.io/en/latest/APIreference/APIfunctions.html#mj-local2global).

Upon experimenting with the function "void mj_contactForce(const mjModel* m, const mjData* d, int id, mjtNum result[6]);", I've discovered that the 'id' is used to assign identifiers to each contact force when there are multiple. This insight has been incredibly useful. I'm truly grateful for your assistance.

yuvaltassa, 2024-02-23:
Just multiply by the transposed xmat, that's all there is to it.