google-deepmind/mujoco #1004: v2.3.7 Python installation runtime error on Intel-based mac

guyazran, 2023-08-02:
I am attempting to install the mujoco python package version 2.3.7.

My machine:
Model: 2017 15-inch MacBook Pro
CPU: Intel Core i7-7920HQ CPU @ 3.10GHz
OS: MacOS 10.13.4.1 (Ventura)
python version: 3.8

I used a fresh python environment (with conda) and ran the following command:
```shell
pip install --no-cache-dir mujoco==2.3.7
```

Installation fails with a runtime error `MUJOCO_PATH environment variable is not set`. For comparison, version 2.3.6 installs successfully without any errors or warnings. 

Below is the full output for the installation command:
```
Collecting mujoco
  Downloading mujoco-2.3.7.tar.gz (593 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 593.3/593.3 kB 896.0 kB/s eta 0:00:00
  Preparing metadata (setup.py) ... done
Collecting absl-py (from mujoco)
  Downloading absl_py-1.4.0-py3-none-any.whl (126 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 126.5/126.5 kB 6.6 MB/s eta 0:00:00
Collecting glfw (from mujoco)
  Obtaining dependency information for glfw from https://files.pythonhosted.org/packages/a4/d7/cc1807354298061c5550aa19b2b52a32975b038c638e96afab9b5f1857b0/glfw-2.6.2-py2.py27.py3.py30.py31.py32.py33.py34.py35.py36.py37.py38-none-macosx_10_6_intel.whl.metadata
  Downloading glfw-2.6.2-py2.py27.py3.py30.py31.py32.py33.py34.py35.py36.py37.py38-none-macosx_10_6_intel.whl.metadata (5.4 kB)
Collecting numpy (from mujoco)
  Obtaining dependency information for numpy from https://files.pythonhosted.org/packages/11/10/943cfb579f1a02909ff96464c69893b1d25be3731b5d3652c2e0cf1281ea/numpy-1.24.4-cp38-cp38-macosx_10_9_x86_64.whl.metadata
  Downloading numpy-1.24.4-cp38-cp38-macosx_10_9_x86_64.whl.metadata (5.6 kB)
Collecting pyopengl (from mujoco)
  Obtaining dependency information for pyopengl from https://files.pythonhosted.org/packages/99/48/00e31747821d3fc56faddd00a4725454d1e694a8b67d715cf20f531506a5/PyOpenGL-3.1.7-py3-none-any.whl.metadata
  Downloading PyOpenGL-3.1.7-py3-none-any.whl.metadata (3.2 kB)
Downloading glfw-2.6.2-py2.py27.py3.py30.py31.py32.py33.py34.py35.py36.py37.py38-none-macosx_10_6_intel.whl (96 kB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 96.9/96.9 kB 6.5 MB/s eta 0:00:00
Downloading numpy-1.24.4-cp38-cp38-macosx_10_9_x86_64.whl (19.8 MB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 19.8/19.8 MB 24.1 MB/s eta 0:00:00
Downloading PyOpenGL-3.1.7-py3-none-any.whl (2.4 MB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 2.4/2.4 MB 55.5 MB/s eta 0:00:00
Building wheels for collected packages: mujoco
  Building wheel for mujoco (setup.py) ... error
  error: subprocess-exited-with-error
  
  × python setup.py bdist_wheel did not run successfully.
  │ exit code: 1
  ╰─> [65 lines of output]
      running bdist_wheel
      running build
      running build_py
      creating build
      creating build/lib.macosx-10.9-x86_64-cpython-38
      creating build/lib.macosx-10.9-x86_64-cpython-38/mujoco
      copying mujoco/viewer.py -> build/lib.macosx-10.9-x86_64-cpython-38/mujoco
      copying mujoco/renderer.py -> build/lib.macosx-10.9-x86_64-cpython-38/mujoco
      copying mujoco/renderer_test.py -> build/lib.macosx-10.9-x86_64-cpython-38/mujoco
      copying mujoco/__init__.py -> build/lib.macosx-10.9-x86_64-cpython-38/mujoco
      copying mujoco/rollout.py -> build/lib.macosx-10.9-x86_64-cpython-38/mujoco
      copying mujoco/rollout_test.py -> build/lib.macosx-10.9-x86_64-cpython-38/mujoco
      copying mujoco/bindings_test.py -> build/lib.macosx-10.9-x86_64-cpython-38/mujoco
      copying mujoco/viewer_test.py -> build/lib.macosx-10.9-x86_64-cpython-38/mujoco
      copying mujoco/render_test.py -> build/lib.macosx-10.9-x86_64-cpython-38/mujoco
      copying mujoco/gl_context.py -> build/lib.macosx-10.9-x86_64-cpython-38/mujoco
      creating build/lib.macosx-10.9-x86_64-cpython-38/mujoco/cgl
      copying mujoco/cgl/cgl.py -> build/lib.macosx-10.9-x86_64-cpython-38/mujoco/cgl
      copying mujoco/cgl/__init__.py -> build/lib.macosx-10.9-x86_64-cpython-38/mujoco/cgl
      creating build/lib.macosx-10.9-x86_64-cpython-38/mujoco/egl
      copying mujoco/egl/__init__.py -> build/lib.macosx-10.9-x86_64-cpython-38/mujoco/egl
      copying mujoco/egl/egl_ext.py -> build/lib.macosx-10.9-x86_64-cpython-38/mujoco/egl
      creating build/lib.macosx-10.9-x86_64-cpython-38/mujoco/osmesa
      copying mujoco/osmesa/__init__.py -> build/lib.macosx-10.9-x86_64-cpython-38/mujoco/osmesa
      creating build/lib.macosx-10.9-x86_64-cpython-38/mujoco/glfw
      copying mujoco/glfw/__init__.py -> build/lib.macosx-10.9-x86_64-cpython-38/mujoco/glfw
      running build_ext
      Traceback (most recent call last):
        File "<string>", line 2, in <module>
        File "<pip-setuptools-caller>", line 34, in <module>
        File "/private/var/folders/0z/f5fklt153rs0bl5z3641bpb40000gn/T/pip-install-lrxvogn5/mujoco_71e2473094134ab18eea141c2b68b51b/setup.py", line 418, in <module>
          setup(**SETUP_KWARGS)
        File "/usr/local/Caskroom/miniconda/base/envs/dmc/lib/python3.8/site-packages/setuptools/__init__.py", line 107, in setup
          return distutils.core.setup(**attrs)
        File "/usr/local/Caskroom/miniconda/base/envs/dmc/lib/python3.8/site-packages/setuptools/_distutils/core.py", line 185, in setup
          return run_commands(dist)
        File "/usr/local/Caskroom/miniconda/base/envs/dmc/lib/python3.8/site-packages/setuptools/_distutils/core.py", line 201, in run_commands
          dist.run_commands()
        File "/usr/local/Caskroom/miniconda/base/envs/dmc/lib/python3.8/site-packages/setuptools/_distutils/dist.py", line 969, in run_commands
          self.run_command(cmd)
        File "/usr/local/Caskroom/miniconda/base/envs/dmc/lib/python3.8/site-packages/setuptools/dist.py", line 1234, in run_command
          super().run_command(command)
        File "/usr/local/Caskroom/miniconda/base/envs/dmc/lib/python3.8/site-packages/setuptools/_distutils/dist.py", line 988, in run_command
          cmd_obj.run()
        File "/usr/local/Caskroom/miniconda/base/envs/dmc/lib/python3.8/site-packages/wheel/bdist_wheel.py", line 325, in run
          self.run_command("build")
        File "/usr/local/Caskroom/miniconda/base/envs/dmc/lib/python3.8/site-packages/setuptools/_distutils/cmd.py", line 318, in run_command
          self.distribution.run_command(command)
        File "/usr/local/Caskroom/miniconda/base/envs/dmc/lib/python3.8/site-packages/setuptools/dist.py", line 1234, in run_command
          super().run_command(command)
        File "/usr/local/Caskroom/miniconda/base/envs/dmc/lib/python3.8/site-packages/setuptools/_distutils/dist.py", line 988, in run_command
          cmd_obj.run()
        File "/usr/local/Caskroom/miniconda/base/envs/dmc/lib/python3.8/site-packages/setuptools/_distutils/command/build.py", line 131, in run
          self.run_command(cmd_name)
        File "/usr/local/Caskroom/miniconda/base/envs/dmc/lib/python3.8/site-packages/setuptools/_distutils/cmd.py", line 318, in run_command
          self.distribution.run_command(command)
        File "/usr/local/Caskroom/miniconda/base/envs/dmc/lib/python3.8/site-packages/setuptools/dist.py", line 1234, in run_command
          super().run_command(command)
        File "/usr/local/Caskroom/miniconda/base/envs/dmc/lib/python3.8/site-packages/setuptools/_distutils/dist.py", line 988, in run_command
          cmd_obj.run()
        File "/private/var/folders/0z/f5fklt153rs0bl5z3641bpb40000gn/T/pip-install-lrxvogn5/mujoco_71e2473094134ab18eea141c2b68b51b/setup.py", line 156, in run
          self._mujoco_framework_path) = self._find_mujoco()
        File "/private/var/folders/0z/f5fklt153rs0bl5z3641bpb40000gn/T/pip-install-lrxvogn5/mujoco_71e2473094134ab18eea141c2b68b51b/setup.py", line 170, in _find_mujoco
          raise RuntimeError(
      RuntimeError: MUJOCO_PATH environment variable is not set
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for mujoco
  Running setup.py clean for mujoco
Failed to build mujoco
ERROR: Could not build wheels for mujoco, which is required to install pyproject.toml-based projects
```

saran-t, 2023-08-02:
Could you run pip in verbose mode to see why it's not picking up the prebuilt wheel? You should be installing from `mujoco-2.3.7-cp38-cp38-macosx_11_0_x86_64.whl` rather than from source.

guyazran, 2023-08-02:
yes. I ran this command:
```shell
pip install -v --no-cache-dir mujoco==2.3.7
```
and got the following output:
```
Using pip 23.2.1 from /usr/local/Caskroom/miniconda/base/envs/dmc/lib/python3.8/site-packages/pip (python 3.8)
Collecting mujoco==2.3.7
  Downloading mujoco-2.3.7.tar.gz (593 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 593.3/593.3 kB 949.8 kB/s eta 0:00:00
  Running command python setup.py egg_info
  running egg_info
  creating /private/var/folders/0z/f5fklt153rs0bl5z3641bpb40000gn/T/pip-pip-egg-info-xhq6vrsc/mujoco.egg-info
  writing /private/var/folders/0z/f5fklt153rs0bl5z3641bpb40000gn/T/pip-pip-egg-info-xhq6vrsc/mujoco.egg-info/PKG-INFO
  writing dependency_links to /private/var/folders/0z/f5fklt153rs0bl5z3641bpb40000gn/T/pip-pip-egg-info-xhq6vrsc/mujoco.egg-info/dependency_links.txt
  writing requirements to /private/var/folders/0z/f5fklt153rs0bl5z3641bpb40000gn/T/pip-pip-egg-info-xhq6vrsc/mujoco.egg-info/requires.txt
  writing top-level names to /private/var/folders/0z/f5fklt153rs0bl5z3641bpb40000gn/T/pip-pip-egg-info-xhq6vrsc/mujoco.egg-info/top_level.txt
  writing manifest file '/private/var/folders/0z/f5fklt153rs0bl5z3641bpb40000gn/T/pip-pip-egg-info-xhq6vrsc/mujoco.egg-info/SOURCES.txt'
  reading manifest file '/private/var/folders/0z/f5fklt153rs0bl5z3641bpb40000gn/T/pip-pip-egg-info-xhq6vrsc/mujoco.egg-info/SOURCES.txt'
  reading manifest template 'MANIFEST.in'
  adding license file 'LICENSE'
  adding license file 'LICENSES_THIRD_PARTY.md'
  writing manifest file '/private/var/folders/0z/f5fklt153rs0bl5z3641bpb40000gn/T/pip-pip-egg-info-xhq6vrsc/mujoco.egg-info/SOURCES.txt'
  Preparing metadata (setup.py) ... done
Collecting absl-py (from mujoco==2.3.7)
  Downloading absl_py-1.4.0-py3-none-any.whl (126 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 126.5/126.5 kB 10.0 MB/s eta 0:00:00
Collecting glfw (from mujoco==2.3.7)
  Obtaining dependency information for glfw from https://files.pythonhosted.org/packages/a4/d7/cc1807354298061c5550aa19b2b52a32975b038c638e96afab9b5f1857b0/glfw-2.6.2-py2.py27.py3.py30.py31.py32.py33.py34.py35.py36.py37.py38-none-macosx_10_6_intel.whl.metadata
  Downloading glfw-2.6.2-py2.py27.py3.py30.py31.py32.py33.py34.py35.py36.py37.py38-none-macosx_10_6_intel.whl.metadata (5.4 kB)
  Link requires a different Python (3.8.17 not in: '>=3.9'): https://files.pythonhosted.org/packages/26/de/437a60a69f7fd0c79264530a97787b2ac7394616e3661236201518f8a47d/numpy-1.25.0rc1.tar.gz (from https://pypi.org/simple/numpy/) (requires-python:>=3.9)
  Link requires a different Python (3.8.17 not in: '>=3.9'): https://files.pythonhosted.org/packages/d0/b2/fe774844d1857804cc884bba67bec38f649c99d0dc1ee7cbbf1da601357c/numpy-1.25.0.tar.gz (from https://pypi.org/simple/numpy/) (requires-python:>=3.9)
  Link requires a different Python (3.8.17 not in: '>=3.9'): https://files.pythonhosted.org/packages/cf/7a/f68d1d658a0e68084097beb212fa9356fee7eabff8b57231cc4acb555b12/numpy-1.25.1.tar.gz (from https://pypi.org/simple/numpy/) (requires-python:>=3.9)
  Link requires a different Python (3.8.17 not in: '>=3.9'): https://files.pythonhosted.org/packages/a0/41/8f53eff8e969dd8576ddfb45e7ed315407d27c7518ae49418be8ed532b07/numpy-1.25.2.tar.gz (from https://pypi.org/simple/numpy/) (requires-python:>=3.9)
Collecting numpy (from mujoco==2.3.7)
  Obtaining dependency information for numpy from https://files.pythonhosted.org/packages/11/10/943cfb579f1a02909ff96464c69893b1d25be3731b5d3652c2e0cf1281ea/numpy-1.24.4-cp38-cp38-macosx_10_9_x86_64.whl.metadata
  Downloading numpy-1.24.4-cp38-cp38-macosx_10_9_x86_64.whl.metadata (5.6 kB)
Collecting pyopengl (from mujoco==2.3.7)
  Obtaining dependency information for pyopengl from https://files.pythonhosted.org/packages/99/48/00e31747821d3fc56faddd00a4725454d1e694a8b67d715cf20f531506a5/PyOpenGL-3.1.7-py3-none-any.whl.metadata
  Downloading PyOpenGL-3.1.7-py3-none-any.whl.metadata (3.2 kB)
Downloading glfw-2.6.2-py2.py27.py3.py30.py31.py32.py33.py34.py35.py36.py37.py38-none-macosx_10_6_intel.whl (96 kB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 96.9/96.9 kB 10.9 MB/s eta 0:00:00
Downloading numpy-1.24.4-cp38-cp38-macosx_10_9_x86_64.whl (19.8 MB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 19.8/19.8 MB 21.5 MB/s eta 0:00:00
Downloading PyOpenGL-3.1.7-py3-none-any.whl (2.4 MB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 2.4/2.4 MB 57.3 MB/s eta 0:00:00
Building wheels for collected packages: mujoco
  Running command python setup.py bdist_wheel
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build/lib.macosx-10.9-x86_64-cpython-38
  creating build/lib.macosx-10.9-x86_64-cpython-38/mujoco
  copying mujoco/viewer.py -> build/lib.macosx-10.9-x86_64-cpython-38/mujoco
  copying mujoco/renderer.py -> build/lib.macosx-10.9-x86_64-cpython-38/mujoco
  copying mujoco/renderer_test.py -> build/lib.macosx-10.9-x86_64-cpython-38/mujoco
  copying mujoco/__init__.py -> build/lib.macosx-10.9-x86_64-cpython-38/mujoco
  copying mujoco/rollout.py -> build/lib.macosx-10.9-x86_64-cpython-38/mujoco
  copying mujoco/rollout_test.py -> build/lib.macosx-10.9-x86_64-cpython-38/mujoco
  copying mujoco/bindings_test.py -> build/lib.macosx-10.9-x86_64-cpython-38/mujoco
  copying mujoco/viewer_test.py -> build/lib.macosx-10.9-x86_64-cpython-38/mujoco
  copying mujoco/render_test.py -> build/lib.macosx-10.9-x86_64-cpython-38/mujoco
  copying mujoco/gl_context.py -> build/lib.macosx-10.9-x86_64-cpython-38/mujoco
[... cut ...]
  Building wheel for mujoco (setup.py) ... error
  ERROR: Failed building wheel for mujoco
  Running setup.py clean for mujoco
  Running command python setup.py clean
  running clean
  removing 'build/lib.macosx-10.9-x86_64-cpython-38' (and everything under it)
  'build/bdist.macosx-10.9-x86_64' does not exist -- can't clean it
  'build/scripts-3.8' does not exist -- can't clean it
  removing 'build'
Failed to build mujoco
ERROR: Could not build wheels for mujoco, which is required to install pyproject.toml-based projects
```

saran-t, 2023-08-02:
It's still not verbose enough to say why pip decide not to pick the wheel (you can see that you're grabbing the .tar.gz sdist).

The build log seems to suggest that your Python installation is applying the macosx_10_9 platform tag. We've decided to only support macOS 11 and above since version 2.3.7. I note that you said you're using 13 so perhaps this is something that's specific to how conda configures the Python environment?

saran-t, 2023-08-02:
This seems related to https://github.com/conda-forge/python-feedstock/issues/445

Try the suggested environment variable workaround there, or just upgrade to Python 3.9 (or better still, Python 3.11) if you can.

guyazran, 2023-08-03:
Thank you @saran-t . The issue is indeed with conda python. When installing python from the default channels, the resulting implementation has an issue with some OS compatibility flags. I run the following command to echo the platform that python thinks it is in:
```shell
python -c "import platform; print(platform.platform())"
```
The output for the default conda implementation is `macOS-10.16-x86_64-i386-64bit` instead of the expected `macOS-13.4.1-x86_64-i386-64bit` (for MacOS 10.13.4.1). Upgrading to 3.9 (or even 3.11) does not help. I present below 3 solutions that enable installation of the latest mujoco package successfully on my machine.

### Solution 1: Env variable workaround
As is suggested in conda-forge/python-feedstock#445, there is an environment variable that can be set to work around this issue. Run `export SYSTEM_VERSION_COMPAT=0` before running the python/pip command. This immediately fixes the issue:
```
$ python -c "import platform; print(platform.platform())"
macOS-10.16-x86_64-i386-64bit
$ export SYSTEM_VERSION_COMPAT=0
$ python -c "import platform; print(platform.platform())"
macOS-13.4.1-x86_64-i386-64bit
```

### Solution 2: Use conda-forge python
The implementation of python found in the conda-forge channel does not suffer from this issue. Use the `-c conda-forge` option when installing python in the environment:
```
$ conda create -c conda-forge -n mj_env python=3.9
...
...
$ conda activate mj_env
$ python -c "import platform; print(platform.platform())"
macOS-13.4.1-x86_64-i386-64bit
```
**Note:** This only works with python version >=3.9. using version 3.8 results in the same issue (which can still be overcome with solution 1)

### Solution 3: Don't use conda
If you must use version 3.8 and you can't use the solution 1 workaround, you can simply use an official python implementation, e.g., using homebrew:
```
$ brew install python@3.8
$ python3.8 -m venv mj_venv
$ source mj_venv/bin/activate
$ python -c "import platform; print(platform.platform())"
macOS-13.4.1-x86_64-i386-64bit
```

closed 2023-08-03 by guyazran