google-deepmind/mujoco #1007: Camera configuration setting in MuJoCo viewer (python)

LewisChoi, 2023-08-03:
Hi,

I'm an engineer, and trying to use MuJoCo for a manipulator simulation.

I'm looking for some help with camera configuration setting in MuJoCo viewer. When using **glfw**, it was possible to configure the camera parameters . See the parameter settings below. 

Example on how to set camera configuration

    context = mj.MjrContext(model, mj.mjtFontScale.mjFONTSCALE_150.value)

    cam.azimuth = 90.0 ; cam.elevation = -15.0 ; cam.distance =  7.0
    cam.lookat =np.array([ 0.0 , 0.0 , 1 ])

...
...
...

in the loop. 

    viewport = mj.MjrRect(0, 0, viewport_width, viewport_height)

    mj.mjv_updateScene(model, data, opt, None, cam, mj.mjtCatBit.mjCAT_ALL.value, scene)
    mj.mjr_render(viewport, scene, context)

Would you help me to set the camera configuration? I was not able to find any examples. 

Thanks

aftersomemath, 2023-08-11:
Hi, 

Do you mean that you want to set the camera configuration of the camera shown by the Python viewer in the GUI window? Currently this is not possible, but it could be done. Recently I ran across a problem where we needed to do this.

yuvaltassa, 2023-08-11:
This should be pretty easy to add. Right, @saran-t ?

closed 2023-10-11 by saran-t