google-deepmind/mujoco #1008: Extremely low rendering fps in wsl2

mantle2048, 2023-08-04:
I installed mujoco on WSL2 via `pip install mujoco`. 

Then I ran the built-in model [humanoid100](https://github.com/deepmind/mujoco/blob/main/model/humanoid100/humanoid100.xml), but the rendering fps  is only 1-2. 

While using the Windows version, running the same model can reach 60 fps.

The following is my screenshot.

[image]

### PC info
```
CPU: Intel i7-10875H (16) @ 2.304GHz
GPU: NVIDIA GeForce RTX 2060
CUDA Version: 11.7
Driver Versino: 517.00
```

### Mujoco info
```
$ echo $PYOPENGL_PLATFORM
glx
```
```
[ins] In [2]: mujoco.__version__
Out[2]: '2.3.7
```

### WSL version
```
WSL version： 1.3.15.0
Core version： 5.15.90.4-1
WSLg version： 1.0.55
MSRDC version： 1.2.4419
Direct3D version： 1.608.2-61064218
DXCore version： 10.0.25880.1000-230602-1350.main
Windows version： 10.0.22621.1992
```

### glxinfo 
```
name of display: :0
display: :0  screen: 0
direct rendering: Yes
Extended renderer info (GLX_MESA_query_renderer):
    Vendor: Microsoft Corporation (0xffffffff)
    Device: D3D12 (NVIDIA GeForce RTX 2060) (0xffffffff)
    Version: 23.1.5
    Accelerated: yes
    Video memory: 14105MB
    Unified memory: no
    Preferred profile: core (0x1)
    Max core profile version: 4.2
    Max compat profile version: 4.2
    Max GLES1 profile version: 1.1
    Max GLES[23] profile version: 3.1
OpenGL vendor string: Microsoft Corporation
OpenGL renderer string: D3D12 (NVIDIA GeForce RTX 2060)
OpenGL core profile version string: 4.2 (Core Profile) Mesa 23.1.5 - kisak-mesa PPA
OpenGL core profile shading language version string: 4.20
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile

OpenGL version string: 4.2 (Compatibility Profile) Mesa 23.1.5 - kisak-mesa PPA
OpenGL shading language version string: 4.20
OpenGL context flags: (none)
OpenGL profile mask: compatibility profile

OpenGL ES profile version string: OpenGL ES 3.1 Mesa 23.1.5 - kisak-mesa PPA
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.10
```

yuvaltassa, 2023-08-04:
I'm not an expert but the `MESA` substring in your printout leads me to think that you are using software renderering.

mantle2048, 2023-08-04:
> I'm not an expert but the `MESA` substring in your printout leads me to think that you are using software renderering.

Thanks for your kind reply! 

Can you teach me which line indicates that I am using software rendering? (I'm a layman of `MESA`)

Because from my perspective, gpu does work when I render the model.

The following is the output of `nvidia-smi` before running mujoco.
```
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 515.67       Driver Version: 517.00       CUDA Version: 11.7     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|                               |                      |               MIG M. |
|===============================+======================+======================|
|   0  NVIDIA GeForce ...  On   | 00000000:01:00.0 Off |                  N/A |
| N/A   48C    P8     4W /  N/A |      0MiB /  6144MiB |      0%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+

+-----------------------------------------------------------------------------+
| Processes:                                                                  |
|  GPU   GI   CI        PID   Type   Process name                  GPU Memory |
|        ID   ID                                                   Usage      |
|=============================================================================|
|    0   N/A  N/A        43      G   /Xwayland                       N/A      |
+-----------------------------------------------------------------------------+
```

The output of `nvidia-smi` after running mujoco.

```
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 515.67       Driver Version: 517.00       CUDA Version: 11.7     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|                               |                      |               MIG M. |
|===============================+======================+======================|
|   0  NVIDIA GeForce ...  On   | 00000000:01:00.0 Off |                  N/A |
| N/A   51C    P8    10W /  N/A |    183MiB /  6144MiB |     20%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+

+-----------------------------------------------------------------------------+
| Processes:                                                                  |
|  GPU   GI   CI        PID   Type   Process name                  GPU Memory |
|        ID   ID                                                   Usage      |
|=============================================================================|
|    0   N/A  N/A        43      G   /Xwayland                       N/A      |
|    0   N/A  N/A      1139      G   /viewer.py                      N/A      |
+-----------------------------------------------------------------------------+
```

aftersomemath, 2023-08-11:
The ability of nvidia-smi to work does not mean mujoco is using the GPU to render.

The easiest way to tell if CPU or GPU is being used for rendering is to look at your CPU and GPU utilization while running mujoco. If the cpu is at 100% in windows task manager while the GPU is at 5%, software rendering is being used.

mantle2048, 2023-08-12:
@aftersomemath Thanks for your guidance ! I then checked the CPU utilization.

The following is the output of `htop` and `windows task manager` (forgive me that my system is in Chinese and CPU utilization rate is in the bottom left of the image). 

It seems that CPU utilization is not high, and GPU is indeed working (11% utilization).

[image]

[image]

[image]

 The output of `glxinfo -B`.
```
name of display: :0
display: :0  screen: 0
direct rendering: Yes
Extended renderer info (GLX_MESA_query_renderer):
    Vendor: Microsoft Corporation (0xffffffff)
    Device: D3D12 (NVIDIA GeForce RTX 2060) (0xffffffff)
    Version: 23.1.5
    Accelerated: yes
    Video memory: 14105MB
    Unified memory: no
    Preferred profile: core (0x1)
    Max core profile version: 4.2
    Max compat profile version: 4.2
    Max GLES1 profile version: 1.1
    Max GLES[23] profile version: 3.1
OpenGL vendor string: Microsoft Corporation
OpenGL renderer string: D3D12 (NVIDIA GeForce RTX 2060)
OpenGL core profile version string: 4.2 (Core Profile) Mesa 23.1.5 - kisak-mesa PPA
OpenGL core profile shading language version string: 4.20
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile

OpenGL version string: 4.2 (Compatibility Profile) Mesa 23.1.5 - kisak-mesa PPA
OpenGL shading language version string: 4.20
OpenGL context flags: (none)
OpenGL profile mask: compatibility profile

OpenGL ES profile version string: OpenGL ES 3.1 Mesa 23.1.5 - kisak-mesa PPA
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.10
```

For comparison, I export `LIBGL_ALWAY_SOFTWARE=1` and force the use of software rendering.

The following is the output. CPU usage increases under software rendering, while GPU is not involved. Interestingly, I found that software rendering fps is even higher than using GPU (CPU: 5fps, GPU: 1 fps), but both are far below the expected fps.

[image]

[image]

[image]

```
name of display: :0
display: :0  screen: 0
direct rendering: Yes
Extended renderer info (GLX_MESA_query_renderer):
    Vendor: Mesa (0xffffffff)
    Device: llvmpipe (LLVM 15.0.7, 256 bits) (0xffffffff)
    Version: 23.1.5
    Accelerated: no
    Video memory: 7876MB
    Unified memory: yes
    Preferred profile: core (0x1)
    Max core profile version: 4.5
    Max compat profile version: 4.5
    Max GLES1 profile version: 1.1
    Max GLES[23] profile version: 3.2
Memory info (GL_ATI_meminfo):
    VBO free memory - total: 524296 MB, largest block: 524296 MB
    VBO free aux. memory - total: 7193 MB, largest block: 7193 MB
    Texture free memory - total: 524296 MB, largest block: 524296 MB
    Texture free aux. memory - total: 7193 MB, largest block: 7193 MB
    Renderbuffer free memory - total: 524296 MB, largest block: 524296 MB
    Renderbuffer free aux. memory - total: 7193 MB, largest block: 7193 MB
Memory info (GL_NVX_gpu_memory_info):
    Dedicated video memory: 526344 MB
    Total available memory: 534220 MB
    Currently available dedicated video memory: 524296 MB
OpenGL vendor string: Mesa
OpenGL renderer string: llvmpipe (LLVM 15.0.7, 256 bits)
OpenGL core profile version string: 4.5 (Core Profile) Mesa 23.1.5 - kisak-mesa PPA
OpenGL core profile shading language version string: 4.50
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile

OpenGL version string: 4.5 (Compatibility Profile) Mesa 23.1.5 - kisak-mesa PPA
OpenGL shading language version string: 4.50
OpenGL context flags: (none)
OpenGL profile mask: compatibility profile

OpenGL ES profile version string: OpenGL ES 3.2 Mesa 23.1.5 - kisak-mesa PPA
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20
```

saran-t, 2023-08-12:
If you run the Windows build on the same system do you get a decent frame rate?

mantle2048, 2023-08-12:
Ok, I run `MUJOCO_GL=glx mujoco humanoid100.xml` and the following is my screenshot. Still 1-2 fps.

[image]

aftersomemath, 2023-08-12:
I was able to replicate the issue (1-2fps mujoco viewer) with a fresh install of WSL2, a RTX3080, 5800X3D. GPU is definitely being used. I tried a bunch of things, but haven't found a good reason for why it is slow.

`glmark2` and `glxgears` show similar GPU and CPU utilization as mujoco despite their job being to max out at least one of  them. I am inclined towards thinking this is a WSL2 problem?

mantle2048, 2023-08-12:
I'm not sure if this is a `mujoco` or `wslg` problem. But I made an additional attempt to verify the rendering performance on other simulators.

I installed `Unity` on wsl2 with `mujoco-plug` and imported `humanoid100`. 

The following is the screenshot and output of `windows task manager`.  It can be seen that GPU is working properly and fps is around 30.

[image]

[image]

saran-t, 2023-08-12:
Just a thought, can you try disabling vsync?

aftersomemath, 2023-08-12:
Turning off vsync didn't help my system.

I noticed `glxgears` achieves 150 fps while `LIBGL_ALWAYS_SOFTWARE=1 glxgears` achieves 2400 fps.

`glmark2` also performs better with software rendering, though the difference is not as drastic, ~1100 fps with hardware rendering, ~1200 with software.

`glxgears` should not be 10 times slower than `glmark2`. So I think this is a problem with WSLg. But since the MuJoCo viewer has similar symptoms as `glxgears` and `glmark2` gets a high framerate, there may be workaround.

mantle2048, 2023-08-13:
> Turning off vsync didn't help my system.
>
> I noticed `glxgears` achieves 150 fps while `LIBGL_ALWAYS_SOFTWARE=1 glxgears` achieves 2400 fps.
>
> `glmark2` also performs better with software rendering, [...]

This is consistent with what I observed. It is likely that WSLg's support for opengl is not sufficient and complete, which limits the rendering performance of mujoco gui. Hope Microsoft can solve this problem someday. (─‿‿─)

closed 2023-09-27 by saran-t

darkopetrovic, 2024-03-31:
I managed to achieve higher fps when I force using `wayland` instead of `x11`.

I am using the python bindings to start the visualizer like this:

`model.py`: 
```python
import mujoco as mj
import mujoco.viewer as vi

def main():
  xml = """
  <mujoco>
    <worldbody>
      <geom name="red_box" type="box" size=".2 .2 .2" rgba="1 0 0 1"/>
      <geom name="green_sphere" pos=".2 .2 .2" size=".1" rgba="0 1 0 1"/>
    </worldbody>
  </mujoco>
  """
  model = mj.MjModel.from_xml_string(xml)
  data = mj.MjData(model)
  vi.launch(model, data)

if __name__ == "__main__":
  main()
```

On windows host I started a X server with `VcSrv` (by disabling native opengl). 
When running `python model.py` on this X server the fps are 4-5. 

To force wayland, I have to set the env variable `XDG_SESSION_TYPE` to `wayland`: 

```bash
XDG_SESSION_TYPE=wayland python model.py
```

The FPS are now 60. 

However, I could not replicate this result with the `simulate` program because forcing `wayland` does not work. 

```
wget https://github.com/google-deepmind/mujoco/releases/download/3.1.2/mujoco-3.1.2-linux-x86_64.tar.gz
tar -xzf mujoco-3.1.2-linux-x86_64.tar.gz
cd mujoco-3.1.2/bin
./simulate ../model/cube/cube_3x3x3.xml
```

This open the viewer on the X server and the fps are 3-4. 

But running with `XDG_SESSION_TYPE=wayland ./simulate ../model/cube/cube_3x3x3.xml ` still opens the viewer on the X server.

If we manage to open `simulate` with `wayland` I suspect we can achieve higher fps in wsl.