google-deepmind/mujoco #1012: What is meaning of size for each geom type?

lucasw, 2023-08-08:
Is there a diagram or document that shows if the first cylinder size element is radius or diameter, and then if the next element is width, and so on for all the types?  Is each box size element the full length/width/height or 1/2 that?

kevinzakka, 2023-08-08:
See https://mujoco.readthedocs.io/en/latest/XMLreference.html#body-geom and https://mujoco.readthedocs.io/en/latest/overview.html?highlight=units#units-are-unspecified.

closed 2023-08-08 by kevinzakka

lucasw, 2023-08-08:
Thanks!

This issue and some copy-and-paste here ought to help with future google searches, it's somewhat unwieldy to look around in that giant page (there are 136 hits for 'size' for instance):

> The cylinder type defines a cylinder. It requires two size parameters: the radius and half-height of the cylinder. The cylinder is oriented along the Z axis of the geom’s frame....
> 
> The box type defines a box. Three size parameters are required, corresponding to the half-sizes of the box along the X, Y and Z axes of the geom’s frame

yuvaltassa, 2023-08-12:
Also for future reference, the correct documentation link is https://mujoco.readthedocs.io/en/latest/XMLreference.html#body-geom-size 

Links to attributes are available in the [schema](https://mujoco.readthedocs.io/en/latest/XMLreference.html#xml-schema).