google-deepmind/mujoco #1013: how to differentiate in mujoco with ocfw

jeongsukchul, 2023-08-09:
**How to ask for help**

First, read our quick guide to
[asking good questions](https://github.com/deepmind/mujoco#asking-questions).
Below is a template for you to use:

Hi,

I'm a student and I'm trying to use MuJoCo for drone simulating.

I'm looking for some help with differentiator.

Here is a model which explains my question:

i want to use data.time - prev_time = dt 
and if the thing i want to differetiate is X, dX=X-prev_X,  X_dot = dX/dt
 like this, but it is not working in my code. 

while not glfw.window_should_close(window):
    time_prev = data.time
    while (data.time - time_prev < 1.0 / 60.0):

I think the problem is because of this code, in glfw that "while (data.time - time_prev < 1.0 / 60.0):" is used, so when i get dt,
and divide some thing with dt, the value got explode. is there any way to get through?

But I think it have solution because 
i can under stand when you explain the algorithm you use in this function
mj.mj_objectVelocity(model, data, mj.mjtObj.mjOBJ_SITE,
                             0, xvel, 0)  # 1 for local, 0 for world frame
because i can't get the velocity with data.xpos[i]/dt

nimrod-gileadi, 2023-08-29:
Do you step the physics within the loop?

If your problem is dividing by zero on the first step, maybe just handle that first step explicitly?

yuvaltassa, 2024-01-08:
Closing due to lack of activity.

closed 2024-01-08 by yuvaltassa