google-deepmind/mujoco #1016: Size of cameras changed with objects which distribute in simulation.

muchvo, 2023-08-12:
**How to ask for help**

Hi,

I'm a student and I'm trying to use MuJoCo for Physical simulation.

I'm looking for some help with the mechanism of camera size.

When I add more objects in the simulation, the size of camera becomes bigger than before. it will cause some problems for me, such as collision detection failure in vision information.

So, I am wandering if it is a default setting and can I control it by tuning some parameters? Any help is appreciated, thanks for your effort to maintain this active community.

Here is a model which explains my question:
This is before adding geoms into simulation.
[image]

This is after adding geoms into simulation
[image]

Here is a minimal example to reproduce this problem, just need to uncomment boxes.
```XML
<mujoco model="MuJoCo Model">
  <default class="main">
  </default>
  <asset>
  </asset>
  <worldbody>
    <geom name="floor" size="5 5 0.1" type="plane" rgba="1 1 1 1"/>
    <camera name="camera1" pos="0 -2 2" quat="0.92388 0.382683 0 0"/>
    <camera name="camera2" pos="0 -5 5" quat="0.92388 0.382683 0 0"/>
    <light pos="0 0 0.5" dir="0 0 -1" directional="true" castshadow="false" cutoff="100" exponent="1" diffuse="1 1 1" specular="0 0 0"/>
    <body name="bo" pos="3 9 0.1" quat="0.979601 0 0 0.200952" gravcomp="0">
      <geom name="bo" size="0.1" friction="1 0.01 0.01" rgba="0.7412 0.0431 0.1843 1"/>
      <camera name="camera3" pos="0 0 0.1" quat="0.585532 0.396424 -0.396424 -0.585532" fovy="90"/>
      <camera name="camera4" pos="0.787411 1.83847 2" quat="0.185655 0.0769009 0.374877 0.905033" mode="track"/>
    </body>
    <!-- <body name="box0" pos="18.4631 -8.54781 0" quat="0.962982 0 0 0.269567" gravcomp="0">
      <joint name="box0" type="free" damping="0"/>
      <geom name="box0" quat="0.707107 0.707107 0 0" type="box" group="4" density="0.001"  rgba="1 1 1 1" size="1 1 1"/>
    </body>
    <body name="box1" pos="18.9097 -19.4892 0" quat="0.0168431 0 0 0.999858" gravcomp="0">
      <joint name="box1" type="free" damping="0"/>
      <geom name="box1" quat="0.707107 0.707107 0 0" type="box" group="4" density="0.001"  rgba="1 1 1 1" size="1 1 1"/>
    </body>
    <body name="box2" pos="25.8451 -0.552142 0" quat="0.203433 0 0 0.979089" gravcomp="0">
      <joint name="box2" type="free" damping="0"/>
      <geom name="box2" quat="0.707107 0.707107 0 0" type="box" group="4" density="0.001"  rgba="1 1 1 1" size="1 1 1"/>
    </body>
    <body name="box3" pos="25.1815 -18.8285 0" quat="0.603496 0 0 0.797366" gravcomp="0">
      <joint name="box3" type="free" damping="0"/>
      <geom name="box3" quat="0.707107 0.707107 0 0" type="box" group="4" density="0.001"  rgba="1 1 1 1" size="1 1 1"/>
    </body>
    <body name="box4" pos="10.7806 -5.31925 0" quat="-0.605844 0 0 0.795583" gravcomp="0">
      <joint name="box4" type="free" damping="0"/>
      <geom name="box4" quat="0.707107 0.707107 0 0" type="box" group="4" density="0.001"  rgba="1 1 1 1" size="1 1 1"/>
    </body>
    <body name="box5" pos="-8.96366 -19.9346 0" quat="0.510522 0 0 0.859865" gravcomp="0">
      <joint name="box5" type="free" damping="0"/>
      <geom name="box5" quat="0.707107 0.707107 0 0" type="box" group="4" density="0.001"  rgba="1 1 1 1" size="1 1 1"/>
    </body>
    <body name="box6" pos="-5.27783 -26.2195 0" quat="0.606906 0 0 0.794774" gravcomp="0">
      <joint name="box6" type="free" damping="0"/>
      <geom name="box6" quat="0.707107 0.707107 0 0" type="box" group="4" density="0.001"  rgba="1 1 1 1" size="1 1 1"/>
    </body>
    <body name="box7" pos="7.98453 -5.97053 0" quat="0.280324 0 0 0.959905" gravcomp="0">
      <joint name="box7" type="free" damping="0"/>
      <geom name="box7" quat="0.707107 0.707107 0 0" type="box" group="4" density="0.001"  rgba="1 1 1 1" size="1 1 1"/>
    </body>
    <body name="box8" pos="-10.6705 -17.0204 0" quat="-0.0707346 0 0 0.997495" gravcomp="0">
      <joint name="box8" type="free" damping="0"/>
[... 747 lines cut ...]
      <joint name="box195" type="free" damping="0"/>
      <geom name="box195" quat="0.707107 0.707107 0 0" type="box" group="4" density="0.001"  rgba="1 1 1 1" size="1 1 1"/>
    </body>
    <body name="box196" pos="-8.76573 -18.395 0" quat="-0.393012 0 0 0.919533" gravcomp="0">
      <joint name="box196" type="free" damping="0"/>
      <geom name="box196" quat="0.707107 0.707107 0 0" type="box" group="4" density="0.001"  rgba="1 1 1 1" size="1 1 1"/>
    </body>
    <body name="box197" pos="19.682 -20.5603 0" quat="0.951222 0 0 0.308508" gravcomp="0">
      <joint name="box197" type="free" damping="0"/>
      <geom name="box197" quat="0.707107 0.707107 0 0" type="box" group="4" density="0.001"  rgba="1 1 1 1" size="1 1 1"/>
    </body>
    <body name="box198" pos="8.6905 -30.7844 0" quat="-0.876401 0 0 0.481582" gravcomp="0">
      <joint name="box198" type="free" damping="0"/>
      <geom name="box198" quat="0.707107 0.707107 0 0" type="box" group="4" density="0.001"  rgba="1 1 1 1" size="1 1 1"/>
    </body>
    <body name="box199" pos="4.43763 -27.4434 0" quat="0.800362 0 0 0.599517" gravcomp="0">
      <joint name="box199" type="free" damping="0"/>
      <geom name="box199" quat="0.707107 0.707107 0 0" type="box" group="4" density="0.001"  rgba="1 1 1 1" size="1 1 1"/>
    </body> -->
  </worldbody>
</mujoco>

```

</details>

yuvaltassa, 2023-08-12:
Yes. As documented [here](https://mujoco.readthedocs.io/en/latest/XMLreference.html#visual-scale), all you need to do is set `<statistic meansize="x"/>` in order to prevent the compiler from automatically computing meansize from the scene.

closed 2023-08-12 by yuvaltassa

muchvo, 2023-08-13:
Thank you so much for your prompt response! I've given it a try and found it to be effective. However, I've noticed that in the primary camera view, when I introduce more objects into the simulation, the camera seems to enlarge compared to before. This causes it to 'clip' or 'intersect' with the objects. This leads to a visual anomaly known as 'z-fighting' or 'clipping errors' where the camera appears to see through the objects or they improperly overlap. This makes the scene look unnatural and disrupts the immersion. I'd greatly appreciate any advice or assistance you can offer to address this issue.

yuvaltassa, 2023-08-13:
Search the XMLreference for `extent`, `znear` and `zfar`.

muchvo, 2023-08-18:
It works! I'm not very proficient in computer graphics. Thank you so much.