google-deepmind/mujoco #1017: How force and torque are simulated?

stm32f303ret6, 2023-08-13:
I'm new to Mujoco and I'm trying to understand how forces/torques applied to joints work. I am using `d->qfrc_applied` to apply torque to a joint and it works fine, but when I implement a P controller I would expect the joint to oscillate, but it always converges to the setpoint (as if it had a full PID) why does this happen? I have also tried using a position actuator and I have the same effect.

The controller implementation is defined in the section of `mjcb_control`

kevinzakka, 2023-08-13:
Is your joint damping nonzero?

stm32f303ret6, 2023-08-13:
yes, this is the xml definition of that joint
```xml
            <body name="fr_steer" pos="2.45 1.25 0.73">
                <inertial pos="0 0 0" mass="30" diaginertia="0.5 0.5 0.2" />
                <joint name="fr_steer_joint" pos="0 0 0" axis="0 0 1" limited="false" />
                <geom size="0.1 0.1 0.2" type="box" rgba="1 1 1 1" />
            </body>
```

kevinzakka, 2023-08-13:
Set it to 0 and you should get oscillations.

stm32f303ret6, 2023-08-13:
> Set it to 0 and you should get oscillations.

apologies for the confusion, by default joint damping is 0 (xml ref.). So I still get no oscillations

kevinzakka, 2023-08-13:
Do you have any defaults set anywhere? Also are you plotting the qpos to make sure? ie not just visually inspecting

stm32f303ret6, 2023-08-15:
I don't have defaults settings, I've been doing isolated tests and I keep getting the same result

yuvaltassa, 2023-08-15:
Can you please post a MWE loadable model?
There is no way to diagnose your problem from the information in this thread.

stm32f303ret6, 2023-08-15:
I'm sorry I don't know what MWE stands for. But I can share the project where I am testing this:
I'm trying to simulate a swerve drive (4 steerings, 4 wheels) and I'm facing the problem of controlling the steering with just a P controller, i was expected the steering oscillating but it just reaches the setpoint without oscillations, once the control is computed i send it to ->qfrc_applied. You can control the setpoint steering angle with the keyboard W and S. Also you can test it using the GUI simulator of mujoco, loading xml/world.xml.
[swerve_drive.zip](https://github.com/deepmind/mujoco/files/12352536/swerve_drive.zip)

yuvaltassa, 2023-08-28:
IIUC correctly your wheels are cylinders, rolling on a plane yes?

I suspect you have two contact points so that when the wheels rotate the friction removes energy from the system. If you switch to ellipsoids you will get the oscillations you expect (if the axis is aligned with the contact).

Why aren't you using the built in position actuators?

PS MWE stands for minimum working example.

closed 2023-08-28 by yuvaltassa