google-deepmind/mujoco #1018: Direct compilation from c++ struct to mjModel

sliptime, 2023-08-15:
Hi,

I'm an engineer and I'm trying to use MuJoCo for a VR game in Unreal Engine.

I'm looking for some help with writing my plugin for Unreal Engine to be able to use the colliders and skeletons to generate a mjModel instance from within c++. I know that the general rule of thumb is to generate a mjModel instance from a predefined xml file but I hope to circumvent that and create a kind of struct that could be passed in instead which is essentially the xml file but in a c++ struct instead. I want to know if the functions from the binary release are capable of manually creating an entire mjModel or is all of that encapsulated in the mj_makeModel function. If so, then I'll have to go with the route of modifying and building this new mj_makeModel file from source. 
Thanks!

sliptime, 2023-08-17:
Ahh nevermind, Unreal had very extensive xml support so conversion from unreal world to xml to mujoco was very fast. Will just leave it as is.

closed 2023-08-18 by sliptime

yuvaltassa, 2023-08-18:
Good to hear. 

We'd love to know more about your hook into Unreal. 
If your codebase is public and readable, please post a link here when you think it might be ready.