google-deepmind/mujoco #1019: Unity Plugin " The type or namespace name...

troytully, 2023-08-16:
Hi,

I'm a Ph.D. student and I'm trying to use MuJoCo for Virtual Reality Sensory Feedback studies.
I'm looking for some help with getting an old Haptix model (my lab was the last lab to be in the Haptix program that ended this year)

We are using the LUKE hand model to pick up a block and receive force values off the block to provide sensory feedback to recreate a fragile egg. (if anyone has a model in Unity that already does this, I'd be happy to drop my project and pick up where you have the project)
[image]
 
I've be able to import the XML Below into Unity:

<mujoco>
	<include file="LukeHand.xml"/>

    <worldbody>
		<geom type="plane" material="MatPlane" pos=".6 0 0.1" zaxis="-1 0 1" size=".1 .5 .025"/>
		<geom type="plane" material="MatPlane" pos="-.6 0 0.1" zaxis="1 0 1" size=".1 .5 .025"/>
		<geom type="plane" material="MatPlane" pos="0 .6 0.1" zaxis="0 -1 1" size=".5 .1 .025"/>
		<geom type="plane" material="MatPlane" pos="0 -.6 0.1" zaxis="0 1 1" size=".5 .1 .025"/>

        <body name="obj1" pos="-0.2 0 .1">
            <geom type="box" rgba="0 1 0 1" size="0.05 0.03 0.05" density="500"/>
            <joint class="free"/>
        </body>

        
            
            
        

        <body name="obj3" pos="0.1 0 .025">
            <geom type="box" pos="-0.1 0 0" material="MatWoodG" size="0.01 0.3 0.1"/>
            
            
            
            
            
            
        </body>
    </worldbody>
</mujoco>

But I'm running into errors when trying to build and run it.

Other information:
I'm running on the Meta Quest Pro
I have Unity experience but it's relatively limited in the scope of Unity

erez-tom, 2023-08-19:
The error messages you posted indicate issues in installing the plug-in. You can skip that step and manually place the folder in your Unity asset folder, provided you also fetch the binary and name it as expected (this is mentioned in the documentation IIRC).
Did you manage to run any simulation yet? Start with verifying your setup with some simple models first, and then try the Haptix model. I'll leave the big open for now - please let us know how it goes!

troytully, 2023-08-21:
After doing the install both ways I'm pretty confident I have the plugin installed correctly. I'm getting the same error after placing the folder into the assets folder manually. I've also tried importing the simple 

I don't know what you mean by naming it as expected. I haven't read anything in the Unity section that would imply naming a file by a specific name.

I imported the hammock.xml and tried building with that and got the same error.

Thank you so much for helping! I don't know what steps I've missed as the documentation doesn't mention much other than importing the plug in.

troytully, 2023-08-21:
The only "renaming" suggested is for OSX and Linux. I'm on windows so no renaming is required. The suggested "manual" action for windows 

After deleting MujocoBinaryRetriever.cs I have new errors:

[image]

There are almost 70 of these errors. It seems like it just can't find the files even though the unity folder is within the assets folder of the project.

troytully, 2023-08-21:
It seems now that I can press play on the scene within unity editor but I still run into errors when trying to build the project.

Balint-H, 2023-08-21:
Hello, those errors on build come up when the editor files are included in the build assembly file. Could you check the assembly file in the Editor folder, and see what platforms are enabled? Everything, except "Editor" should be disabled.

troytully, 2023-08-21:
I just downloaded this plug-in so it should be up to date.

Which editor folder? The one in the unity project or the editor folder under 
mujocounity-main\unity\Editor

What is the name of the assembly file?

Thank you all for the help! I've been spending hours trying to figure out my issue.

Balint-H, 2023-08-21:
Hello, it's at MuJoCo/Editor/Mujoco.Editor.asmdef. Could you send a screenshot of how that file looks in your inspector when you click on it in Unity?

troytully, 2023-08-21:
Here is the file I believe you are asking for.
[image]

Balint-H, 2023-08-21:
Interestingly its missing the editor only platform criteria. 

Adding it either in a text editor or through the Unity inspector should fix it for you. I'm not sure what could have changed it.

If you do remember the steps you've taken to install could you detail them? I'd like to try to reproduce it and see what could have caused it.

troytully, 2023-08-21:
open the MuJoCo repo on github, click the green "<>code" icon, and then download zip.

Balint-H, 2023-08-21:
And then you added the unzipped package via Unity's package manager and the package.json file in unity folder of the repo?

troytully, 2023-08-21:
Yes exactly.

troytully, 2023-08-21:
I was able to build and run the flag.xml as well as an old HAPTIX model we are hoping to utilize. 

Thank you both, I'm sure I'll be looking for help again. (Is this the best place to ask for help?)