google-deepmind/mujoco #1020: Body not appearing as geom option in mjmodel.geom()

manavkulshrestha, 2023-08-16:
**How to ask for help**

Hi,

I'm a student and I'm trying to use MuJoCo for doing some robot control. I'm looking for help with some API things that are unclear to me. This is the MJCF I'm using

<details>
  <summary>MJCF for my scene</summary>

```XML
<mujoco model="turtlebot3_burger">
  <compiler angle="radian" meshdir="./" autolimits="true"/>
  <statistic meansize="0.0480478" extent="0.284544" center="-0.0316037 -1.28119e-06 0.0998247"/>

  <asset>
    <!-- sky and plane -->
    <texture type="skybox" builtin="gradient" width="2048" height="2048"/>
    <texture name="grid" type="2d" builtin="checker" width="512" height="512" rgb1=".1 .2 .3" rgb2=".2 .3 .4"/>
    <material name="grid" texture="grid" texrepeat="1 1" texuniform="true" reflectance=".2"/>

    <!-- robot -->
    <mesh name="burger_base" file="robot/burger_base.stl" scale="0.001 0.001 0.001"/>
    <mesh name="left_tire" file="robot/left_tire.stl" scale="0.001 0.001 0.001"/>
    <mesh name="right_tire" file="robot/right_tire.stl" scale="0.001 0.001 0.001"/>
    <mesh name="lds" file="robot/lds.stl" scale="0.001 0.001 0.001"/>
  </asset>

  <worldbody>
    <!-- sky and plane -->
    <light cutoff="100" diffuse="1 1 1" dir="0 0 -1.3" directional="true" exponent="1" pos="0 0 1.3" specular="0.1 0.1 0.1"/>
	  <geom name="floor" type="plane" size="100 100 .2" material="grid"/>

    <!-- robot -->
    <body name="robot" pos="-0.032 0 0.01">
      <freejoint/>
      <geom pos="0 0 0" type="mesh" mesh="burger_base"/>
      <geom size="0.015 0.0045 0.01" pos="-4.90000e-02 7.96327e-07 -5.00000e-03" quat="0.707388 -0.706825 0 0" type="box"/>
      <geom pos="0 0 0.172" type="mesh" mesh="lds"/>
      
      <body name="wheel_left_link" pos="0.032 0.08 0.023" quat="0.707388 -0.706825 0 0">
        <inertial pos="0 0 0" quat="-0.000890159 0.706886 0.000889646 0.707326" mass="0.0284989" diaginertia="2.07126e-05 1.11924e-05 1.11756e-05"/>
        <joint name="wheel_left_joint" pos="0 0 0" axis="0 0 1"/>
        <geom quat="0.707388 0.706825 0 0" type="mesh" mesh="left_tire"/>
      </body>

      <body name="wheel_right_link" pos="0.032 -0.08 0.023" quat="0.707388 -0.706825 0 0">
        <inertial pos="0 0 0" quat="-0.000890159 0.706886 0.000889646 0.707326" mass="0.0284989" diaginertia="2.07126e-05 1.11924e-05 1.11756e-05"/>
        <joint name="wheel_right_joint" pos="0 0 0" axis="0 0 1"/>
        <geom quat="0.707388 0.706825 0 0" type="mesh" mesh="right_tire"/>
      </body>
    </body>

  </worldbody>

  <actuator>
    <motor name="left_motor" joint="wheel_left_joint" gear="1"/>
    <motor name="right_motor" joint="wheel_right_joint" gear="1"/>
  </actuator>

</mujoco>
```

</details>

Now, despite the fact that I have a  body named "robot" in the world body, it is not appearing as an option in the traceback when I run model.geom():
```
Traceback (most recent call last):
  File "<string>", line 1, in <module>
KeyError: "Invalid name ''. Valid names: ['floor']"
```
I wanted to know if I've made some mistake in the XML or maybe I'm not doing what I intend to be doing. My end goal for now is to try and get the xpos value for the robot. I may also have multiple robots so it would be better to have some systematic way to do this instead of simply parsing `data.xpos` to find the associated 3d vector (since those also seem to contain the wheel links and floor etc.)

Any guidance would be appreciated as this is a fairly new API for me, coming from pybullet.

yuvaltassa, 2023-08-16:
1. Bodies and geoms are not the same thing. Read the [overview](https://mujoco.readthedocs.io/en/latest/overview.html) and/or go through the [tutorial](https://mujoco.readthedocs.io/en/latest/python.html#tutorial-notebook).

2. Regarding named access, try going through the [tutorial](https://mujoco.readthedocs.io/en/latest/python.html#tutorial-notebook) and/or reading the [relevant docs](https://mujoco.readthedocs.io/en/latest/python.html#named-access).

If you still can't figure it out report back here and we will help you 🙂

manavkulshrestha, 2023-08-16:
Thank you for the prompt response! I had gone through the tutorial, but probably not thoroughly enough. The reference for named access in python bindings and going over geoms vs body was very helpful, thank you! TLDR; using `model.body('robot').xpos` works well

closed 2023-08-16 by manavkulshrestha