google-deepmind/mujoco #1023: add custom geoms to mujoco.viewer.launch_passive

nopaddleboat, 2023-08-20:
Hi,

I am using mujoco.viewer.launch_passive which is very convenient to use except that I found it hard to add custom geoms as mentioned in issues  #382 #132 etc. I know it is easy to add geoms using C/C++ version of mujoco(#132), but it would be faster to use the python version for prototype. I tried to add the geoms as follows (part of the exmple from the doc) but failed. I wonder if there is a way to add geoms to the passive viewer. Thanks.

```python
with mujoco.viewer.launch_passive(m, d) as viewer:
  while viewer.is_running():
with viewer.lock():
      viewer.scn.ngeom+=1
       mujoco.mjv_initGeom(viewer.scn.geoms[viewer.scn.ngeom-1],
                         mujoco.mjtGeom.mjGEOM_SPHERE, np.array([0.38,0.42,1.0]),
                      np.array([1,1,1.0]), np.zeros(9), rgba.astype(np.float32))
viewer.sync()
```

HuNingHe, 2023-08-25:
I have the same problem, here's part of my code:

```
iter_ct += 1
with viewer.lock():
    viewer.scn.ngeom += 1
    mujoco.mjv_initGeom(viewer.scn.geoms[viewer.scn.ngeom - 1],
                        mujoco.mjtGeom.mjGEOM_CAPSULE, np.array([0.38, 0.42, 1.0]),
                        np.array([1, 1, 1.0]), np.zeros(9), np.array([0.5, 0.5, 0.5, 1]))

    mujoco.mjv_connector(viewer.scn.geoms[viewer.scn.ngeom - 1], mujoco.mjtGeom.mjGEOM_CAPSULE, 0.02,
                     np.array([0.0001 * iter_ct, 0.0001 * iter_ct, 0.0001 * iter_ct]),
                     np.array([0.001 * iter_ct, 0.001 * iter_ct, 0.001 * iter_ct]))
viewer.sync()
```

No geoms would be visualized, but when I removed `with viewer.lock()`, the geom would constantly flicker.

nopaddleboat, 2023-08-25:
The reason why users cannot add new geoms through exposed viewer.scn may be that in passive view, viewer.scn will be overwritten by scnstate_ in the function mjv_updateSceneFromState as shown in simulate.cc.
```C
     if (this->fully_managed_) {
        Sync();
      } else {
        scnstate_.data.warning[mjWARN_VGEOMFULL].number += mjv_updateSceneFromState(
            &scnstate_, &this->opt, &this->pert, &this->cam, mjCAT_ALL, &this->scn);      
       }
```
Currently I wrapped up another MjvScene scn_geo which can be used to add new geoms by the python users and not used by the original code. To make it work, the above code block has to be modified as follows. I tried it and it works. However, this may not be a solution in the long run.
```C
if (this->fully_managed_) {
        Sync();
      } else {
        scnstate_.data.warning[mjWARN_VGEOMFULL].number += mjv_updateSceneFromState(
            &scnstate_, &this->opt, &this->pert, &this->cam, mjCAT_ALL, &this->scn);
        if(this->scn_geo.ngeom>0){
          memcpy(this->scn.geoms+this->scn.ngeom, this->scn_geo.geoms, sizeof(mjvGeom) * this->scn_geo.ngeom);
          this->scn.ngeom+=this->scn_geo.ngeom;
        }        
      }
```

saran-t, 2023-08-25:
This isn't possible currently but I'm thinking about ways to implement this.

saran-t, 2023-09-06:
Sorry for taking this long, but I've just reviewed my own code again and actually think that this should already be possible -- you just need to call `viewer.sync()` _before_ modifying `viewer.scn`, i.e.

```
v.sync()
with v.lock():
  viewer.scn.ngeom += 1
  mujoco.mjv_initGeom(
      viewer.scn.geoms[viewer.scn.ngeom-1],
      mujoco.mjtGeom.mjGEOM_SPHERE, np.array([0.38,0.42,1.0]),
      np.array([1,1,1.0]), np.zeros(9), rgba.astype(np.float32))
```

Could you please try this and report back?

closed 2023-10-10 by copybara-service[bot]

saran-t, 2023-10-10:
Sorry that it's taken so long to roll this out. We're planning to release a new version that incorporates 2e15574b58df4bfda59e1642cf67699cb0a58667 in the middle of next week.

If you want to try this out before then, you can `pip install mujoco --pre -f https://py.mujoco.org/`.

vikashplus, 2023-10-15:
@saran-t  -- this update is amazing!
Wasted half a day only to learn about the new feature. 
Implemented it. Using it. And loving it already.

saran-t, 2024-11-22:
@yourj4m this isn't actually related to the original issue, which was about the passive python GUI

In your case the problem is that you're calling `visualize_debug_grid` before `mjv_updateScene`, and the latter call clears the `scn`.

harshal-14, 2026-10-10:
I have a python package to exactly solve this drawing mismatch: https://github.com/harshal-14/mujoco-debugdraw

commit 2e15574b (2023-10-10): Allow user-defined visualization geoms to be added to the passive viewer.

Fixes #1023, #1082

PiperOrigin-RevId: 572303635
Change-Id: Ia8399f816f311b38a7b137f5bd4f359f77a1da1c