google-deepmind/mujoco #1024: Actuator type change at runtime: Not working after upgrading from 2.1.5 to 2.3.1

kim135797531, 2023-08-20:
Hello,

Thank you for the deepmind's continued dedication to managing the code for MuJoCo.
This inquiry concerns the process of modifying actuators at runtime.

In simulations involving a model with a muscle actuator within Python binding,
sometimes I need to switch the `muscle` actuator to a `general` actuator at runtime for debugging.

Here's how the actuator type is modified:
```
env.model.actuator_dyntype[:] = 0  # none
env.model.actuator_gaintype[:] = 0  # fixed
env.model.actuator_biastype[:] = 0  # none
env.model.actuator_dynprm[:] = 0
env.model.actuator_gainprm[:] = 0
env.model.actuator_gainprm[:, 0] = 1  # p = a*u
env.model.actuator_biasprm[:] = 0
```

Recently, after moving from MuJoCo version 2.1.5 to 2.3.1, the simulation stopped to work with the same code.

Upon further examination, it was discovered that a check for `actuator_actadr == -1` has been newly included in engine_forward.c.
(https://github.com/deepmind/mujoco/blame/36b30e4500e2d1950b7992621cac3d193de3c980/src/engine/engine_forward.c#L234)

By adding the following lines, movement as a general actuator was restored:
```
env.model.actuator_actadr[:] = -1
env.model.actuator_actnum[:] = 0
```

However, this detail is not readily apparent in the XMLreference document.

It would be beneficial to include this information in the documentation, or explicitly state that modifying the actuator at runtime might lead to unintended consequences.

Thank you.

yuvaltassa, 2023-08-21:
Modifying anything in the `mjModel` is not recommended, recompilation is the only safe thing to do. 

That said, we've been considering adding a feature to allow the user to disable/enable actuators at runtime. This would mean that you could add you general actuators but mark them as disabled and at runtime safely enable them while disabling the muscles. Note that the size of the `ctrl` vector will reflect the full number of actuators, and it will be up to you to know which `ctrl` entries are being currently ignored. Is this a feature that you would use?

closed 2023-08-21 by yuvaltassa

kim135797531, 2023-08-21:
Thank you for your response.

That's an interesting idea. However, in my current program, many parts refer to the total number of actuators, so I fear it might lead to other bugs. In my case, I will create a separate XML with general actuators instead.

After updating from 2.1.5 to 2.3.1, my simulation has become almost 10 times faster! Thank you.