google-deepmind/mujoco #1025: how to setup the weld equality rigidness?

linnsong, 2023-08-21:
Hi,

Please see following examples, i weld 2 ball togather to simulate a finger structure, when i set the hanging ball mass little, eg. 1, it's fine, but if i add the mass of the hanging ball to 50 , the 2 weld ball will have a displacement between them. it's confuse me that how to setup the rigidness of the weld, solref works but i have no idea what's the safe boundary.  this issue get worse in example 2, a ten percentage little one. which i will simulate a finger sturcture.

example 1: 10 times big

<mujoco>
  <option>
    <flag contact="disable"/>
  </option>
  <compiler angle="radian" autolimits="true"/>
  <visual>
    <global azimuth="220" elevation="-20"/>
  </visual>
  <default>
    <default class="50cm">
      <geom size="0.05 0.25 0.05" rgba=".9 .9 .5 .3"/>
    </default>
    <default class="100cm">
      <geom type="box" size=".05" fromto="0 -0.5 0 0 0.5 0" rgba=".4 .7 .6 .3"/>
    </default>
    <tendon width="0.0005" damping="0.8" stiffness="1000" springlength="-1"/>
    <default class="pivot">
        <joint type="hinge" axis="1 0 0" damping="0.001"/>
    </default>
    <default class="crank">
        <joint type="slide" axis="0 1 0" damping="0.001" range="0 0.01"/>
    </default>
    <default class="F_slidedriver">
        <intvelocity forcerange="-1000 1000" ctrlrange="-10 10" gear="200" actrange="-10 10" kp="20"/>
    </default>
  </default>

  <worldbody>
    <geom pos="0 0 -2" type="plane" size="10 10 .01"/>
    <light pos="0 0 20"/>

    <body name="framebase" pos="0 -0.5 0">
        <geom type="box" class="100cm"/>
        <body pos="0 1 0">
            <geom type="box" class="100cm"/>
            <joint name="axis1" pos="0 -0.5 0" class="pivot" range="-0.75 0.75"/>
            <site name="site1" pos="0 0.5 0" size="0.005" rgba="1 .2 .2 .3"/>
            <body name="w2" pos="0 -0.45 0.1">
                <geom type="sphere" size="0.03" rgba="1 0 1 0.5"/>
            </body>
        </body>
        <body pos="0 -0.25 0.1">
             <joint name="axis2" pos="0 -0.25 0.00" class="pivot"/>
             <geom type="box" class="50cm"/>
             <body pos="0 0.5 0">
                <joint name="slide1" pos="0 -0.25 0" class="crank"/>
                <geom type="box" class="50cm" rgba="0 0 0.9 0.8"/>
                <body name="w1" pos="0 0.3 0">
                    <joint name="axis3" pos="0 0.0 0" class="pivot"/>
                    <geom type="sphere" size="0.05" rgba="1 0 0 0.5"/>
                </body>
             </body>
        </body>
    </body>

    <body name="obj" pos="0 1 -0.5">
        <freejoint/>
        <geom type="sphere" size="0.2" rgba="0.2 0.2 0.8 0.3" mass="50"/>
        <site name="site2" pos="0 0 0" size="0.01"/>
    </body>
  </worldbody>

  <equality>
    <weld name="hard"  body1="w1" body2="w2" torquescale="1"/>
  </equality>
  <tendon>
    <spatial name="palmedge1">   
        <site site="site1"/>
        <site site="site2"/> 
    </spatial>
  </tendon>
  <actuator>
        <intvelocity name="slidecontrol" joint="slide1" class="F_slidedriver"/>
  </actuator>
</mujoco>

example2 : finger size

<mujoco>
  <option>
    <flag contact="disable"/>
  </option>
  <compiler angle="radian" autolimits="true"/>
  <visual>
    <global azimuth="220" elevation="-20"/>
  </visual>

  <default>
    <default class="5cm">
      <geom size="0.005 0.025 0.005" rgba=".9 .9 .5 .3"/>
    </default>
    <default class="10cm">
      <geom type="box" size=".005" fromto="0 -0.05 0 0 0.05 0" rgba=".4 .7 .6 .3"/>
    </default>
    <tendon width="0.0005" damping="0.8" stiffness="1000" springlength="-1"/>
    <default class="pivot">
        <joint type="hinge" axis="1 0 0" damping="0.001"/>
    </default>
    <default class="crank">
        <joint type="slide" axis="0 1 0" damping="0.001" range="0 0.01"/>
    </default>
    <default class="F_slidedriver">
        <intvelocity forcerange="-70 70" ctrlrange="-1 1" gear="20" actrange="-0.2 0.2" kp="20"/>
    </default>
  </default>

  <worldbody>
    <geom pos="0 0 -2" type="plane" size="10 10 .01"/>
    <light pos="0 0 20"/>

    <body name="framebase" pos="0 -0.05 0">
        <geom type="box" class="10cm"/>
        <body pos="0 0.1 0">
            <geom type="box" class="10cm"/>
            <joint name="axis1" pos="0 -0.05 0" class="pivot" range="-0.75 0.75"/>
            <site name="site1" pos="0 0.05 0" size="0.005" rgba="1 .2 .2 .3"/>
            <body name="w2" pos="0 -0.045 0.01">
                <geom type="sphere" size="0.005" rgba="1 0 1 0.5"/>
            </body>
        </body>
        <body pos="0 -0.025 0.01">
             <joint name="axis2" pos="0 -0.025 0.00" class="pivot"/>
             <geom type="box" class="5cm"/>
             <body pos="0 0.05 0">
                <joint name="slide1" pos="0 -0.025 0" class="crank"/>
                <geom type="box" class="5cm" rgba="0 0 0.9 0.8"/>
                <body name="w1" pos="0 0.03 0">
                    <joint name="axis3" pos="0 0.0 0" class="pivot"/>
                    <geom type="sphere" size="0.0025" rgba="1 0 0 0.5"/>
                </body>
             </body>
        </body>
    </body>
    
    <body name="obj" pos="0 0.1 -0.1">
        <freejoint/>
        <geom type="sphere" size="0.02" rgba="0.2 0.2 0.8 0.3" mass="0.1"/>
        <site name="site2" pos="0 0 0" size="0.01"/>
    </body>
  </worldbody>

  <equality>
    <weld name="hardweld"  body1="w1" body2="w2" torquescale="1"/>
  </equality>
  <tendon>
    <spatial name="palmedge1">   
        <site site="site1"/>
        <site site="site2"/> 
    </spatial>
  </tendon>
  <actuator>
        <intvelocity name="slidecontrol" joint="slide1" class="F_slidedriver"/>
  </actuator>
</mujoco>

br
linn

yuvaltassa, 2023-08-21:
Your question is unclear, I see no welds in you model?

Perhaps [this section](https://mujoco.readthedocs.io/en/latest/modeling.html#solver-parameters) will be helpful.

linnsong, 2023-08-21:
sorry, copy paste error, 
here 10times big:

<mujoco>
  <option>
    <flag contact="disable"/>
  </option>
  <compiler angle="radian" autolimits="true"/>
  <visual>
    <global azimuth="220" elevation="-20"/>
  </visual>
  <default>
    <default class="50cm">
      <geom size="0.05 0.25 0.05" rgba=".9 .9 .5 .3"/>
    </default>
    <default class="100cm">
      <geom type="box" size=".05" fromto="0 -0.5 0 0 0.5 0" rgba=".4 .7 .6 .3"/>
    </default>
    <tendon width="0.0005" damping="0.8" stiffness="1000" springlength="-1"/>
    <default class="pivot">
        <joint type="hinge" axis="1 0 0" damping="0.001"/>
    </default>
    <default class="crank">
        <joint type="slide" axis="0 1 0" damping="0.001" range="0 0.01"/>
    </default>
    <default class="F_slidedriver">
        <intvelocity forcerange="-1000 1000" ctrlrange="-10 10" gear="200" actrange="-10 10" kp="20"/>
    </default>
  </default>

  <worldbody>
    <geom pos="0 0 -2" type="plane" size="10 10 .01"/>
    <light pos="0 0 20"/>

    <body name="framebase" pos="0 -0.5 0">
        <geom type="box" class="100cm"/>
        <body pos="0 1 0">
            <geom type="box" class="100cm"/>
            <joint name="axis1" pos="0 -0.5 0" class="pivot" range="-0.75 0.75"/>
            <site name="site1" pos="0 0.5 0" size="0.005" rgba="1 .2 .2 .3"/>
            <body name="w2" pos="0 -0.45 0.1">
                <geom type="sphere" size="0.03" rgba="1 0 1 0.5"/>
            </body>
        </body>
        <body pos="0 -0.25 0.1">
             <joint name="axis2" pos="0 -0.25 0.00" class="pivot"/>
             <geom type="box" class="50cm"/>
             <body pos="0 0.5 0">
                <joint name="slide1" pos="0 -0.25 0" class="crank"/>
                <geom type="box" class="50cm" rgba="0 0 0.9 0.8"/>
                <body name="w1" pos="0 0.3 0">
                    <joint name="axis3" pos="0 0.0 0" class="pivot"/>
                    <geom type="sphere" size="0.05" rgba="1 0 0 0.5"/>
                </body>
             </body>
        </body>
    </body>

    <body name="obj" pos="0 1 -0.5">
        <freejoint/>
        <geom type="sphere" size="0.2" rgba="0.2 0.2 0.8 0.3" mass="5"/>
        <site name="site2" pos="0 0 0" size="0.01"/>
    </body>
  </worldbody>

  <equality>
    <weld name="hard"  body1="w1" body2="w2" torquescale="1" solref="0.004 0.4"/>
  </equality>
  <tendon>
    <spatial name="palmedge1">   
        <site site="site1"/>
        <site site="site2"/> 
    </spatial>
  </tendon>
  <actuator>
        <intvelocity name="slidecontrol" joint="slide1" class="F_slidedriver"/>
  </actuator>
</mujoco>

small one:

<mujoco>
  <option>
    <flag contact="disable"/>
  </option>
  <compiler angle="radian" autolimits="true"/>
  <visual>
    <global azimuth="220" elevation="-20"/>
  </visual>

  <default>
    <default class="5cm">
      <geom size="0.005 0.025 0.005" rgba=".9 .9 .5 .3"/>
    </default>
    <default class="10cm">
      <geom type="box" size=".005" fromto="0 -0.05 0 0 0.05 0" rgba=".4 .7 .6 .3"/>
    </default>
    <tendon width="0.0005" damping="0.8" stiffness="1000" springlength="-1"/>
    <default class="pivot">
        <joint type="hinge" axis="1 0 0" damping="0.001"/>
    </default>
    <default class="crank">
        <joint type="slide" axis="0 1 0" damping="0.001" range="0 0.01"/>
    </default>
    <default class="F_slidedriver">
        <intvelocity forcerange="-70 70" ctrlrange="-1 1" gear="20" actrange="-0.2 0.2" kp="20"/>
    </default>
  </default>

  <worldbody>
    <geom pos="0 0 -2" type="plane" size="10 10 .01"/>
    <light pos="0 0 20"/>

    <body name="framebase" pos="0 -0.05 0">
        <geom type="box" class="10cm"/>
        <body pos="0 0.1 0">
            <geom type="box" class="10cm"/>
            <joint name="axis1" pos="0 -0.05 0" class="pivot" range="-0.75 0.75"/>
            <site name="site1" pos="0 0.05 0" size="0.005" rgba="1 .2 .2 .3"/>
            <body name="w2" pos="0 -0.045 0.01">
                <geom type="sphere" size="0.005" rgba="1 0 1 0.5"/>
            </body>
        </body>
        <body pos="0 -0.025 0.01">
             <joint name="axis2" pos="0 -0.025 0.00" class="pivot"/>
             <geom type="box" class="5cm"/>
             <body pos="0 0.05 0">
                <joint name="slide1" pos="0 -0.025 0" class="crank"/>
                <geom type="box" class="5cm" rgba="0 0 0.9 0.8"/>
                <body name="w1" pos="0 0.03 0">
                    <joint name="axis3" pos="0 0.0 0" class="pivot"/>
                    <geom type="sphere" size="0.0025" rgba="1 0 0 0.5"/>
                </body>
             </body>
        </body>
    </body>
    
    <body name="obj" pos="0 0.1 -0.1">
        <freejoint/>
        <geom type="sphere" size="0.02" rgba="0.2 0.2 0.8 0.3" mass="0.1"/>
        <site name="site2" pos="0 0 0" size="0.01"/>
    </body>
  </worldbody>

  <equality>
    <weld name="hardweld"  body1="w1" body2="w2" torquescale="1" solref="0.004 0.4"/>
  </equality>
  <tendon>
    <spatial name="palmedge1">   
        <site site="site1"/>
        <site site="site2"/> 
    </spatial>
  </tendon>
  <actuator>
        <intvelocity name="slidecontrol" joint="slide1" class="F_slidedriver"/>
  </actuator>
</mujoco>

closed 2024-08-30 by kevinzakka