google-deepmind/mujoco #1026: Can I make a parallel robot mechanism using MJCF format?

hi-an, 2023-08-21:
Hello,

I'm a student and I'm trying to use MuJoCo for training a parallel mechanism robot via reinforcement learning algorithms.

I'm looking for some help with generating a parallel mechanism robot. I want to add one more joint to `link_b` and `rim` parts. 

As my understanding, `joint` feature only can use in the `body` tag because it set degree of freedoms between parent and child bodies. If I add a `rim` body inside the `link_b` body(line 23~27), there is an error `ValueError: Error: repeated name in body array`. I don't want to multiply `rim` parts by setting a different name.

Are there solutions about this problem??

Here is my current model which explains my question:

<details>
  <summary>Parallel rboot mechanism xml</summary>

```XML
<mujoco model="onshape">
  <compiler angle="radian" autolimits="true"/>
  <statistic meansize="0.652061" extent="2.87506" center="0 0.0912037 -0.0610164"/>
  <asset>
    <mesh name="center" file="center.stl"/>
    <mesh name="motor" file="motor.stl"/>
    <mesh name="rot_plate_b" file="rot_plate_b.stl"/>
    <mesh name="link_b" file="link_b.stl"/>
    <mesh name="rot_plate_r" file="rot_plate_r.stl"/>
    <mesh name="link_r" file="link_r.stl"/>
    <mesh name="rim" file="rim.stl"/>
  </asset>

  <worldbody>
    <body name="wheel" pos="0 0 0">
      <geom type="mesh" rgba="0.4 0.4 0.4 1" mesh="center"/>
      <geom pos="0.34 0 0" type="mesh" rgba="0.6 0.580392 0.529412 1" mesh="motor"/>
      <geom pos="-0.34 0 0" type="mesh" rgba="0.6 0.580392 0.529412 1" mesh="motor"/>
      <body name="rot_plate_b" pos="0.2175 0 0" quat="0.151895 -0.6906 0.151895 -0.6906">
        <inertial pos="0 0 -0.0645256" quat="0 0.707107 0 0.707107" mass="59.9046" diaginertia="3.13893 1.63884 1.63884"/>
        <joint name="rot_b_j" pos="0 0 0" axis="0 0 1"/>
        <geom pos="0 0 -0.0975" quat="0.5 -0.5 -0.5 -0.5" type="mesh" rgba="0 0 1 1" mesh="rot_plate_b"/>
        <body name="link_b" pos="-0.480644 -0.2775 -0.08" quat="0 -0.852969 -0.521962 0">
          <inertial pos="0 0.3775 -0.0175" quat="0.5 0.5 -0.5 0.5" mass="7.87695" diaginertia="0.463448 0.457843 0.00721361"/>
          <joint name="link1_b_j" pos="0 0 0" axis="0 0 1"/>
          <geom pos="0 0.3775 -0.0175" quat="0.5 -0.5 -0.5 -0.5" type="mesh" rgba="0 0 1 1" mesh="link_b"/>
        </body>
        <body name="link_b_2" pos="0.480644 -0.2775 -0.08" quat="0 -0.025548 0.999674 0">
          <inertial pos="0 0.3775 -0.0175" quat="0.5 0.5 -0.5 0.5" mass="7.87695" diaginertia="0.463448 0.457843 0.00721361"/>
          <joint name="link3_b_j" pos="0 0 0" axis="0 0 1"/>
          <geom pos="0 0.3775 -0.0175" quat="0.5 -0.5 -0.5 -0.5" type="mesh" rgba="0 0 1 1" mesh="link_b"/>
        </body>
        <body name="link_b_3" pos="0 0.555 -0.08" quat="0 0.878517 -0.477712 0">
          <inertial pos="0 0.3775 -0.0175" quat="0.5 0.5 -0.5 0.5" mass="7.87695" diaginertia="0.463448 0.457843 0.00721361"/>
          <joint name="link2_b_j" pos="0 0 0" axis="0 0 1"/>
          <geom pos="0 0.3775 -0.0175" quat="0.5 -0.5 -0.5 -0.5" type="mesh" rgba="0 0 1 1" mesh="link_b"/>
        </body>
      </body>

      <body name="rot_plate_r" pos="-0.2175 0 0" quat="-0.609585 0.358338 0.609585 -0.358338">
        <inertial pos="0 0 -0.0645256" quat="0.5 0.5 -0.5 0.5" mass="59.9046" diaginertia="3.13893 1.63884 1.63884"/>
        <joint name="rot_r_j" pos="0 0 0" axis="0 0 1"/>
        <geom pos="0 0 -0.0975" quat="0.5 -0.5 -0.5 -0.5" type="mesh" rgba="1 0 0 1" mesh="rot_plate_r"/>
        <body name="link_r" pos="0.480644 -0.2775 -0.08" quat="0 -0.999674 -0.025548 0">
          <inertial pos="0 -0.3775 -0.0175" quat="0.5 0.5 -0.5 0.5" mass="7.87695" diaginertia="0.463448 0.457843 0.00721361"/>
          <joint name="link1_r_j" pos="0 0 0" axis="0 0 1"/>
          <geom pos="0 -0.3775 -0.0175" quat="0.5 0.5 0.5 -0.5" type="mesh" rgba="1 0 0 1" mesh="link_r"/>
          <body name="rim" pos="0 -0.755 0" quat="0 0.926466 -0.376377 0">
            <inertial pos="-0.105737 -0.314431 -0.170074" quat="0.681036 0.684974 -0.183545 -0.182515" mass="120.247" diaginertia="34.4863 21.9667 15.7317"/>
            <joint name="rim1_r_j" pos="0 0 0" axis="0 0 1"/>
            <geom pos="-0.65 0 -0.17" quat="0.5 0.5 0.5 -0.5" type="mesh" rgba="0.623529 0.666667 0.701961 1" mesh="rim"/>
          </body>
        </body>
        <body name="link_r_2" pos="0 0.555 -0.08" quat="0 0.477712 0.878517 0">
          <inertial pos="0 -0.3775 -0.0175" quat="0.5 0.5 -0.5 0.5" mass="7.87695" diaginertia="0.463448 0.457843 0.00721361"/>
          <joint name="link2_r_j" pos="0 0 0" axis="0 0 1"/>
          <geom pos="0 -0.3775 -0.0175" quat="0.5 0.5 0.5 -0.5" type="mesh" rgba="1 0 0 1" mesh="link_r"/>
          <body name="rim_2" pos="0 -0.755 0" quat="0 -0.926466 0.376377 0">
            <inertial pos="-0.105737 -0.314431 -0.170074" quat="0.681036 0.684974 -0.183545 -0.182515" mass="120.247" diaginertia="34.4863 21.9667 15.7317"/>
            <joint name="rim2_r_j" pos="0 0 0" axis="0 0 1"/>
            <geom pos="-0.65 0 -0.17" quat="0.5 0.5 0.5 -0.5" type="mesh" rgba="0.623529 0.666667 0.701961 1" mesh="rim"/>
          </body>
        </body>
        <body name="link_r_3" pos="-0.480644 -0.2775 -0.08" quat="0 0.521962 -0.852969 0">
          <inertial pos="0 -0.3775 -0.0175" quat="0.5 0.5 -0.5 0.5" mass="7.87695" diaginertia="0.463448 0.457843 0.00721361"/>
          <joint name="link3_r_j" pos="0 0 0" axis="0 0 1"/>
          <geom pos="0 -0.3775 -0.0175" quat="0.5 0.5 0.5 -0.5" type="mesh" rgba="1 0 0 1" mesh="link_r"/>
          <body name="rim_3" pos="0 -0.755 0" quat="0 -0.926466 0.376377 0">
            <inertial pos="-0.105737 -0.314431 -0.170074" quat="0.681036 0.684974 -0.183545 -0.182515" mass="120.247" diaginertia="34.4863 21.9667 15.7317"/>
            <joint name="rim3_r_j" pos="0 0 0" axis="0 0 1"/>
            <geom pos="-0.65 0 -0.17" quat="0.5 0.5 0.5 -0.5" type="mesh" rgba="0.623529 0.666667 0.701961 1" mesh="rim"/>
          </body>
        </body>
      </body>
    </body>
  </worldbody>
</mujoco>

```

</details>

Here is a screenshot / video, illustrating my question:
This is my parallel robot model.

[image]

yuvaltassa, 2023-08-21:
Yes, it's possible, you need to use "connect" equality constraints. Besides reading the docs, a quick search in issues would have helped you [find](https://github.com/deepmind/mujoco/issues/84) what you need.

closed 2023-08-21 by yuvaltassa

hi-an, 2023-08-24:
I'm trying to use "connect" equality constraints but I have problems.
I want to connect a new hinge joint to the exact position and the position is different with the already connected.

I tried to place the anchor the desired position and add cylinder geoms to the desired position but it doesn't work.
Here is a description picture and adding code.
[image]

```xml
<equality>
    <connect name="rim_b_j" body1="b_point_1" body2="pin_point_1" anchor="-0.01 0.755 -0.015"/>
    <connect name="rim2_b_j" body1="b_point_2" body2="pin_point_2" anchor="-0.01 0.755 -0.015"/>
    <connect name="rim3_b_j" body1="b_point_3" body2="pin_point_3" anchor="-0.01 0.755 -0.015"/>
</equality>
```

yuvaltassa, 2023-08-24:
It's not entirely clear what is the problem you are experiencing. I recommend:
1. Enable constraint visualisation. This will show you the two points that the equality is trying to push together.
2. Perhaps examining [this model](https://github.com/deepmind/mujoco/blob/main/test/engine/testdata/weld.xml) (again, with constraint visualisation enabled), will help you further.